realsenseai/realsense-ros #2006: args for static_transform_publisher for d435i and l515

aditdoshi333, 2021-07-29:
Hello,

I am trying to use rtabmap with multiple d435i and l515 for mapping. Rtabmap supports mapping with multiple rgbd cameras. I am using the following example for the same.
Link to example: https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_two_kinects.launch

They are using some static tf publisher. Similar like

`<node pkg="tf" type="static_transform_publisher" name="base_to_camera1_tf"
      args="0.0 0.0 0.0 0.0 0.0 0.0 /base_link /camera1_link 100" />
  <node pkg="tf" type="static_transform_publisher" name="base_to_camera2_tf"
      args="-0.1325 -0.1975 0.0 -1.570796327 0.0 0.0 /base_link /camera2_link 100" />`

I am not able to understand how to find args of static transform publisher. My use case is as following

- Two d435i (60° apart)
- Three d435i (60° apart)
- Two l515 (90° apart)

MartyG-RealSense, 2021-07-29:
Hi @aditdoshi333  Before starting, may I confirm please that any solution must involve defining the static transform between cameras inside an RTABMAP launch file?

aditdoshi333, 2021-07-29:
@MartyG-RealSense Inside or outside doesn't matter actually. I just want to understand how these transforms are defined.

MartyG-RealSense, 2021-07-29:
The RealSense ROS wrapper takes the approach with the pairing of a 400 Series camera and a T265 Tracking Camera of providing a launch file called **rs_d400_and_t265.launch** and defining the **static_transform_publisher** instruction at the bottom of the launch file.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_d400_and_t265.launch#L49

A list of args are defined for both cameras (camera1 and camera2) and then the static_transform_publisher line at the bottom of the script calls upon the defined camera1 and camera2.

https://github.com/IntelRealSense/realsense-ros/blob/6a9cc994e3a9f63ddff1938a86ff04ca990d2551/realsense2_camera/launch/rs_d400_and_t265.launch#L22-L46

I would speculate that you could create multiple-camera custom launch files for D435i only, L515 only or D435i+L515 that take account of static transform by editing the details in the launch file.  I do not know of an existing example to check as a reference though.

closed 2021-08-11 by MartyG-RealSense