realsenseai/realsense-ros #2008: release_power_failed

fiorano10, 2021-07-30:
We have 4 cameras connected to the motherboard, each has its own USB 3.1 Gen 1 port.

Running the following config
Ubuntu 18.04
ROS Melodic
Librealsense v2.42
realsense-ros v2.2.22
Camera fw: 5.12.11

```
 26/07 10:30:11,273 ERROR [140158473058048] (types.h:315) xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 26/07 10:30:11,273 WARNING [140157256722176] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 26/07 10:30:11,273 ERROR [140158473058048] (sensor.cpp:547) release_power failed: xioctl(VIDIOC_REQBUFS) failed Last Error: No such device

 26/07 10:30:11,274 WARNING [140157785200384] (backend-hid.h:48) write_fs_attribute with 0 failed. The specified path /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.5/0003:8086:0B3A.0001/HID-SENSOR-200076.3.auto/iio:device1/buffer/enable is not valid
 26/07 10:30:11,274 WARNING [140157785200384] (backend-hid.cpp:715) HID set_power 0 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.5/0003:8086:0B3A.0001/HID-SENSOR-200076.3.auto/iio:device1/buffer/enable

 26/07 10:30:11,275 WARNING [140157768414976] (backend-hid.h:48) write_fs_attribute with 0 failed. The specified path /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.5/0003:8086:0B3A.0001/HID-SENSOR-200073.2.auto/iio:device0/buffer/enable is not valid
 26/07 10:30:11,275 WARNING [140157768414976] (backend-hid.cpp:715) HID set_power 0 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.5/0003:8086:0B3A.0001/HID-SENSOR-200073.2.auto/iio:device0/buffer/enable
 26/07 10:30:11,275 WARNING [140157256722176] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 26/07 10:30:11,275 WARNING [140157256722176] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 26/07 10:30:11,275 WARNING [140157256722176] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 26/07 10:30:11,275 WARNING [140157256722176] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 26/07 10:30:11,275 WARNING [140157256722176] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 26/07 10:30:11,275 WARNING [140157256722176] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 26/07 10:30:11,275 WARNING [140157256722176] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 26/07 10:30:11,275 WARNING [140157256722176] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 26/07 10:30:11,275 ERROR [140158473058048] (types.h:315) Failed to enable_sensor /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.5/0003:8086:0B3A.0001/HID-SENSOR-2000e1.4.auto/enable_sensor Last Error: No such file or directory

```

MartyG-RealSense, 2021-07-30:
Hi @fiorano10   There were a small number of cases with this error in SDK 2.41.0, just before SDK version 2.42.0 was released.  The errors did not occur in SDK 2.40. 

https://github.com/IntelRealSense/librealsense/issues/8318

 The problem was found to occur when a stream was closed or frames were released and was attributed to v4l changes that were reverted by a fix in 2.42.0. 

https://github.com/IntelRealSense/librealsense/issues/8154#issuecomment-767175826

[image]

If you are currently using SDK 2.42.0 and it is indeed this problem that is affecting you then I am not certain why the reversion did not solve it for you.  You can of course try downgrading to SDK 2.40.0 to see whether the issue disappears.  Bear in mind that you would also need to downgrade to ROs wrapper **2.2.20** to match with SDK 2.40.0.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.20

fiorano10, 2021-07-30:
I can confirm that we're seeing this on v2.42 and downgrading is not really a viable resolution. Can this be resolved in the new release?

MartyG-RealSense, 2021-07-30:
I do not have knowledge of any specific feature inbetween the 2.43.0 and 2.48.0 releases that would be guaranteed to resolve your problem, though you are certainly welcome to try 2.48.0 and matching ROS wrapper 2.3.1 and see if it provides improvement.

MartyG-RealSense, 2021-08-04:
Were you able to test with 2.48.0 and wrapper 2.3.1 to see whether the error still occurs, please?

closed 2021-08-16 by MartyG-RealSense