realsenseai/realsense-ros #2012: Custom json file

greendev32, 2021-07-30:
Hi, I'd like the preset setting to be set to short range when I run a launch file, so that the camera can find depths closer than 200mm. I added a reference to a .json file downloaded from the realsense-viewer to my launch file so that the camera will have the same settings seen in the viewer. For some reason the settings do not take effect. I know the file is found because the launch output will give me a warning if the filepath to the .json file is wrong. Also, anything I change in the launch file is ignored, like the output width and height.

The camera I'm using is the L515. Any help would be appreciated!

Here is my output when I run rs_camera.launch.

roslaunch realsense2_camera rs_camera.launch
... logging to /home/adev/.ros/log/f20cb66a-f18d-11eb-87a6-fcb3bcdfa6c4/roslaunch-Adev-8274.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Adev:34055/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
[... 72 lines cut ...]
[ INFO] [1627687754.782087409]: Device FW version: 01.05.08.01
[ INFO] [1627687754.782124523]: Device Product ID: 0x0B64
[ INFO] [1627687754.782156953]: Enable PointCloud: Off
[ INFO] [1627687754.782205444]: Align Depth: On
[ INFO] [1627687754.782250648]: Sync Mode: On
[ INFO] [1627687754.782343814]: Device Sensors: 
[ INFO] [1627687754.783174148]: L500 Depth Sensor was found.
[ INFO] [1627687754.788696607]: RGB Camera was found.
[ INFO] [1627687754.789097723]: Motion Module was found.
[ INFO] [1627687754.789971559]: num_filters: 1
[ INFO] [1627687754.790026183]: Setting Dynamic reconfig parameters.
[ INFO] [1627687754.841714289]: Done Setting Dynamic reconfig parameters.
[ INFO] [1627687754.841999456]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1627687754.842952639]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1627687754.843678704]: setupPublishers...
[ INFO] [1627687754.845232066]: Expected frequency for depth = 30.00000
[ INFO] [1627687754.862030702]: Expected frequency for color = 30.00000
[ INFO] [1627687754.873514147]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1627687754.885455325]: setupStreams...
[ INFO] [1627687754.937577234]: insert Depth to L500 Depth Sensor
[ INFO] [1627687754.937623663]: insert Color to RGB Camera
[ INFO] [1627687754.978691313]: SELECTED BASE:Depth, 0
[ INFO] [1627687755.130773468]: RealSense Node Is Up!
[ WARN] [1627687755.145328326]: 
[ WARN] [1627687755.145371372]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.

MartyG-RealSense, 2021-07-31:
Hi @greendev32  Whilst not necessarily related to your json problem, a good starting point in investigating your case will be to address the setion of your log below:

**[ INFO] [1627687754.439254726]: RealSense ROS v2.3.1
[ INFO] [1627687754.439338998]: Built with LibRealSense v2.45.0
[ INFO] [1627687754.439387062]: Running with LibRealSense v2.48.0
[ WARN] [1627687754.439427662]: ***************************************************
[ WARN] [1627687754.439473566]: ** running with a different librealsense version **
[ WARN] [1627687754.439499578]: ** than the one the wrapper was compiled with! **
[ WARN] [1627687754.439535329]: *****************************************************

This indicates that the RealSense ROS wrapper was first built on librealsense SDK 2.45.0 ans then you upgraded to librealsense 2.48.0.  When the librealsense version is updated, the RealSense ROS wrapper needs to be rebuilt on order to remove the above warning message.  Thanks!

greendev32, 2021-07-31:
Hi Marty, I just re-installed the ros wrapper using the instructions on the homepage:
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
cd ..

etc..

and the wrapper version is still 2.45.0. I don't think this is causing my problem either, however it would be good to get them on the same version. Should I be checking out a different branch? 
Thanks
-Allison

MartyG-RealSense, 2021-07-31:
RealSense ROS is the wrapper and librealsense is the SDK.  The ROS wrapper (2.3.1) is the part that should be rebuilt.  It looks from your description above that this is what you have done.

Ideally the log should state:

[ INFO] [1627687754.439254726]: RealSense ROS v2.3.1
[ INFO] [1627687754.439338998]: Built with LibRealSense v2.48.0
[ INFO] [1627687754.439387062]: Running with LibRealSense v2.48.0

You can confirm which librealsense SDK version is currently installed by launching the **RealSense Viewer** program and looking at the version number at the top of the Viewer window.

You can use a version number directly with git checkout instead of retrieving it with the tag command.

**git checkout 2.3.1**

greendev32, 2021-07-31:
Even when I checkout the 3.2.1 branch and rebuild the ROS wrapper, I still get the same warning message. 

[ INFO] [1627754516.866077932]: RealSense ROS v2.3.1
[ INFO] [1627754516.866107416]: Built with LibRealSense v2.45.0
[ INFO] [1627754516.866122780]: Running with LibRealSense v2.48.0
[ WARN] [1627754516.866135335]: ***************************************************
[ WARN] [1627754516.866145548]: ** running with a different librealsense version **
[ WARN] [1627754516.866155028]: ** than the one the wrapper was compiled with!   **
[ WARN] [1627754516.866165264]: ***************************************************

-Allison

MartyG-RealSense, 2021-07-31:
Okay, let's put the librealsense version issue aside for a moment and return to your original question.  "I'd like the preset setting to be set to short range when I run a launch file, so that the camera can find depths closer than 200mm".

Do you mean that you want to set a maximum depth distance of 200 mm and exclude depth data beyond that distance?  Ifcso then you do not need a json for that as you can set a maximum distance in meters with the **clip_distance** launch setting.  For example, for 200 mm (0.2 meters):

roslaunch realsense2_camera rs_camera.launch clip_distance:=0.2

greendev32, 2021-07-31:
I gave that a try, but it output all 0's for all ranges (even under 200). What I really want is for my output in ROS to match what I see in the viewer when I have short range selected as the preset. In the viewer when I hover over the depth image, every pixel has the correct depth for the scene I'm pointing at. When I run the launch file with no parameters, it seems like it's defaulting to long rage preset. It can calculate the range further away, but not when the object is closer than 200mm. I appreciate your help with this!

MartyG-RealSense, 2021-07-31:
I wonder if it is because the L515 camera model has a default minimum depth sensing distance of 0.25 cm (255 mm) according to its technical specification and so the depth values are not appearing because the camera simply cannot observe depth at that close range, at least in the ROS wrapper.  In librealsense and the Viewer there are ways to reduce the minimum depth distance to allow sensing of closer ranges, such as increasing the value of a setting called **disparity shift**.

greendev32, 2021-07-31:
Previously, I used a D435 camera, set all settings as desired in the realsense viewer, exported the json file and then called the json file inside the launch file on the appropriate line- this worked as expected, the camera performed in the ROS wrapper exactly as it did in the realsense viewer. I'm showing a portion of the launch file below where I changed the json file path. I expected this to work the same way with the L515 where in the realsense viewer I can get ranges down to 100mm, but despite the launch settings stating that the json file was loaded, the camera does not function as it does in the viewer. The json file that is created by the realsense viewer app should change the disparity shift along with several other camera parameters from the default settings. I've just noticed that disparity shift is not a parameter in the .json filed produced by the viewer for the L515, so I don't know what .json file to modify to change the disparity shift as you suggest. So I believe my question is now - what json file is the L515 camera using in ros? Then I can change the settings to match the viewer and it will probably solve my problem. 

launch file snippet:
 <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default="/home/adev/ros_workspace/src/realsense-ros/realsense2_camera/launch/test2.json"/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

Viewer generated .json file for L515
{
    "Alternate IR": 0.0,
    "Apd Temperature": 35.0911178588867,
    "Confidence Threshold": 1,
    "Depth Offset": 4.5,
    "Depth Units": 0.000250000011874363,
    "Digital Gain": 2,
    "Enable IR Reflectivity": 0.0,
    "Enable Max Usable Range": 0.0,
    "Error Polling Enabled": 1,
    "Frames Queue Size": 16,
    "Freefall Detection Enabled": 1,
    "Global Time Enabled": 0.0,
    "Host Performance": 0.0,
    "Humidity Temperature": 37.8816680908203,
    "Inter Cam Sync Mode": 0.0,
    "Invalidation Bypass": 0.0,
    "LDD temperature": 44.0405693054199,
    "Laser Power": 100,
    "Ma Temperature": 36.7489242553711,
    "Mc Temperature": 35.5791397094727,
    "Min Distance": 100,
    "Noise Estimation": 984,
    "Noise Filtering": 4,
    "Post Processing Sharpening": 1,
    "Pre Processing Sharpening": 0.0,
    "Receiver Gain": 18,
    "Sensor Mode": 0.0,
    "Visual Preset": 0.0,
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "480",
    "stream-ir-format": "Y8",
    "stream-width": "640"
}

greendev32, 2021-07-31:
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default="/home/adev/ros_workspace/src/realsense-ros/realsense2_camera/launch/test2.json"/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

greendev32, 2021-07-31:
launch file snippet: 
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default="/home/adev/ros_workspace/src/realsense-ros/realsense2_camera/launch/test2.json"/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

greendev32, 2021-07-31:
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default="~desktop/test2.json"/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

MartyG-RealSense, 2021-07-31:
The D435 and D435i camera models have a default minimum depth sensing distance of 0.1 meters / 100 mm and so can depth sense at closer range from objects / surfaces than the other RealSense models without needing the assistance of settings such as disparity shift.

I checked the settings availabile on L515 and confirmed that disparity shift or export of it into an L515 json was not supported on that model.  I do apologize.

I believe that if a path to a json file is not defined then the default values of settings for a particular camera model are used.

Is there any possibility of moving the camera further away from the observed object so that the object is above the L515 minimum depth distance range?

greendev32, 2021-07-31:
launch file snippet:

 arg name="serial_no"           default=""/>
 arg name="usb_port_id"         default=""/>
  arg name="device_type"         default=""/>
  arg name="json_file_path"      default="~desktop/test2.json"/>
  arg name="camera"              default="camera"/>
 arg name="tf_prefix"           default="$(arg camera)"/>
 arg name="external_manager"    default="false"/>
  arg name="manager"             default="realsense2_camera_manager"/>
  arg name="output"              default="screen"/>
  arg name="respawn"              default="false"/>

greendev32, 2021-07-31:
I must reiterate that the L515 Camera works great in the realsense viewer all the way down to 100mm  when using the built-in short range settings. This means that the camera can work in short ranges and that the settings can be changed. I am defining a path to the json file, but the json file I am using does not include all camera parameters necessary to adjust the camera to short range mode. 
Right now the camera, while running in ROS, appears to be in long range mode which sets a minimum distance of 490mm and i've confirmed that I don't get measurements less than 490mm while running in ROS. I'm looking for a way to reduce the minimum range.

MartyG-RealSense, 2021-07-31:
As the RealSense ROS wrapper does not have a setting for a minimum distance (just maximum distance with **clip_distance**), the only means that I know of for setting a minimum distance is the 'RangeMin' setting of **RTABMAP**.  I do not have any references for its use with L515 though, so I cannot advise whether it would make a difference to your particular situation.

https://github.com/IntelRealSense/realsense-ros/issues/1859#issuecomment-836933868

A RealSense ROS user said that they were able to select the Short Range preset via a drop down in the dynamic_reconfigure interface of ROS, though I have not tested this myself.

https://github.com/IntelRealSense/realsense-ros/issues/1294

You can access the dynamic_reconfigure interface by entering the line below into the ROS terminal.

**rosrun rqt_reconfigure rqt_reconfigure**

MartyG-RealSense, 2021-08-06:
Hi @greendev32  Do you require further assistance with this case, please?  Thanks!

greendev32, 2021-08-09:
Hi, I apologize for the delayed response. The dynamic reconfigure interface solved my problem! I am no able to get aligned range measurements down to about 8 cm in range through the ROS topic aligned_depth_to_color/image_raw.

thanks for your help-

closed 2021-08-09 by greendev32

MartyG-RealSense, 2021-08-09:
Great to hear that you achieved a solution, @greendev32 - thanks very much for the update!