realsenseai/realsense-ros #2014: D435: Max 7 fps when streaming `/color/image_raw` and `/depth/color/points`

admantium-sg, 2021-08-01:
My goal is to use the D435 with ROS1 on an headless RasperryPi4B to stream image data, `/color/image_raw ` and depth data, `/depth/color/points` as well as  `depth/image_rect_raw`, via WLAN to a Linux Workstation.

The concrete versions are:

- Raspberry Pi4B
- Ubuntu 20.04 server (headless)
- ROS1 Noetic (installed as Debian package)
- librealsense 2.0.48 (installed from source)
- ros-realsense 2.3.1 (installed from source)

Starting point to run ros-realsense on the Raspberry Pi4B is the default command:

```bash
roslaunch realsense2_camera rs_camera.launch camera:=camera1
```

Which I systematically evolved to 

```bash
roslaunch realsense2_camera rs_camera.launch camera:=camera1 depth_width:=640 color_width:=640 depth_height:=480 color_height:=480 depth_fps:=6 color_fps:=6 pointcloud_texture_stream:=RS2_STREAM_COLOR enable_sync:=false align_depth:=false initial_reset:=false filters:=pointcloud
```

Systematically setting these parameters, and measuring the performance with `rostopic hz` from my Linux workstation, I came to the following results:

| Parameter           |            |            |            |            |            |            |            |            |
| ------------------ | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- |
| depth&color width  | not set    | not set    | 640        | 640        | 640        | 640        | 640        | 640        |
| depth&color height | not set    | not set    | 480        | 480        | 480        | 480        | 480        | 480        |
| depth\_fps         | not set    | not set    | 5          | 5          | 5          | 6          | 6          | 6          |
| color\_fps         | not set    | not set    | 5          | 5          | 5          | 6          | 6          | 6          |
| initial\_reset     | TRUE       | TRUE       |            | TRUE       | TRUE       | TRUE       | TRUE       | TRUE       |
| enable\_sync       |            |            |            | TRUE       |            |            |            |            |
| align\_depth       |            |            |            |            |            |            |            |            |
| filters            | pointcloud | pointcloud | pointcloud | pointcloud | pointcloud | pointcloud | pointcloud | pointcloud |
| texture\_stream    | any        | color      | any        | any        | color      | color      | color      | color      |
| ordered\_pc        |            |            |            |            |            |            | yes        |            |
|                    |            |            |            |            |            |            |            |            |
| Topic HZ           |            |            |            |            |            |            |            |            |
| ------------------ | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- |
| /color/image\_raw   | 6.5        | no data    | 1.5        | 1.5        | no data    | 6          | 6          | 6          |
| /depth/color/points | no data    | no data    | no data    | no data    | no data    | 3          | 1          | 3          |
| /depth/image-rect   | 7          | 6.5        | no data    | no data    | 6          | 6          | 6          | 6          |

So, to summarize:
- I can not get more than 7FPS for any stream. 
- I can only get pointcloud data when manually setting the fps to 6

A network ping between the Raspberry Pi4B and the Linux workstation is 2ms.

Why is the performance so slow? Is it not possible to stream with 30fps in ROS1?

MartyG-RealSense, 2021-08-01:
Hi @admantium-sg  In the lik below, a RealSense ROS user tried converting the SDK's open-source ethernet networking system to use ROS networking protocols and had similar performance limitations with the project's Raspberry Pi 4.  They said "I only get an average of 2Hz when I run rostopic hz on the camera/depth/image_rect_raw, and 10Hz on camera/color/image_rect_raw".

https://github.com/IntelRealSense/realsense-ros/issues/1808

There was no solution achieved at the conclusion of that particular case, unfortunately.

MartyG-RealSense, 2021-08-14:
Case closed due to no further comments received.

closed 2021-08-14 by MartyG-RealSense

DavidePatria, 2021-10-21:
I'm actually working with /camera/image_raw (or similar name) and camera/aligned_color_with_depth/image_raw (or similar) so at the momento I'm not using the pointcloud. I would use this setup to register a rosbag to later use it for slam purposes

DavidePatria, 2021-10-21:
I'll definitely look into that, thank you.
Anyway, the ability to use it has I described would make things much easier, given that the file formats (slam and real sense) are probably not compatible and only shared ros as a common platform. The slam I'm using is openvslam, which allows for the usage of mp4 files, but doesn't mention anything about files containing depth, so I'm afraid of compatibility therefore decided to use ros, that is anyway used for my project, to exchange data.

Another solution would be to make the process use all the available threads at the maximum speed, since I've noticed that only one is maxed out and the others are not fully used, but I guess this one is a bit more complicated.