realsenseai/realsense-ros #2015: [ERROR] Build fails.

adbidwai, 2021-08-02:
Hello,
My system configurations are as follows : 
**OS** : Ubuntu 18.04
**ROS Distro** : ROS Melodic
I am facing the following issue when I try to build the `development` branch of `realsense-ros` from source. Is it safe to just change the `2.48.0` at line 44 of `/home/adbidwai/nus_ws/src/realsense-ros/realsense2_camera/CMakeLists.txt` to `2.45.0`?
```
Errors     << realsense2_camera:cmake /home/adbidwai/nus_ws/logs/realsense2_camera/build.cmake.004.log          
CMake Warning at /home/adbidwai/nus_ws/src/realsense-ros/realsense2_camera/CMakeLists.txt:44 (find_package):
  Could not find a configuration file for package "realsense2" that is
  compatible with requested version "2.48.0".

  The following configuration files were considered but not accepted:

    /opt/ros/melodic/lib/cmake/realsense2/realsense2Config.cmake, version: 2.45.0

CMake Error at /home/adbidwai/nus_ws/src/realsense-ros/realsense2_camera/CMakeLists.txt:48 (message):

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

```

MartyG-RealSense, 2021-08-02:
Hi @adbidwai  The error message is highlighting that RealSense SDK 2.0 (librealsense) is missing.  When building the RealSense ROs wrapper from source with **Method 2**, the librealsense SDK should be installed first and then the ROS wrapper built separately from source secondly after that.  

Each time that librealsense is updated to a new version, the RealSense ROS wrapper should be built again.  Ideally, you should match a particular ROS wrapper version to a specific librealsense SDK version listed in that wrapper verision's release notes.

https://github.com/IntelRealSense/realsense-ros/releases

For example, the recomennded ROS1 wrapper version to use with librealsense 2.48.0 is **2.3.1**, whilst for librealsense 2.45.0 the recommend ROS1 wrapper is **2.3.0**

adbidwai, 2021-08-03:
Thank you so much for your answer @MartyG-RealSense 
So this is what I have understood,
1) `realsense-ros` package releases are tied with specific versions of `librealsense`, details about which I can look under the "Releases" section in the GitHub repository, right?
2) So, now I have 2 options, either install the correct `librealsense` library version or checkout to the correct version of the `realsense-ros` package, right?

MartyG-RealSense, 2021-08-03:
You are very welcome.   :)

1.  Yes, that is correct.

2. If the **Method 1** processs of installing librealsense and the wrapper together from packages instead of installing them separately is used then you are guaranteed a compatible match between SDK and wrapper versions. 

3.  The version installed by the packages with **sudo apt-get install ros-melodic-librealsense** may depend on how long ago the latest librealsense version was released.  If it is close to the release date of a new librealsense then the packages may still be using the previous librealsense version and its matching wrapper, as it can take time to update the packages.  Building from source code with **Method 2** ensures that you are installing the latest wrapper version.

MartyG-RealSense, 2021-08-09:
Hi @adbidwai  Do you require further assistance with this case, please?  Thanks!

adbidwai, 2021-08-09:
@MartyG-RealSense Hi again, I'm sorry I did not close this issue. I have got all that I wanted to know, Thanks a lot! You can close this issue :+1:

MartyG-RealSense, 2021-08-09:
No problem at all @adbidwai - it's excellent to hear that you are satisfied with the outcome of the case.  Thanks for the update!

closed 2021-08-09 by MartyG-RealSense