realsenseai/realsense-ros #2016: rs-enumerate-devices /realsense-viewer not working after launching realsense2_camera node

joelbudu, 2021-08-02:
rs-enumerate-devices /realsense-viewer  doesn't seem to be working after launching a realsense2_camera node. It only works after physically disconnecting and reconnecting the USB cable.

MartyG-RealSense, 2021-08-02:
Hi @joelbudu  RealSense programs currently streaming with the camera should be closed before running another RealSense program, because the program that was run first will be blocking access to other applications using the streams that it currently has a 'claim' on.

Unplugging the camera from the USB ports stops all currently active streams and so makes all streams available to use for the next program that enables those streams and 'claims' them.

joelbudu, 2021-08-02:
Dear @MartyG-RealSense . This issue happens even after closing the running application.

MartyG-RealSense, 2021-08-02:
I have seen a couple of past cases similar to your situation, where the Viewer does not work after first running RealSense ROS.  The link below has one such case.

https://github.com/IntelRealSense/realsense-ros/issues/1187#issuecomment-624895849

Since librealsense and the RealSense ROS wrapper operate independently and you do not need the Viewer open in order to use the ROS wrapper, a simple workaround may be to not use librealsense programs in sessions where you are using RealSense ROS.   The RealSense ROS wrapper publishes a topic called **/diagnostics** that you can check for camera information.

MartyG-RealSense, 2021-08-09:
Hi @joelbudu  Do you require further assistance with this case, please?  Thanks!

joelbudu, 2021-08-09:
I did realize based on the links you shared that these is some disparity between the version of librealsense installed and the realsense-ros package I have installed.

I tried to build the latest realsesnse-ros from source but I'm receiving this warning now:

`[ INFO] [1628513177.458658821]: RealSense ROS v2.3.1
[ INFO] [1628513177.458721123]: Built with LibRealSense v2.45.0
[ INFO] [1628513177.458755288]: Running with LibRealSense v2.48.0
[ WARN] [1628513177.458783389]: ***************************************************
[ WARN] [1628513177.458803737]: ** running with a different librealsense version **
[ WARN] [1628513177.458824719]: ** than the one the wrapper was compiled with!   **
[ WARN] [1628513177.458848879]: ***************************************************
`

joelbudu, 2021-08-09:
Update: @MartyG-RealSense Apparently I had another librealsense2 package installed: ros-melodic-librealsense2 that was also creating a conflict. With matching librealsense2 versions this works fine and there is no issue. Thank you very much for the guidance.

closed 2021-08-09 by joelbudu

MartyG-RealSense, 2021-08-09:
Great news @joelbudu that you found a solution - thanks very much for the update   :)