realsenseai/realsense-ros #2017: Not able to work with compressed image topic

aditdoshi333, 2021-08-03:
Hello,

I want to record the .bag file for around 20 to 30 mins and running rtabmap ros on the same. If I record raw image topics like the following:
`
/camera/aligned_depth_to_color/image_raw  
/camera/color/image_raw/compressed 
/camera/color/camera_info
`
The size of the bag file is crossing ~200gb. I can't use such a huge bag file. So I am thinking to use compressed image topics like the following:

`
/camera/color/image_raw/compressed /camera/aligned_depth_to_color/image_raw/compressed /camera/color/camera_info
`
But I am getting some errors while running the realsense node

**Realsense Node**
`[ERROR] [1627983160.926561915]: Client [/rtabmap/rgbd_odometry] wants topic /camera/color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
[ERROR] [1627983160.926853260]: Client [/rtabmap/rgbd_odometry] wants topic /camera/aligned_depth_to_color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
[ERROR] [1627983160.929256854]: Client [/rtabmap/rtabmap] wants topic /camera/color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
[ERROR] [1627983160.929974768]: Client [/rtabmap/rtabmap] wants topic /camera/aligned_depth_to_color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
[ERROR] [1627983160.934051363]: Client [/rtabmap/rtabmapviz] wants topic /camera/color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
[ERROR] [1627983160.934547045]: Client [/rtabmap/rtabmapviz] wants topic /camera/aligned_depth_to_color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
`

**Rtabmap Node**

`[ WARN] [1627984776.467773886]: [image_transport] It looks like you are trying to subscribe directly to a transport-specific image topic '/camera/color/image_raw/compressed', in which case you will likely get a connection error. Try subscribing to the base topic '/camera/color/image_raw' instead with parameter ~image_transport set to 'compressed' (on the command line, _image_transport:=compressed). See http://ros.org/wiki/image_transport for details.
[ WARN] [1627984776.469969621]: [image_transport] It looks like you are trying to subscribe directly to a transport-specific image topic '/camera/aligned_depth_to_color/image_raw/compressed', in which case you will likely get a connection error. Try subscribing to the base topic '/camera/aligned_depth_to_color/image_raw' instead with parameter ~image_transport set to 'compressed' (on the command line, _image_transport:=compressed). See http://ros.org/wiki/image_transport for details.
`

Any light is appreciated. 
Thank you

MartyG-RealSense, 2021-08-03:
Hi @aditdoshi333  You could divide a single huge bag into multiple smaller bags using the **--split** command of rosbag record.  It enables you to specify a maximum time length or file size, and when that defined limit is reached then rosbag record automatically stops the recording and begins a new bag file.  For example:

**$ rosbag record --split --size=1024 /chatter
$ rosbag record --split --duration=30 /chatter
$ rosbag record --split --duration=5m /chatter
$ rosbag record --split --duration=2h /chatter**

http://wiki.ros.org/rosbag/Commandline#rosbag_record

If you wish to use compressed-image topics then a section about compression within RealSense ROS documentation on the RealSense website recommends having the **compressed-image-transport** and the 
**compressed-depth-image-transport** packages installed.

https://dev.intelrealsense.com/docs/ros-wrapper#section-compression-packages

aditdoshi333, 2021-08-03:
Hello, @MartyG-RealSense Thanks for the reply. I can't use the split option as I have limited disk space. I wish to use compressed topics of the real sense node. But I am not sure how to decompress them? Trying to do with image_transport but was not able to do. Can you please help with the decompression of the topics?

aditdoshi333, 2021-08-03:
@MartyG-RealSense I am able to use compress colour image. But while uncompressing depth image I am getting the following error.

`[ERROR] [1627993449.847817656]: Unsupported image format: 16UC1; jpeg compressed `

The following cmd used for image transport 

`rosrun image_transport republish compressedDepth  in:=/camera/aligned_depth_to_color/image_raw raw out:=/full_depth`

MartyG-RealSense, 2021-08-03:
The link below has an example of how another RealSense ROS user handled the problem of using compressed depth by addng params to their launch file.

https://github.com/IntelRealSense/realsense-ros/issues/1484#issuecomment-721256624

closed 2021-08-16 by MartyG-RealSense