realsenseai/realsense-ros #2021: Cannot found device after initial reset

lantica, 2021-08-04:
Hello, when I launch the node with `initial_rest:=true`, it seems that the node cannot find the device after power cycle.

The versions are as follow:
- ubuntu 20.04
- ros noetic
- realsense-ros 2.2.22 (installed by `apt-get install ros-noetic-realsense2-camera`)
- librealsense 2.42.0 
- device firmware 05.11.01.100

Logs: 
```
-- Logs begin at Tue 2021-08-03 07:49:37 UTC, end at Wed 2021-08-04 02:56:56 UTC. --
Aug 04 02:52:40 localhost 295c98dd546f[1126]: INFO  [ros.nodelet] Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:
Aug 04 02:52:40 localhost 295c98dd546f[1126]: 
Aug 04 02:52:40 localhost 295c98dd546f[1126]: INFO  [ros.nodelet] /camera/depth/color/points -> /camera/points
Aug 04 02:52:40 localhost 295c98dd546f[1126]: 
Aug 04 02:52:40 localhost 295c98dd546f[1126]: INFO  [ros.roscpp] waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...
Aug 04 02:52:40 localhost 295c98dd546f[1126]: 
Aug 04 02:52:40 localhost 295c98dd546f[1126]: INFO  [ros.roscpp] waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...
Aug 04 02:52:40 localhost 295c98dd546f[1126]: 
Aug 04 02:52:40 localhost 295c98dd546f[1126]: INFO  [ros.nodelet] Initializing nodelet with 4 worker threads.
Aug 04 02:52:40 localhost 295c98dd546f[1126]: 
Aug 04 02:52:40 localhost 295c98dd546f[1126]: INFO  [ros.roscpp] waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.
Aug 04 02:52:40 localhost 295c98dd546f[1126]: 
Aug 04 02:52:40 localhost 295c98dd546f[1126]: INFO  [ros.roscpp] waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.
Aug 04 02:52:40 localhost 295c98dd546f[1126]: 
Aug 04 02:52:40 localhost 295c98dd546f[1126]: INFO  [ros.realsense2_camera] RealSense ROS v2.2.22
Aug 04 02:52:40 localhost 295c98dd546f[1126]: 
Aug 04 02:52:40 localhost 295c98dd546f[1126]: INFO  [ros.realsense2_camera] Built with LibRealSense v2.42.0
Aug 04 02:52:40 localhost 295c98dd546f[1126]: 
Aug 04 02:52:40 localhost 295c98dd546f[1126]: INFO  [ros.realsense2_camera] Running with LibRealSense v2.42.0
Aug 04 02:52:40 localhost 295c98dd546f[1126]: 
Aug 04 02:52:40 localhost 295c98dd546f[1126]: INFO  [ros.realsense2_camera]  
Aug 04 02:52:40 localhost 295c98dd546f[1126]: 
Aug 04 02:52:41 localhost 295c98dd546f[1126]: INFO  [ros.realsense2_camera] Device with serial number 047322070642 was found.
Aug 04 02:52:41 localhost 295c98dd546f[1126]: 
Aug 04 02:52:41 localhost 295c98dd546f[1126]: INFO  [ros.realsense2_camera] Device with physical ID 2-2-2 was found.
Aug 04 02:52:41 localhost 295c98dd546f[1126]: 
Aug 04 02:52:41 localhost 295c98dd546f[1126]: INFO  [ros.realsense2_camera] Device with name Intel RealSense D435I was found.
Aug 04 02:52:41 localhost 295c98dd546f[1126]: 
Aug 04 02:52:41 localhost 295c98dd546f[1126]: INFO  [ros.realsense2_camera] Device with port number 2-2 was found.
Aug 04 02:52:41 localhost 295c98dd546f[1126]: 
Aug 04 02:52:41 localhost 295c98dd546f[1126]: INFO  [ros.realsense2_camera] Device USB type: 3.2
Aug 04 02:52:41 localhost 295c98dd546f[1126]: 
Aug 04 02:52:41 localhost 295c98dd546f[1126]: INFO  [ros.realsense2_camera] Resetting device...
Aug 04 02:52:41 localhost 295c98dd546f[1126]: 
Aug 04 02:52:42 localhost kernel: usb 2-2: USB disconnect, device number 2
Aug 04 02:52:42 localhost kernel: usb 2-2: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
Aug 04 02:52:42 localhost kernel: usb 2-2: New USB device found, idVendor=8086, idProduct=0b3a, bcdDevice=50.b1
Aug 04 02:52:42 localhost kernel: usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 04 02:52:42 localhost kernel: usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 435i
Aug 04 02:52:42 localhost kernel: usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
Aug 04 02:52:42 localhost kernel: usb 2-2: SerialNumber: 046523051862
Aug 04 02:52:42 localhost kernel: uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
Aug 04 02:52:42 localhost kernel: uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
Aug 04 02:52:42 localhost kernel: input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/input/input16
Aug 04 02:52:42 localhost kernel: uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
Aug 04 02:52:42 localhost kernel: uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
Aug 04 02:52:47 localhost 295c98dd546f[1126]: INFO  [ros.realsense2_camera]  
Aug 04 02:52:47 localhost 295c98dd546f[1126]: 
Aug 04 02:52:47 localhost 295c98dd546f[1126]:  04/08 10:52:47,229 ERROR [139953082185472] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
Aug 04 02:52:47 localhost 295c98dd546f[1126]: WARN  [ros.realsense2_camera] Device 1/1 failed with exception: failed to set power state
Aug 04 02:52:47 localhost 295c98dd546f[1126]: ERROR [ros.realsense2_camera] The requested device with serial number 047322070642 and device name containing d435 is NOT found. Will Try again.
Aug 04 02:52:47 localhost 295c98dd546f[1126]: 
Aug 04 02:52:47 localhost 295c98dd546f[1126]:  04/08 10:52:47,233 ERROR [139953193350912] (sensor.cpp:523) acquire_power failed: failed to set power state
Aug 04 02:52:47 localhost 295c98dd546f[1126]:  04/08 10:52:47,234 WARNING [139953193350912] (rs.cpp:306) null pointer passed for argument "device"

```

MartyG-RealSense, 2021-08-04:
Hi @lantica  Using **initial_reset:=true** in the launch is optional.  It is a helpful function to reset the camera at launch if there are problems with a launch.  If you can launch correctly without including it in your launch instruction then you do not need to use it.

lantica, 2021-08-04:
I'd prefer set this param to true as there is a high chance that the node cannot be launched properly when I kill and relaunch the node. It is very annoying that I have to power cycle the whole system to relaunch the node.

MartyG-RealSense, 2021-08-04:
If the camera is always plugged into the same USB port then you could try finding the camera by port ID during launch with the **usb_port_id** instruction instead of finding by serial number.

The ROS wrapper documentation states: "usb_port_id: will attach to the device with the given USB port (usb_port_id). i.e 4-1, 4-2 etc. 
 Default, ignore USB port when choosing a device".

For example:

**roslaunch realsense2_camera rs_camera.launch usb_port_id:=2-4-2**

fiorano10, 2021-08-10:
@MartyG-RealSense is it preferable to launch using the usb_port_id? Does it provide better stability?

MartyG-RealSense, 2021-08-10:
It is better if you can launch by serial number in a multiple-cam arrangement (a serial number should not be needed for only one camera), because it makes life easier if you or the end-user will be unplugging and re-plugging the cameras and cannot guarantee that the same camera will be put in the same port every time.  

The usb_port_id instruction was designed as an alternative to serial numbers in multiple-camera situations where identifying the camera by serial number was not working.  So if the camera cannot be found with serial number and it is always going to remain inserted in the same USB port then it is fine to consider using usb_port_id.

closed 2021-09-27 by MartyG-RealSense