realsenseai/realsense-ros #2022: process has died.

github-lile, 2021-08-05:
SUMMARY
========

PARAMETERS
 * /left_camera/realsense2_camera/accel_fps: 250
 * /left_camera/realsense2_camera/accel_frame_id: left_camera_accel...
 * /left_camera/realsense2_camera/accel_optical_frame_id: left_camera_accel...
 * /left_camera/realsense2_camera/align_depth: True
 * /left_camera/realsense2_camera/aligned_depth_to_color_frame_id: left_camera_align...
 * /left_camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: left_camera_align...
 * /left_camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: left_camera_align...
 * /left_camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: left_camera_align...
 * /left_camera/realsense2_camera/aligned_depth_to_infra1_frame_id: left_camera_align...
 * /left_camera/realsense2_camera/aligned_depth_to_infra2_frame_id: left_camera_align...
 * /left_camera/realsense2_camera/allow_no_texture_points: False
 * /left_camera/realsense2_camera/base_frame_id: left_camera_link
 * /left_camera/realsense2_camera/calib_odom_file: 
 * /left_camera/realsense2_camera/clip_distance: -1.0
 * /left_camera/realsense2_camera/color_fps: 15
 * /left_camera/realsense2_camera/color_frame_id: left_camera_color...
 * /left_camera/realsense2_camera/color_height: 480
 * /left_camera/realsense2_camera/color_optical_frame_id: left_camera_color...
 * /left_camera/realsense2_camera/color_width: 640
 * /left_camera/realsense2_camera/depth_fps: 30
 * /left_camera/realsense2_camera/depth_frame_id: left_camera_depth...
 * /left_camera/realsense2_camera/depth_height: 480
 * /left_camera/realsense2_camera/depth_optical_frame_id: left_camera_depth...
 * /left_camera/realsense2_camera/depth_width: 640
 * /left_camera/realsense2_camera/device_type: 
 * /left_camera/realsense2_camera/enable_accel: True
 * /left_camera/realsense2_camera/enable_color: True
 * /left_camera/realsense2_camera/enable_depth: True
 * /left_camera/realsense2_camera/enable_fisheye1: True
 * /left_camera/realsense2_camera/enable_fisheye2: True
 * /left_camera/realsense2_camera/enable_fisheye: True
 * /left_camera/realsense2_camera/enable_gyro: True
 * /left_camera/realsense2_camera/enable_infra1: True
 * /left_camera/realsense2_camera/enable_infra2: True
 * /left_camera/realsense2_camera/enable_pointcloud: False
 * /left_camera/realsense2_camera/enable_sync: True
 * /left_camera/realsense2_camera/filters: 
 * /left_camera/realsense2_camera/fisheye1_frame_id: left_camera_fishe...
 * /left_camera/realsense2_camera/fisheye1_optical_frame_id: left_camera_fishe...
 * /left_camera/realsense2_camera/fisheye2_frame_id: left_camera_fishe...
 * /left_camera/realsense2_camera/fisheye2_optical_frame_id: left_camera_fishe...
 * /left_camera/realsense2_camera/fisheye_fps: 30
 * /left_camera/realsense2_camera/fisheye_frame_id: left_camera_fishe...
 * /left_camera/realsense2_camera/fisheye_height: 480
 * /left_camera/realsense2_camera/fisheye_optical_frame_id: left_camera_fishe...
 * /left_camera/realsense2_camera/fisheye_width: 640
 * /left_camera/realsense2_camera/gyro_fps: 400
 * /left_camera/realsense2_camera/gyro_frame_id: left_camera_gyro_...
 * /left_camera/realsense2_camera/gyro_optical_frame_id: left_camera_gyro_...
 * /left_camera/realsense2_camera/imu_optical_frame_id: left_camera_imu_o...
 * /left_camera/realsense2_camera/infra1_frame_id: left_camera_infra...
 * /left_camera/realsense2_camera/infra1_optical_frame_id: left_camera_infra...
 * /left_camera/realsense2_camera/infra2_frame_id: left_camera_infra...
 * /left_camera/realsense2_camera/infra2_optical_frame_id: left_camera_infra...
 * /left_camera/realsense2_camera/infra_fps: 30
 * /left_camera/realsense2_camera/infra_height: 480
 * /left_camera/realsense2_camera/infra_width: 640
 * /left_camera/realsense2_camera/initial_reset: False
 * /left_camera/realsense2_camera/json_file_path: 
 * /left_camera/realsense2_camera/linear_accel_cov: 0.01
 * /left_camera/realsense2_camera/odom_frame_id: left_camera_odom_...
 * /left_camera/realsense2_camera/pointcloud_texture_index: 0
 * /left_camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /left_camera/realsense2_camera/pose_frame_id: left_camera_pose_...
 * /left_camera/realsense2_camera/pose_optical_frame_id: left_camera_pose_...
 * /left_camera/realsense2_camera/publish_odom_tf: True
 * /left_camera/realsense2_camera/publish_tf: True
 * /left_camera/realsense2_camera/rosbag_filename: 
 * /left_camera/realsense2_camera/serial_no: 001622071918
 * /left_camera/realsense2_camera/tf_publish_rate: 0.0
 * /left_camera/realsense2_camera/topic_odom_in: left_camera/odom_in
 * /left_camera/realsense2_camera/unite_imu_method: none
 * /left_camera/realsense2_camera/usb_port_id: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /left_camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[left_camera/realsense2_camera_manager-1]: started with pid [18180]
process[left_camera/realsense2_camera-2]: started with pid [18181]
process[left_camera/color_rectify_color-3]: started with pid [18182]
process[left_camera/points_xyzrgb_hw_registered-4]: started with pid [18184]
[ INFO] [1628139974.514158996]: Initializing nodelet with 8 worker threads.
[ INFO] [1628139974.938156856]: RealSense ROS v2.2.13
[ INFO] [1628139974.938216440]: Running with LibRealSense v2.33.1
[ INFO] [1628139974.962490422]:  
[ INFO] [1628139974.976214648]: Device with serial number 008222070950 was found.

[ INFO] [1628139974.976247534]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.0/video4linux/video3 was found.
[ INFO] [1628139974.976257009]: Device with name Intel RealSense D435 was found.
[ INFO] [1628139974.976645520]: Device with port number 1-4.1 was found.
[ INFO] [1628139974.985209612]: Device with serial number 001622071918 was found.

[ INFO] [1628139974.985239781]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.4/1-4.4:1.0/video4linux/video6 was found.
[ INFO] [1628139974.985250120]: Device with name Intel RealSense D435 was found.
[ INFO] [1628139974.985563935]: Device with port number 1-4.4 was found.
[ INFO] [1628139974.990492942]: getParameters...
[ INFO] [1628139975.033630969]: setupDevice...
[ INFO] [1628139975.033674217]: JSON file is not provided
[ INFO] [1628139975.033688243]: ROS Node Namespace: left_camera
[ INFO] [1628139975.033749009]: Device Name: Intel RealSense D435
[ INFO] [1628139975.033761667]: Device Serial No: 001622071918
[ INFO] [1628139975.033792021]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.4/1-4.4:1.0/video4linux/video6
[ INFO] [1628139975.033827248]: Device FW version: 05.12.13.50
[ INFO] [1628139975.033841217]: Device Product ID: 0x0B07
[ INFO] [1628139975.033886651]: Enable PointCloud: Off
[ INFO] [1628139975.033899267]: Align Depth: On
[ INFO] [1628139975.033910881]: Sync Mode: On
[ INFO] [1628139975.034046962]: Device Sensors: 
[ INFO] [1628139975.034101079]: Stereo Module was found.
[ INFO] [1628139975.034119918]: RGB Camera was found.
[ INFO] [1628139975.034175900]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1628139975.034188572]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1628139975.034215015]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1628139975.034241362]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1628139975.034253492]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1628139975.034265483]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1628139975.034285773]: num_filters: 0
[ INFO] [1628139975.034297465]: Setting Dynamic reconfig parameters.
 05/08 13:06:15,055 WARNING [140292252456704] (types.cpp:49) hwmon command 0x7d failed. Error type: No data to return (-21).
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x7d failed. Error type: No data to return (-21).
[left_camera/realsense2_camera_manager-1] process has died [pid 18180, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/bulldog/.ros/log/9e567484-f5a7-11eb-b87f-1c697a64284c/left_camera-realsense2_camera_manager-1.log].
log file: /home/bulldog/.ros/log/9e567484-f5a7-11eb-b87f-1c697a64284c/left_camera-realsense2_camera_manager-1*.log
[left_camera/realsense2_camera-2] process has finished cleanly
log file: /home/bulldog/.ros/log/9e567484-f5a7-11eb-b87f-1c697a64284c/left_camera-realsense2_camera-2*.log
[left_camera/points_xyzrgb_hw_registered-4] process has finished cleanly
log file: /home/bulldog/.ros/log/9e567484-f5a7-11eb-b87f-1c697a64284c/left_camera-points_xyzrgb_hw_registered-4*.log

MartyG-RealSense, 2021-08-05:
Hi @github-lile  Your RealSense ROS wrapper and librealsense SDK versions are correctly matched.

As a starting point in investigating your case, may I ask if you installed **ddynamic_reconfigure** for ROS Kinetic during installation of the ROS wrapper?  If you have not installed it and you are using Linux then you should be able to do so with the command below:

**sudo apt install ros-kinetic-ddynamic-reconfigure**

github-lile, 2021-08-06:
> Hi @github-lile Your RealSense ROS wrapper and librealsense SDK versions are correctly matched.
>
> As a starting point in investigating your case, may I ask if you installed **ddynamic_reconfigure** [...]

github-lile, 2021-08-06:
@MartyG-RealSense Thank you, I should have installed ddynamic_reconfigure. I started three D435 cameras through roslaunch at the same time, only two of them can be started, but the left one cannot be started. The above is the logs of starting the camera alone.

MartyG-RealSense, 2021-08-06:
What computer or computing device do you have the three D435 cameras attached to please?  (PC, Raspberry Pi, Nvidia Jetson, etc)

github-lile, 2021-08-09:
@MartyG-RealSense PC，ubuntu20.04

MartyG-RealSense, 2021-08-09:
Thank you very much.  And which roslaunch instruction are you using to launch the cameras?  For example, are you using **rs_multiple_devices.launch** and specifying the serial numbers for the three cameras in a single roslaunch instruction?

https://github.com/IntelRealSense/realsense-ros#work-with-multiple-cameras

github-lile, 2021-08-10:
@MartyG-RealSense Thank you. Yes, I did exactly what you said.

MartyG-RealSense, 2021-08-10:
There was a past case in https://github.com/IntelRealSense/realsense-ros/issues/1326 of a multiple camera launch where only two cameras out of a set would be detected.  They were able to get it working on a PC by using a mains electricity powered USB 3.2 hub.  Is it practical to use a powered hub that is plugged into a mains socket in your particular project, please?

 You could alternatively test the launching of the three cameras in separate ROS terminals (one terminal for each camera) using the **rs_camera.launch** instruction instead of a single terminal using the **rs_multiple_devices.launch** instruction, as suggested by the multiple camera section of the RealSense ROS documentation linked to above.

github-lile, 2021-08-10:
@MartyG-RealSense Thank you very much. 
The left camera can be displayed by the system tools of the PC when plugged into the PC by USB, but it can't be started by ros.
I also used three rs_camera.launch with different serial numbers to launch different three cameras separately, but the left one does not work.

MartyG-RealSense, 2021-08-10:
Is the affected camera able to work normally in the **RealSense Viewer** program, please?

github-lile, 2021-08-11:
@MartyG-RealSense  Yes, the affected camera can work normally in the RealSense Viewer program.

MartyG-RealSense, 2021-08-11:
I'm not certain about this point, but I wonder whether 2 cameras on the same computer in the RealSense ROS wrapper is a purposeful limitation.  The section of the wrapper documentation about multiple camera use only provides guidance for defining two cameras.  And Intel's ROS multiple-camera guide for three cameras puts two on the same computer and the third on a second computer.

https://github.com/IntelRealSense/realsense-ros#work-with-multiple-cameras

closed 2021-08-23 by MartyG-RealSense