realsenseai/realsense-ros #2023: ROS2 with L515: message rates from other nodes are greatly reduced

AndyZe, 2021-08-05:
We have a relatively new, powerful desktop PC (Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz, 8 threads). When running the L515 we find that overall ROS messages are greatly reduced. For example, streaming position commands to the robot arm drops from 500 Hz to ~12 Hz!

I will try to provide as much information as I can below. **Any suggestions to improve performance?** (Likely related to #1850)

ROS2 Foxy
Only one RealSense L515 and one UR5e robot connected, no other significant devies
Camera config attached.
System monitor with CPU load and Network bandwidth snapshot attached. Note, all CPU loads are around 70-90%.

[realsense.launch.py.txt](https://github.com/IntelRealSense/realsense-ros/f
[image]
iles/6940956/realsense.launch.py.txt)

MartyG-RealSense, 2021-08-07:
Hi @AndyZe   Yes, the L515 lidar depth camera model only supports an FPS of 30 for depth and infrared streams.

I see that you are using the **rs_launch.py** launch file.  Can you tell me the full roslaunch instruction that you used, please?

Also, would it be possible to paste the launch log that was generated in the terminal during roslaunch?  Thanks!

AndyZe, 2021-08-08:
Hi Marty, the launch log stuff pertaining to RealSense is attached. Also attached the result of `ros2 param dump /wrist_l515/wrist_l515`  (our wrist camera node name).
[ros2_param_dump.txt](https://github.com/IntelRealSense/realsense-ros/files/6951321/ros2_param_dump.txt)
[L515_launch_log.txt](https://github.com/IntelRealSense/realsense-ros/files/6951322/L515_launch_log.txt)

MartyG-RealSense, 2021-08-09:
Do you see any improvement in performance if you launch moveit_studio_agent and realsense.launch.py in two separate ROS terminals instead of with a single launch instruction?

AndyZe, 2021-08-10:
It seems like this is mostly fixed by building in Release mode. Sorry to bother you. (Will re-open if it becomes an issue again.)

closed 2021-08-10 by AndyZe

MartyG-RealSense, 2021-08-10:
It's no trouble at all   :)   Please do feel free to re-open at a later date if you have follow-up questions.   Thanks for the update!

bbferka, 2021-11-22:
If anyone reads this thread because of high CPU consumption while using L515 cameras: we had a similar issue: extremely poor performance using ROS2 on same hardware where ROS1 drivers were performing well. Turned out to be a bug in here: https://github.com/ros-perception/image_transport_plugins/blob/foxy-devel/compressed_depth_image_transport/src/compressed_depth_publisher.cpp#L68  since setting the PNG compression level does not work as is currently implemented and will always default to 9;