realsenseai/realsense-ros #2025: some warnning when use d435i

SongJ11, 2021-08-07:
After I use d435i to execute realsense viewer, some warnings appear. Why？
such as:
WARNING [139766324201216] (backend-v4l2.cpp:1358) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 WARNING [139766105855744] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.5/0003:8086:0B3A.0003/HID-SENSOR-200076.3.auto/iio:device0/buffer/enable
WARNING [139765882025728](sensor.cpp:985) HID timestamp not found, switching to Host timestamps.
WARNING [139765202544384] (ds5-auto-calibration.cpp:340) hwmon command 0x80( 3 0 0 0 ) failed (response -7= HW not ready)

MartyG-RealSense, 2021-08-07:
Hi @SongJ11  May I first ask the questions below, please.

-  Are you using Windows or Linux, please?

-  If you are using Linux, did you build the librealsense SDK from Debian packages or from source code?

-  If you built the librealsense SDK from source code, did you patch the kernel using the command below?

**./scripts/patch-realsense-ubuntu-lts.sh**

SongJ11, 2021-08-07:
> Hi @SongJ11 May I first ask the questions below, please.
>
>     * Are you using Windows or Linux, please?
>
>     * If you are using Linux, did you build the librealsense SDK from Debian packages or from [...]

I use Linux. If I use. / scripts / patch-realsense-ubuntu-lts.sh, how should I solve it

SongJ11, 2021-08-07:
> I use Linux. If I use. / scripts / patch-realsense-ubuntu-lts.sh, how should I solve it
    I use the librealsense SDK

SongJ11, 2021-08-07:
> If you are using Linux, did you build the librealsense SDK from Debian packages or from source code

I use source code to build librealsense SDK

MartyG-RealSense, 2021-08-07:
The error **Pixel format ... a likely requires patch for fourcc code RW16!** may be caused by an unpatched kernel.  Please go to the root directory of your librealsense source code folder (the 'librealsense' one with the list of files) and input the command **./scripts/patch-realsense-ubuntu-lts.sh**

[image]

SongJ11, 2021-08-07:
> The error **Pixel format ... likely requires patch for fourcc code RW16!** may be caused by an unpatched kernel. Please go to the root directory of your librealsense source code folder (the 'librealsense' one with the list of files) and input the command **./scripts/patch-realsense-ubuntu-lts.sh**
> 
> [image]

OK,I will try it,thanks.

MartyG-RealSense, 2021-08-14:
Hi @SongJ11  Do you require further assistance with this case, please?  Thanks!

SongJ11, 2021-08-14:
> Hi @SongJ11 Do you require further assistance with this case, please? Thanks!

yes,I have solved those warnings using your method.Thanks you again.

MartyG-RealSense, 2021-08-14:
Excellent news that a solution has been achieved, @SongJ11 - thanks for the update.  As the problem has been solved, I will close the case.  Thanks again!

closed 2021-08-14 by MartyG-RealSense