realsenseai/realsense-ros #2026: Auto exposure goes too low and freezes

fiorano10, 2021-08-07:
We have 4 cameras connected to the motherboard, each has its own USB 3.1 Gen 1 port. Running the following config
Ubuntu 18.04
ROS Melodic
Librealsense v2.42
realsense-ros v2.2.22
Camera fw: 5.12.11
Auto exposure ROI: bottom 2/3 of the image

After launching the camera, it suddenly drops the exposure after a few minutes but never recovers, until the node is restarted.

dropped exposure in Orange
[image]

Before the exposure drop
[image]

After the exposure drop
[image]

MartyG-RealSense, 2021-08-08:
Hi @fiorano10  Which model of RealSense camera are you using, please?

There have been past cases of this effect of an exposure drop to minimum level that may require a camera reset or a covering over of the camera lenses with the hand to correct the image.  

https://github.com/IntelRealSense/librealsense/issues/4928
https://support.intelrealsense.com/hc/en-us/community/posts/360033445713-D435-Sunlight-saturation-problem

As a starting point in this investigation: since the effect occurs after a few minutes, would it be possible to touch the casing of the cameras when the problem occurs and see whether they are hot to the touch?  This could indicate that the problem is related to the camera being overheated due to bad USB cables or a glitch on the USB ports of the computer.  The computer is more likely as a cause, as all 4 cameras are unlikely to have bad USB cables.  There has also been a past case with a single-board computer where a power component on the board was causing the camera overheating.

fiorano10, 2021-08-09:
Wer're using D435i x 4, `/diagnostics_agg` says OK for all cameras. It is just one camera, and is fixed after blocking the view for a short period. Any way to do it without resetting the camera?

MartyG-RealSense, 2021-08-09:
You could try disabling and then re-enabling auto-exposure during runtime with a rosrun instruction.

https://github.com/IntelRealSense/realsense-ros/issues/1472#issuecomment-785887174

closed 2021-08-21 by MartyG-RealSense