realsenseai/realsense-ros #2028: (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61 IntelRealSense

SarenWraith, 2021-08-09:
I got to the same point referred to in: https://github.com/IntelRealSense/realsense-ros/issues/1510
except that the image is not being displayed in Rviz even when using the add tab

I am doing this on the jetson nano but have not attempted repeating it on a real pc yet. Could that resolve my problem?

MartyG-RealSense, 2021-08-09:
Hi @SarenWraith  Could you post the launch log in the terminal that is generated by the roslaunch instruction in a comment below, please?  Thanks!

SarenWraith, 2021-08-09:
I hope I am doing this right, I ran the command line 

`roslaunch realsense2_camera rs_camera.launch initial_reset:=true`

... logging to /home/cthulhu/.ros/log/4b79829e-f8f2-11eb-ac36-c006c3430634/roslaunch-jetson-8974.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.17.0.1:44321/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 84 lines cut ...]
[ INFO] [1628500521.444461552]: Setting Dynamic reconfig parameters.
[ WARN] [1628500521.896631522]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 165000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1628500521.911276834]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1628500522.044296508]: Done Setting Dynamic reconfig parameters.
[ INFO] [1628500522.045672062]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1628500522.048503428]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1628500522.048635666]: setupPublishers...
[ INFO] [1628500522.061476578]: Expected frequency for depth = 30.00000
[ INFO] [1628500522.150681985]: Expected frequency for color = 30.00000
[ INFO] [1628500522.214217547]: setupStreams...
[ INFO] [1628500522.227463193]: insert Depth to Stereo Module
[ INFO] [1628500522.227629701]: insert Color to RGB Camera
 09/08 13:15:22,317 WARNING [547490861440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1628500522.368077250]: SELECTED BASE:Depth, 0
[ INFO] [1628500522.413880562]: RealSense Node Is Up!
[ WARN] [1628500522.437925066]: 
 09/08 13:15:22,438 WARNING [547490861440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 13:15:22,489 WARNING [547490861440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 13:15:22,540 WARNING [547490861440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 09/08 13:15:22,783 WARNING [547490861440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 13:15:22,834 WARNING [547490861440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 13:15:22,885 WARNING [547490861440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 13:15:22,936 WARNING [547490861440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2021-08-09:
Yes, that is the correct log.  Thanks very much!

Your librealsense SDK and RealSense ROS wrapper versions are correctly matched, and depth and color streams are being published.

Are the **control_transfer returned** messages only appearing a small number of times during launch and then stopping, or at a rate of once or twice per minute after launch?  If so then they can be safely ignored.  However, if they are being generated continuously without stopping then this can indicate a serious communication problem with the device.

SarenWraith, 2021-08-09:
I am afraid they are being generated continuously without stopping. This is all being done on a jetson nano, which has a limited computing power. Is it possible the problem is due to the computing power? Or is there something more sinister going on? Also the command `realsense-viewer` works perfectly, and one launch file by the name of `rs_camera_with_model.launch` (I may have miswritten it, but its something close to this), worked once and then did not work a second time. Thank you for the assistance MartyG, it is much appreciated believe me.

MartyG-RealSense, 2021-08-09:
Jetson boards are very capable, so I would not strongly suspect processing limitations to be the problem.  You can though check CPU usage percentage in the Ubuntu OS whilst an application is running by using a system monitoring application such as **htop** though.

https://vitux.com/how-to-use-htop-to-monitor-system-processes-in-ubuntu-20-04/

However, a **power supply** related issue could potentially cause problems if there is power instability at the USB port.  Do you have your camera directly connected to the Nano board (so that the camera's power needs are being supplied by the Nano) or are you using a USB hub, please?  And if you are using a hub, is it a 'powered hub' that is powered by a mains electricity socket?

SarenWraith, 2021-08-09:
I have the camera connected directly to the Nano, but I have been powering the Jetson Nano from the usb port rather than the barrel jack, I will attempt a run with a barrel jack connector and update you.

SarenWraith, 2021-08-09:
The issue was not resolved but the error log looks a little different now:

... logging to /home/cthulhu/.ros/log/c29f2cfc-f8f7-11eb-933e-c006c3430634/roslaunch-jetson-7953.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.17.0.1:41137/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 95 lines cut ...]
[ INFO] [1628512062.890246023]: insert Color to RGB Camera
 09/08 16:27:42,981 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1628512063.032038887]: SELECTED BASE:Depth, 0
 09/08 16:27:43,052 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1628512063.073663939]: RealSense Node Is Up!
[ WARN] [1628512063.110185658]: 
 09/08 16:27:43,111 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 16:27:43,161 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 16:27:43,212 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 16:27:43,263 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 16:27:43,398 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 16:27:43,449 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 16:27:43,499 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 16:27:43,550 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 09/08 16:27:50,072 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 09/08 16:27:53,124 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 09/08 16:27:56,178 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 09/08 16:28:01,232 WARNING [546912059776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2021-08-09:
Is there any improvement if you set RViz to use **compressed** data instead of raw data, like in the link below?

https://github.com/IntelRealSense/realsense-ros/issues/1510#issuecomment-839616982

SarenWraith, 2021-08-09:
I just tried your suggested solution, I am afraid that the image still sends a No image received error. However in my case the fixed frame (under the global status tab) shows the error **Fixed Frame [map] does not exist**. The error summary remains the same, alas.

MartyG-RealSense, 2021-08-09:
Do you get the same warning messages if you leave **initial_reset:=true** out of the roslaunch instruction?

Do you also still get the Fixed Frame error if you set Fixed Frame to **camera_link** instead of map?

[image]

SarenWraith, 2021-08-10:
Sorry for the late reply, The good news is that the global status is showing a green check mark as well as the fixed frame for the system. There is no image showing, however, and the error log seems much the same

roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_fps:=6 color_fps:=6
... logging to /home/cthulhu/.ros/log/0d789d26-f9a2-11eb-a46d-c006c3430634/roslaunch-jetson-10954.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.17.0.1:42487/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 6
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 6
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 84 lines cut ...]
[ INFO] [1628576001.922057574]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1628576001.922171852]: setupPublishers...
[ INFO] [1628576001.931122513]: Expected frequency for depth = 30.00000
[ INFO] [1628576002.011709857]: Expected frequency for color = 30.00000
[ INFO] [1628576002.075643000]: setupStreams...
 10/08 10:13:22,076 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[ INFO] [1628576002.138172404]: insert Depth to Stereo Module
[ INFO] [1628576002.138359187]: insert Color to RGB Camera
[ INFO] [1628576002.253970485]: SELECTED BASE:Depth, 0
[ INFO] [1628576002.302115278]: RealSense Node Is Up!
[ WARN] [1628576002.384628261]: 
 10/08 10:13:22,386 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:13:22,436 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:13:22,488 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:13:22,538 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:13:22,589 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 10/08 10:13:23,267 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:13:23,317 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:13:23,771 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:13:37,133 WARNING [546794627456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

[image]

SarenWraith, 2021-08-10:
Interestingly enough, while the realsense viewer does work, it also outputs a similar  error log

 10/08 10:22:33,891 WARNING [548195520960] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 10/08 10:22:35,269 WARNING [548195520960] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 10/08 10:22:36,781 WARNING [548195520960] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 10/08 10:22:36,857 WARNING [548195520960] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 10/08 10:22:40,129 WARNING [548426729712] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 10/08 10:22:42,066 WARNING [548195520960] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:22:42,117 WARNING [548195520960] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:22:42,209 WARNING [548195520960] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:22:42,289 WARNING [548195520960] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2021-08-10:
The ROS wrapper is overriding your custom roslaunch configuration of 6 FPS and using 30 FPS for depth and color.  This is likely because since ROS wrapper version 2.2.22, the wrapper requires a custom stream definition to have three factors specified for each stream type used: width, height and FPS.  Otherwise, the configuration is deemed invalid and the default configuration of that particular camera model is applied instead.

Please try the roslaunch instruction below:

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=848 depth_height:=480 depth_fps:=6 color_width:=1280 color_height:=720 color_fps:=6**

SarenWraith, 2021-08-10:
Good news is I am finally seeing an image on RVIZ, but the error log eventually returns to the 768 index after a while. 

[image]

started roslaunch server http://172.17.0.1:42105/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 6
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 720
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1280
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 6
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 848
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
[... 125 lines cut ...]
 10/08 10:55:23,914 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:55:23,964 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:55:24,015 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:55:24,065 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:55:24,116 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:55:27,269 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:55:29,321 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:55:37,377 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:55:38,428 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:55:40,480 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:55:54,539 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 10/08 10:56:33,116 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:56:42,181 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:56:55,240 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 10:56:56,291 WARNING [547339891072] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

SarenWraith, 2021-08-10:
Ok so I decided to try `roslaunch realsense2_camera opensource_tracking.launch filters:=pointcloud depth_width:=848 depth_height:=480 depth_fps:=6 color_width:=1280 color_height:=720 color_fps:=6` and no caamera output was shown even when I tried the compressed image trick.

[ WARN] [1628578987.926859165]: Failed to meet update rate! Took 0.012039967000000000377
[ WARN] [1628578988.019059308]: Failed to meet update rate! Took 0.018950675000000000159
[ WARN] [1628578988.020832489]: Failed to meet update rate! Took 0.017387710000000000665
[ WARN] [1628578988.158730781]: Failed to meet update rate! Took 0.0090624040000000013334
[ WARN] [1628578988.285580444]: Failed to meet update rate! Took 0.0092549450000000005323
[ WARN] [1628578988.290726075]: Failed to meet update rate! Took 0.01107307600000000114
[ INFO] [1628578988.405249882]: Odom: quality=184, std dev=0.004497m|0.044810rad, update time=0.587390s
[ WARN] [1628578988.919908972]: Failed to meet update rate! Took 0.010196325000000000874
[ WARN] [1628578988.922253369]: Failed to meet update rate! Took 0.0092049399999999999972
[ WARN] [1628578988.968909053]: Failed to meet update rate! Took 0.0085029350000000013948
[ INFO] [1628578989.288900957]: Odom: quality=154, std dev=0.004413m|0.043116rad, update time=0.514359s
[ WARN] [1628578989.466632776]: Failed to meet update rate! Took 0.012891863000000001668
[ WARN] [1628578989.467138885]: Failed to meet update rate! Took 0.010098651000000000155
[ WARN] [1628578989.772252455]: Failed to meet update rate! Took 0.011809902000000000594
[ WARN] [1628578989.779374553]: Failed to meet update rate! Took 0.015601480000000000903
[ WARN] [1628578989.780751836]: Failed to meet update rate! Took 0.0082556230000000001279
[ WARN] [1628578989.804646960]: Failed to meet update rate! Took 0.025068650000000001349
[ WARN] [1628578989.808893186]: Failed to meet update rate! Took 0.020211830999999999547
[ INFO] [1628578990.001802980]: Odom: quality=194, std dev=0.004326m|0.038960rad, update time=0.532669s
[ WARN] [1628578990.119098207]: Failed to meet update rate! Took 0.0073326700000000003335
[ INFO] [1628578990.323704494]: RTAB-Map detection rate = 1.000000 Hz
[ INFO] [1628578990.326265303]: rtabmap: Deleted database "/home/cthulhu/.ros/rtabmap.db" (--delete_db_on_start or -d are set).
[ INFO] [1628578990.326529999]: rtabmap: Using database from "/home/cthulhu/.ros/rtabmap.db" (0 MB).
[ WARN] [1628578990.429046222]: Failed to meet update rate! Took 0.0072756120000000007453
[ WARN] [1628578990.548537350]: Failed to meet update rate! Took 0.006766805000000000736
[ INFO] [1628578990.733063479]: rtabmap: Database version = "0.20.9".
[ INFO] [1628578990.734092001]: rtabmap: SLAM mode (Mem/IncrementalMemory=true)
[ INFO] [1628578991.031075911]: Odom: quality=203, std dev=0.005517m|0.043116rad, update time=0.590095s
[ WARN] [1628578991.229421307]: Failed to meet update rate! Took 0.007652131000000000384
[ INFO] [1628578991.244340938]: /rtabmap/rtabmap: subscribe_depth = true
[ INFO] [1628578991.244611207]: /rtabmap/rtabmap: subscribe_rgb = true
[ INFO] [1628578991.245943436]: /rtabmap/rtabmap: subscribe_stereo = false
[ INFO] [1628578991.246057606]: /rtabmap/rtabmap: subscribe_rgbd = false (rgbd_cameras=1)
[ INFO] [1628578991.246131515]: /rtabmap/rtabmap: subscribe_odom_info = true
[ INFO] [1628578991.246205163]: /rtabmap/rtabmap: subscribe_user_data = false
[ INFO] [1628578991.246270113]: /rtabmap/rtabmap: subscribe_scan = false
[ INFO] [1628578991.246343396]: /rtabmap/rtabmap: subscribe_scan_cloud = false
[ INFO] [1628578991.246491526]: /rtabmap/rtabmap: subscribe_scan_descriptor = false
[ INFO] [1628578991.246591321]: /rtabmap/rtabmap: queue_size    = 10
[ INFO] [1628578991.246662260]: /rtabmap/rtabmap: approx_sync   = true
[ INFO] [1628578991.247006229]: Setup depth callback
[ WARN] [1628578991.261949247]: Failed to meet update rate! Took 0.0068362720000000000084
[ WARN] [1628578991.305205855]: Failed to meet update rate! Took 0.0067654360000000005396
[ INFO] [1628578991.624639085]: 
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info
[ INFO] [1628578991.716473716]: Odom: quality=206, std dev=0.003369m|0.036256rad, update time=0.594077s
[ WARN] [1628578991.905327670]: Failed to meet update rate! Took 0.010225854000000001234
[ WARN] [1628578991.906059828]: Failed to meet update rate! Took 0.0076460860000000007222
[ WARN] [1628578991.965405323]: Failed to meet update rate! Took 0.0093885740000000002248
[ WARN] [1628578991.967327883]: Failed to meet update rate! Took 0.0079966029999999996974
[ WARN] [1628578992.422802915]: Still waiting for data on topic /imu/data_raw...
 10/08 11:03:12,500 WARNING [546719125888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1628578992.585043266]: Odom: quality=175, std dev=0.006249m|0.043116rad, update time=0.560497s
 10/08 11:03:12,661 ERROR [546257756544] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
[ WARN] [1628578992.864563508]: Failed to meet update rate! Took 0.0118911720000000002
[ WARN] [1628578992.869621113]: Failed to meet update rate! Took 0.013607111000000001594
[ WARN] [1628578992.889037914]: Failed to meet update rate! Took 0.016724997000000001862
[... cut ...]
[ WARN] [1628579059.892876520]: Failed to meet update rate! Took 0.0080019740000000002395
 10/08 11:04:19,931 ERROR [546245669248] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 10/08 11:04:19,959 ERROR [546245669248] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
[ INFO] [1628579059.975802814]: Odom: quality=299, std dev=0.002989m|0.036256rad, update time=0.455006s
[ WARN] [1628579059.989814902]: Failed to meet update rate! Took 0.0082669279999999997766

MartyG-RealSense, 2021-08-10:
Possibly 6 FPS is not delivering frames quickly enough for RTABMAP.  Do you see improvement if you change 6 to 15 in the roslaunch instruction, please?

SarenWraith, 2021-08-10:
Ok so good news, the camera is showing a feed at a decent frame rate and the IMU is being processed as I can see it changing.

[image]

 The map is not forming however like in the 

Here is the Error log:

 10/08 11:14:31,428 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 10/08 11:14:31,526 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 10/08 11:14:31,943 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 10/08 11:14:32,052 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 10/08 11:14:33,247 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 10/08 11:14:35,184 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 10/08 11:14:36,070 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,222 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,292 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,374 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,484 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,536 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,587 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,638 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,688 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,741 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,791 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,842 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,893 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,945 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:36,996 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:37,216 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:37,368 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:37,821 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:38,090 WARNING [547901847792] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 10/08 11:14:42,246 ERROR [547272782272] (http-downloader.cpp:143) Download error from URL: https://librealsense.intel.com/Releases/rs_versions_db.json, error info: Timeout was reached
 10/08 11:14:46,077 WARNING [547667042752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/08 11:14:55,765 ERROR [546400367040] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 10/08 11:14:55,797 ERROR [546400367040] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
cthulhu@jetson:~$ roslaunch realsense2_camera opensource_tracking.launch filters:=pointcloud depth_width:=848 depth_height:=480 depth_fps:=15 color_width:=1280 color_height:=720 color_fps:=15
... logging to /home/cthulhu/.ros/log/b226122e-f9aa-11eb-b5ea-c006c3430634/roslaunch-jetson-8298.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.17.0.1:38497/

SUMMARY
========

CLEAR PARAMETERS
 * /ukf_se/

PARAMETERS
 * /ImuFilter/_publish_tf: False
 * /ImuFilter/_world_frame: enu
 * /ImuFilter/use_mag: False
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
[... cut ...] ffort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_plugin_tutorials/Imu
[ WARN] [1628579720.785188759]: Failed to meet update rate! Took 0.0069380130000000006865

MartyG-RealSense, 2021-08-10:
I note that in your latest launch log, the gyro and accel topics are disabled.  They are disabled by default from ROS wrapper **2.2.15** onwards.  to enable them, add the instructions below to your roslaunch instruction:

**enable_gyro:=true enable_accel:=true**

SarenWraith, 2021-08-10:
I tried your command and when I did it the camera no longer functioned. I am nit sure why this happened. Hoever, the camera did track itself rotating 

[image]

MartyG-RealSense, 2021-08-10:
If you wish to use the **opensource_tracking** launch file to take advantage of the IMU of the D435i then Intel's D435i SLAM tutorial may be a good guide to use if you have not done so already (remembering to enable gyro and accel in the launch instruction, as the guide does not include that information)

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

SarenWraith, 2021-08-10:
I actually started with this guide before all of this, I followed the steps to install the imu-madgwick, the robot localization, and the rtab-map. There are improvements now that I have received your help, but the picture shown in the guide is still far off from what I currently have.

MartyG-RealSense, 2021-08-10:
If your setup is based on the SLAM D435i guide then may I confirm the following points with you:

-  That you launch with **roslaunch realsense2_camera opensource_tracking.launch** and do not include the pointcloud filter.  Because as the D435i SLAM guide shows, a pointcloud is already generated when using **opensource_tracking.launch** in combination with the other installed pre-requisite modules of the guide.

-  That the Fixed Frame option in RViz is set to **map** as shown in the small animated image when magnified.

[image]

-  In the Frames list of tick-boxes in RViz, only the 'camera_link' and 'map' boxes are ticked, and 'Image Topic' just below the Frames list is set to 'camera/color/image/raw'

[image]

SarenWraith, 2021-08-11:
Sorry for the late reply. Alright so I am receiving an image on the image pad on the RVIZ, obviously the IMU is no longer functional as it was checked off. My question is how to obtain the map shown [intel SLAM guide](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i)? 
I have attached a picture of the operation and the error log in case that helps

[image]

000011981
[ WARN] [1628663958.891571937]: Failed to meet update rate! Took 0.0079370100000000012863
[ WARN] [1628663958.903848784]: Failed to meet update rate! Took 0.0068417220000000006017
[ WARN] [1628663958.947588900]: Failed to meet update rate! Took 0.0069947050000000003667
[ WARN] [1628663958.951202359]: Failed to meet update rate! Took 0.0075658750000000005512
[ WARN] [1628663959.197977588]: Failed to meet update rate! Took 0.007669566000000000057
[ WARN] (2021-08-11 10:39:19.426) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=50) between -1 and 875"
[ INFO] [1628663959.429575222]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.525421s
[ WARN] [1628663959.648755332]: Failed to meet update rate! Took 0.0084504280000000011469
[ WARN] [1628663959.803875200]: Failed to meet update rate! Took 0.0068982280000000001721
[ WARN] [1628663959.807149075]: Failed to meet update rate! Took 0.0067690820000000000503
[ WARN] (2021-08-11 10:39:20.193) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=45) between -1 and 876"
[ INFO] [1628663960.197076493]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.522544s
[ERROR] (2021-08-11 10:39:20.203) Rtabmap.cpp:1156::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 10072 is ignored!
[ INFO] [1628663960.203388251]: rtabmap (44): Rate=1.00s, Limit=0.000s, Conversion=0.0049s, RTAB-Map=0.0002s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[ WARN] (2021-08-11 10:39:20.763) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=40) between -1 and 877"
[ INFO] [1628663960.766380754]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.493513s
[ WARN] [1628663960.973740946]: Failed to meet update rate! Took 0.0067639350000000004332
[ WARN] [1628663961.103637064]: Failed to meet update rate! Took 0.0066791810000000001804
[ WARN] (2021-08-11 10:39:21.356) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=39) between -1 and 878"
[ INFO] [1628663961.359352088]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.513053s
[ERROR] (2021-08-11 10:39:21.364) Rtabmap.cpp:1156::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 10092 is ignored!
[ INFO] [1628663961.365012644]: rtabmap (44): Rate=1.00s, Limit=0.000s, Conversion=0.0048s, RTAB-Map=0.0002s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[ WARN] [1628663961.365814472]: Failed to meet update rate! Took 0.0088584370000000002554
[ WARN] [1628663961.414416284]: Failed to meet update rate! Took 0.0074207750000000001142
[ WARN] [1628663961.517282547]: Failed to meet update rate! Took 0.0069364230000000005882
[ WARN] [1628663961.528292407]: Failed to meet update rate! Took 0.01462920000000000019
[ WARN] [1628663961.528958088]: Failed to meet update rate! Took 0.012004570000000000865
[ WARN] (2021-08-11 10:39:21.913) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=37) between -1 and 879"
[ INFO] [1628663961.917171694]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.471766s
[ WARN] [1628663962.016469966]: Failed to meet update rate! Took 0.0079516159999999998925
[ WARN] [1628663962.168968204]: Failed to meet update rate! Took 0.0071066920000000003907
[ WARN] (2021-08-11 10:39:22.508) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=40) between -1 and 880"
[ INFO] [1628663962.511446109]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.496347s
[ERROR] (2021-08-11 10:39:22.518) Rtabmap.cpp:1156::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 10112 is ignored!
[ INFO] [1628663962.525372029]: rtabmap (44): Rate=1.00s, Limit=0.000s, Conversion=0.0063s, RTAB-Map=0.0002s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[ WARN] [1628663962.606094925]: Failed to meet update rate! Took 0.0075732490000000006469
[ WARN] [1628663962.628910692]: Failed to meet update rate! Took 0.0070434970000000001258
[ WARN] (2021-08-11 10:39:23.090) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=52) between -1 and 881"
[ INFO] [1628663963.094153526]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.505922s
[ WARN] [1628663963.165822564]: Failed to meet update rate! Took 0.0073076630000000006998
[ WARN] [1628663963.299838239]: Failed to meet update rate! Took 0.0077576720000000001787
[ WARN] [1628663963.317397410]: Failed to meet update rate! Took 0.0088749720000000000691
[ WARN] [1628663963.440697550]: Still waiting for data on topic /imu/data_raw...
[ WARN] [1628663963.712011525]: Failed to meet update rate! Took 0.0068428760000000003402
[ WARN] (2021-08-11 10:39:23.735) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=47) between -1 and 882"
[... cut ...]
[ INFO] [1628663980.954123372]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.520496s
[ WARN] [1628663981.081284265]: Failed to meet update rate! Took 0.010203198000000000129
[ WARN] [1628663981.083138391]: Failed to meet update rate! Took 0.0088097729999999999095
^C[ WARN] [1628663981.182717811]: Failed to meet update rate! Took 0.0072229770000000002167
[ukf_se-9] killing on exit
[ WARN] [1628663981.267634067]: Failed to meet update rate! Took 0.0088008040000000006126

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

MartyG-RealSense, 2021-08-11:
There is little documentation available about the **Failed to meet update rate! ** error.  When ROS does generate this error though, it sometimes provides this advice: "Try decreasing the rate, limiting sensor output frequency, or limiting the number of sensors".

I believe that 'sensor output frequency' is referring to the IMU frequency.  Like with setting an FPS for the image streams, RealSense ROS supports setting a 'frequency' speed for the gyro and accel streams in the roslaunch instruction.  The minimum supported frequencies are 200 for gyro and 63 for accel.

**gyro_fps:=200 accel_fps:=63**

[image]

Advice about the **Not enough inliers** warning is provided by Doronhi the RealSense ROS wrapper developer in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1080

Doronhi advises adjusting the position of the camera to see whether the inlier warning goes away.

In regard to **Out of frame resources!** and the **Exception was thrown during user processing callback** warnings, Doronhi provided the advice below in https://github.com/IntelRealSense/realsense-ros/issues/1161

"The "Out of frame resources!" message and the "Exception...processing callback!" could be related to lack of system resources - the app can't keep up with the incoming flow of images.  For that you can reduce frame rate and resolution.  Building with CUDA support should really help in that case on NVidia machines".

SarenWraith, 2021-08-13:
Apologies but I won’t be able to respond as my office closed for the time being. I’ll respond again in two days

MartyG-RealSense, 2021-08-13:
No problem at all, @SarenWraith - thanks for the update!

SarenWraith, 2021-08-16:
It worked!! For the record what I did was change the depth cloud and the color image under it to image_raw.

[image]

Alright but is there a way for the camera to retain that the wall it mapped is at this postion, because as I move it it is constantly forgetting the area it mapped.

MartyG-RealSense, 2021-08-16:
The link below contains a guide based around Intel's D435i SLAM tutorial for ROS that was created by a RealSense ROS user and expands upon the original guide.

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

In the section of the guide headed **Localization with rtabmap**, it suggests using RTABMAP to study a recorded pointcloud map file and obtain the relative position from that map.

Giggus, 2021-08-17:
Hi, 

I also have the same issue: "WARNING [139828389717760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61", tried to lower the color_fps and to set init_reset to true but I still get the message (even without using rviz). Did you find any solution to this?

MartyG-RealSense, 2021-08-23:
Hi @Giggus  Do you require further assistance with the above problem, please?  Thanks!

Giggus, 2021-08-23:
Hi @MartyG-RealSense, yes I do.
This problem is still very present and it is not solved by using solutions already posted on this forum unfortunately.

MartyG-RealSense, 2021-08-23:
Hi @Giggus  What computer / computing device are you using with the camera?  Could you also post a roslaunch log in a comment please.  Thanks!

Giggus, 2021-08-25:
Hi @MartyG-RealSense, 

I run the L515 on a MSI-GF75-Thin-10UEK (gaming laptop)
- ntel® Core™ i7-10750H CPU @ 2.60GHz × 12
- NVIDIA® GeForce RTX™ 3060
- 16 GB of RAM
-  Ubuntu 20.04.2 LTS

`$ roslaunch realsense2_camera rs_camera.launch`                                                                                                   

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gc-GF75-Thin-10UEK:38937/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
[... 195 lines cut ...]
 25/08 09:06:29,785 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:31,036 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:32,287 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:33,537 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:34,788 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:36,039 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:37,289 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:38,540 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:39,790 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:41,022 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:41,072 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:42,323 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:43,573 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 25/08 09:06:48,576 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/08 09:06:49,827 WARNING [140524029888256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Giggus, 2021-08-25:
No, always the same

Giggus, 2021-08-30:
@MartyG-RealSense yes

MartyG-RealSense, 2021-08-30:
@Giggus   It sounds as though the L515 camera hardware is okay, since it is working normally in the RealSense Viewer, and that it is a software problem related to your RealSense ROS setup.

Are you using any form of ROS networking (connecting another computer remotely to the gaming laptop that your L515 is attached to)

Also, your roslaunch log has the message **frame's time domain is HARDWARE_CLOCK.  Timestamps may reset periodically**.  I have seen this message occurring in very few cases, and it suggests to me that **Global Time** may be disabled in your launch.  Do you therefore see a positive difference if you include the instruction **enable_global_time:=true** in your roslaunch instruction to enable global time?

Giggus, 2021-08-31:
@MartyG-RealSense I am connecting to a remote robot but this error happens when I just launch the camera node on itself. 

I tried to disable " Global Time" but in any launch file the "global_time_enabled:=true" parameter actually exists... 
How can I change the time setting?

MartyG-RealSense, 2021-08-31:
If a launch file does not contain global_time_enabled then it can be added as a rosparam in the launch file for use with L515 using details provided by **Doronhi** the RealSense ROS wrapper developer in https://github.com/IntelRealSense/realsense-ros/issues/1607#issuecomment-760128960

MartyG-RealSense, 2021-09-06:
Hi @Giggus  Do you require further assistance with this case, please? Thanks!

Giggus, 2021-09-06:
Hi @MartyG-RealSense, even after adding 

  <rosparam>
      /camera/motion_module/global_time_enabled: true
      /camera/l500_depth_sensor/global_time_enabled: true
      /camera/rgb_camera/global_time_enabled: true
  </rosparam>

I still get the same warning 
"WARNING [140065227831040] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61"

MartyG-RealSense, 2021-09-06:
Is the launch still exiting or can you now view the data even though the control_transfer returned error is still occurring?

Giggus, 2021-09-06:
the launch never exits (unless I stop it myself) but I always have this warning message

MartyG-RealSense, 2021-09-06:
If the data is published correctly in RViz and does not seem to be affected by the warning messages then it is likely safe to ignore them.

Giggus, 2021-09-07:
@MartyG-RealSense ok thanks. I thought it was related with the low performance I have

Giggus, 2021-09-13:
No thanks, I'll keep the warning message

MartyG-RealSense, 2021-09-13:
Thanks very much @Giggus for the update.  As you do not require further assistance and @SarenWraith the creator of the case had their problem resolved earlier in https://github.com/IntelRealSense/realsense-ros/issues/2028#issuecomment-899283012 I will close the case.  Thanks again!

closed 2021-09-13 by MartyG-RealSense