realsenseai/realsense-ros #2030: Localization in pre-existing map only with d435i

AlaricLcrs, 2021-08-09:
Hello ! I'm currently working on an A1 Robot created by Unitree (https://www.unitree.com/products/a1/). I would like the robot to achieve autonomous navigation. I have already managed to create maps using the SLAM part of realsense-ros (https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i), and it works great. But in order to do that, I'm wondering if it's possible for the robot to locate himself in a preexisting map. I've looked where I could, but I don't know if it's possible with JUST a RealSense d435i. A lot of the solutions I've found would include a T265 camera or a LIDAR camera, which I don't have. 

Thanks !

MartyG-RealSense, 2021-08-09:
Hi @AlaricLcrs  A RealSense ROS user created an adaption of the SLAM with D435i guide to add localization with a saved map and trajectory plotting with **matplotlib.**

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM.html

AlaricLcrs, 2021-08-10:
That's exactly what I was looking for, thank you very much !

closed 2021-08-10 by AlaricLcrs

MartyG-RealSense, 2021-08-10:
You are very welcome @AlaricLcrs - I'm pleased that I could help   :)