realsenseai/realsense-ros #2033: [ROS2 foxy] D415 parameters do not have an effect

AndyZe, 2021-08-10:
I'm launching a D415, following the example in `rs_camera_multi_launch.py`. For now, it's just one camera but I want to run 2 soon.

My launch file is literally copy/paste of `rs_camera_multi_launch.py`, which calls `rs_launch.py`. The only thing I've modified is `depth_fps=6` and `color_fps=6` and commented launching of the second camera.

When launching the camera, terminal output says 30 fps:

`color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8`

ros2 param dump says 6 fps like it should:

`color_fps: 6.0`

`ros2 topic hz -w 100 /camera/camera/image_raw` says ~15 fps:

`average rate: 14.892`

AndyZe, 2021-08-10:
Can anybody confirm that parameters are set correctly from `rs_camera_multi_launch.py`?

MartyG-RealSense, 2021-08-10:
Hi @AndyZe  From ROS wrapper 2.2.15 onwards, three factors should be provided for each stream type being used when defining a custom stream configuration - width, height and FPS.  If these three factors are not provided then the ROS wrapper deems the custom configuration invalid and instead applies the default configuration of the particular camera model being used.

**Doronhi** the RealSense ROS wrapper provides an example custom configuration for ROS2 in https://github.com/IntelRealSense/realsense-ros/issues/1751#issuecomment-798834284

AndyZe, 2021-08-10:
Thanks for a very fast response @MartyG-RealSense! Can confirm that it works when all 3 parameters are set

AndyZe, 2021-08-10:
Oh, let me suggest printing a warning to terminal if the custom config file is rejected.

closed 2021-08-10 by AndyZe

MartyG-RealSense, 2021-08-10:
You're very welcome, @AndyZe - I'm pleased that I could help!

I will highlight your above feature request to Doronhi

MartyG-RealSense, 2021-08-12:
Hi @AndyZe  Doronhi has created a pull request (above) to address your feature request.  Thanks!