realsenseai/realsense-ros #2038: Issue with device node in linux

freesolo92, 2021-08-16:
Hi,

it is not a issue with the wrapper or the package itselfs but maybe with the kernel driver.
Every time I start the `realsense2_camera rs_camera` launchfile the device node in `/dev/video0` or `/dev/video1` dissapears and I do not know why.
Can anyone tell me why this is happen?
I have another package for object detection which accesses to the device node in `/dev/video2` and it is crashing all the time.

Thanks for any reply!

MartyG-RealSense, 2021-08-16:
Hi @freesolo92  The case https://github.com/IntelRealSense/realsense-ros/issues/1620 involved all of the /dev/video channels disappearing when performing roslaunch.  The RealSense ROS user in that case was using **Nvidia DeepStream**.  Are you using DeepStream for your object detection, please?

freesolo92, 2021-08-16:
Hi @MartyG-RealSense 
No I do not use DeepStream SDK, at least I don't use it consciously.
I have not installed the SDK.
The Python script from Nvidia only make use of the v4l2-framework for USB cams

But what I noticed is, that when I start `realsense-viewer`, the device node /dev/video0 is still available. Other than described in the other case posted.

Is this maybe a problem with the two ways how to build the LibRealSense SDK Backend?

I use a Jetson Nano with ROS Melodic on top of a Ubuntu 18.04.

MartyG-RealSense, 2021-08-17:
If you are building librealsense from source code then you can build it with the **V4L2 backend** by including the CMake build flag **-DFORCE_LIBUVC=OFF** 

The V4L2 backend enables standard Linux tools to make use of raw data from RealSense cameras.  

https://github.com/IntelRealSense/librealsense/issues/6841#issuecomment-660859774

freesolo92, 2021-08-17:
Yes, I have build the SDK from source and all packages working in seperate. 
I also have access via v4l2 framework tool.

But the point is, that the device node in `/dev/video0` till `/dev/video2` disappears all the time I start the realsense launch file.
In the case you mentioned above, they said that the device node will also disappear when starting the `realsense-viewer`. But I can not agree with this statement. 
When I work with `realsense viewer`, the device node is available. 

So, I do not know the resone why the device node disappears inside ros. But for my development I need both. After reconnecting the camera I am able to get access to the video stream via device node.

MartyG-RealSense, 2021-08-17:
In general, it is inadvisable for the RealSense Viewer and the RealSense ROS wrapper to be used together at the same time.  If you activate particular streams in one tool then they will likely become inaccessible in another tool if it tries to enable the same stream type.  This is because an enabled stream is 'claimed' by the program that first started it and it is not available to another program until the claim is released by stopping the stream in that first program.  

This principle is described in the SDK's **Multi-Streaming Model** documentatio in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

freesolo92, 2021-08-18:
Hi @MartyG-RealSense ,

Yes I know. I do not need a access from different tools at the same time. 
But after I used the ROS wrapper I need access via v4l2 tool and the device node. But after using ROS wrapper, the device node is not available. 
This is my problem. But since no one else had such a problem before, I guess I have to find my own solution. 
Unfortunately I don't currently have time to take a closer look at the ROS wrapper, how the backend works.

Thank you for the reference with the document!

MartyG-RealSense, 2021-08-18:
I will refer your question above to **Doronhi** the RealSense ROS wrapper developer to seek his input.  Thanks for your patience!

freesolo92, 2021-08-20:
Hi @MartyG-RealSense ,

Thank you very much for your support!

doronhi, 2021-08-30:
It sounds like you have 2 installations of librealsense2 on your machine. The one the ROS wrapper use, is using the RSUSB backend. It is a known issue that an app using RSUSB does not release it properly and the device is not available for 4vl backend. To Disable in the build you should use the -DFORCE_RSUSB_BACKEND=OFF option.
In any case, it is not a good idea to have two versions.
It may help if you poset the ouput of `apt list --installed | grep realsense` to help figure out which librealsense installations you have.
For instance, if you have both "ros-galactic-librealsense2" and "librealsense2-tools" it could mean that realsense-viewer uses a different version of librealsense2 than realsense2_camera.
If that is the case and realsense-viewer works good I suggest removing realsense2_camera installation and it's ros-librealsense2 dependencies and build the wrapper from code using the librealsense2 from librealsense2-dev installation.

freesolo92, 2021-08-31:
Hi @doronhi

Thanks for your message. You are right. I tried the 2 methods on my jetson. But after I flashed the Nano with a new image and only installed the realsense sdk via the package manager and installed the realsense ros from source in my workspace, it worked. 
I have access to the device driver via v4l2 tool after running the launch file. And I also get no erros while the launch file is running. 
Thank you very much!!! I will close this topic. 
Have a nice day

closed 2021-08-31 by freesolo92

MartyG-RealSense, 2021-08-31:
Hi @freesolo92   It's great to hear that you achieved success - thanks very much for the update!