realsenseai/realsense-ros #2039: Very large delay in ROS image topic between machines in wlan

RamsesReyes, 2021-08-18:
I'm working with a scaled model car which has a realsense SR300 camera, this is the car: [https://github.com/AutoModelCar](https://github.com/AutoModelCar). It has an odroid XU4 with Ubuntu 16.04 running ROS kinetic.

The car sends the camera image to a ROS topic by a realsense nodelet. The ROS_MASTER  is in a laptop, also with Ubuntu 16.04 running ROS kinetic,  in a wlan in which the car is too. The problem is that there are tons of delay in the topic that publishes the camera images:

    :~$ > rostopic delay /app/camera/rgb/image_raw/compressed
    subscribed to [/app/camera/rgb/image_raw/compressed]
    average delay: 1359.301
             min: 1358.945s max: 1359.655s std dev: 0.23105s window: 9

And the delay grows and grows even more. This is the ping from the laptop to the car:

    :~$ > ping 192.168.1.103
    PING 192.168.1.103 (192.168.1.103) 56(84) bytes of data.
    64 bytes from 192.168.1.103: icmp_seq=1 ttl=64 time=2.44 ms
    64 bytes from 192.168.1.103: icmp_seq=2 ttl=64 time=2.45 ms
    64 bytes from 192.168.1.103: icmp_seq=3 ttl=64 time=1.98 ms

This is the bandwidth consumption of the topic:

    :~& > rostopic bw /app/camera/rgb/image_color/compressed
    subscribed to [/app/camera/rgb/image_color/compressed]
    average: 365.30KB/s
    	   mean: 35.36KB min: 35.29KB max: 35.43KB window: 8

The clock time of the machines is sync with ntp. The delay is similar for the non-compressed image topics. The camera was initially taking images at 30Hz, then I set it at 10Hz but the problem persists. It is worth to mention that the delay is almost zero at starting the realsense nodelet, but it grows along the time.

I would really appreciate if you can tell me what I'm missing.
Thanks in advance, kind regards.

MartyG-RealSense, 2021-08-19:
Hi @RamsesReyes   May I first ask if you have tried the instruction **sudo ufw allow ntp** to allow the ntp protocol to work with Ubuntu if your system has a firewall?

RamsesReyes, 2021-08-19:
Hello @MartyG-RealSense , thank you for your time.

> Hi @RamsesReyes May I first ask if you have tried the instruction **sudo ufw allow ntp** to allow the ntp protocol to work with Ubuntu if your system has a firewall?

No, I haven't. Should I try it in the master or in the car, or maybe both?

MartyG-RealSense, 2021-08-19:
You could first check if there is a UFW firewall enabled on each Ubuntu machine with the instruction **sudo ufw status**

If the result come back as Active (indicating an enabled firewall on that particular machine) then you could proceed with trying **sudo ufw allow ntp **

https://www.digitalocean.com/community/tutorials/ufw-essentials-common-firewall-rules-and-commands

RamsesReyes, 2021-08-20:
Firewall is inactive in both machines.
```
root@Alan-3:~# ufw status
WARN: uid is 0 but '/' is owned by 252739
Status: inactive```

MartyG-RealSense, 2021-08-20:
I went back to the beginning of the case to research it again for fresh insights.  As this car is based on Period XU4, it occurs to me that this could be a potential source of problems in regard to the librealsense part of the project.  This is because Intel provide special Odroid-specific adaptations to the standard installation instructions for it in the link below.

https://dev.intelrealsense.com/docs/odroid

A RealSense user provides their own librealsense installation guide for XU4 here:

https://jack-kawell.com/2019/09/30/installing-realsense-sdk-on-an-odroid-xu4/