realsenseai/realsense-ros #2040: Intel said it’s shutting down

ranjitkathiriya, 2021-08-19:
Intel said it’s shutting down its RealSense computer vision division to focus on its core businesses, marking the end for a portfolio of high-tech cameras and sensors.

https://www.crn.com/news/components-peripherals/intel-says-it-s-winding-down-realsense-camera-business

Is that true? Can you please, provide in detail that how it will affect software support for the person who has brought real sense?

RealSenseSupport, 2021-08-21:
Intel has decided to wind down the RealSense business and is announcing the EOL of LiDAR, Facial Authentication and Tracking product lines this month. Intel will continue to provide select Stereo products to its current distribution customers.  
We will provide a 6-month EOL and Last Time Buy period for these EOL products.

jan-krueger, 2021-08-22:
@RealSenseSupport Will you continue to maintain this repository, and/or will the known issue regarding launching multiple T265s be fixed before they are discontinued entirely?

closed 2021-08-23 by ranjitkathiriya