realsenseai/realsense-ros #2042: Cannot read launch params for D455(eg: stereo_module/auto_exposure_limit)

vzhr, 2021-08-20:
when I use D455, I had a warnning: " _Param '/camera/realsense2_camera/stereo_module/auto_exposure_limit' has value 0 outside the range [1, 165000]. Removing this parameter from dynamic reconfigure options._".
Then I add  param "_stereo_module/auto_exposure_limit_" in _nodelet.launch.xml_ file,  I still had this warnning, but have this param: /camera/stereo_module/auto_exposure_limit.
I think this was caused by ros namespace, so I check codes and find this in _base_realsense_node.cpp_:

```
void BaseRealSenseNode::publishTopics()
{
    getParameters();
    setupDevice();
    setupFilters();
    registerHDRoptions();
    registerDynamicReconfigCb(_node_handle); //here
    setupErrorCallback();
    enable_devices();
    setupPublishers();
    setupStreams();
    SetBaseStream();
    registerAutoExposureROIOptions(_node_handle); //here
    publishStaticTransforms();
    publishIntrinsics();
    startMonitoring();
    ROS_INFO_STREAM("RealSense Node Is Up!");
}
```
Then I change `_node_handle` to `_pnh`
```
void BaseRealSenseNode::publishTopics()
{
    getParameters();
    setupDevice();
    setupFilters();
    registerHDRoptions();
    registerDynamicReconfigCb(_pnh); //here
    setupErrorCallback();
    enable_devices();
    setupPublishers();
    setupStreams();
    SetBaseStream();
    registerAutoExposureROIOptions(_pnh); //here
    publishStaticTransforms();
    publishIntrinsics();
    startMonitoring();
    ROS_INFO_STREAM("RealSense Node Is Up!");
}
```

stereo_module param loaded correctly.

Is this a bug? Maybe this help.

MartyG-RealSense, 2021-08-20:
Hi @vzhr  This warning appears in most RealSense ROS users' launch logs and therefore is not something to be concerned about and I would not consider it to be a bug.

When the feature was added in librealsense SDK version **2.42.0**, the SDK release notes provided the following description about it.

***********

Auto-Exposure and Gain max limits for IR sensors.  The new APIs set the effective upper bound for Exposure and Gain values for Depth Auto-Exposure.  This allows to control and limit the changes in FPS rate while enabling for AE auto-adjustments.  Requires FW5.12.11.0 or later.

**********

Thanks very much for sharing your method for enabling it in ROS with the RealSense ROS community!

vzhr, 2021-08-23:
> Hi @vzhr This warning appears in most RealSense ROS users' launch logs and therefore is not something to be concerned about and I would not consider it to be a bug.
>
> When the feature was added in [...]
You're very welcome,  @MartyG-RealSense.  I'm pleased that this could help!

MartyG-RealSense, 2021-08-23:
@vzhr  Do you require further assistance with this case after kindly providing your solution?   Thanks!

closed 2021-08-29 by MartyG-RealSense