realsenseai/realsense-ros #2043: Trying to launch the camera with a default yaml file

rythm07, 2021-08-20:
Hello,

Thanks in advance for your help.

I'm trying to launch my camera d435i with a default configuration that I wrote in the **d435i.yam**l in the config folder. On **rs_launch.py** file, there's a parameter called **config file**. I tried to put the path to the file as you do in the example that's on ReadMe but I wasn't successful.

Anyone knows the right way to do it?

MartyG-RealSense, 2021-08-20:
Hi @rythm07  If you are using the ROS2 wrapper - as indicated by your use of rs_launch.py - then a good starting point may be to use RealSense ROS2 wrapper version **3.2.2** and its matching librealsense version **2.48.0** if possible for you to do so, as this wrapper contains a fix for reading a .yaml config file.

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.2.2

Further references about using .yaml can be found in the links below.

https://github.com/IntelRealSense/librealsense/issues/8863
https://github.com/IntelRealSense/realsense-ros/issues/1640

rythm07, 2021-08-20:
I still don't know how can I read a yaml config file from the launch. I already tried the fixes described in this issue https://github.com/IntelRealSense/librealsense/issues/8863#issuecomment-823026235 but without success. I still have this warning appearing: 
[image]

Here is how I do it in the launch file:

[image]

MartyG-RealSense, 2021-08-20:
I have examined the details of your case carefully but cannot see or find anything obviously wrong.  So I will refer it to **Doronhi** the RealSense ROS wrapper developer to seek his advice.  Thanks for your patience!

MartyG-RealSense, 2021-08-20:
BTW, were you able to try the suggestion in https://github.com/IntelRealSense/realsense-ros/issues/2043#issuecomment-902637256 to upgrade to at least ROS2 wrapper 3.2.2 and librealsense 2.48.0 in order to have a wrapper containing the .yaml config file fix?

rythm07, 2021-08-20:
Thanks a lot for your help. I really appreciate that! I will wait then for @doronhi's help.

Yeah, I build the version 3.2.2 of the ROS2 Wrapper as you recommended, but I have my librealsense version is 2.49.0 instead of the version 2.48.0. 

Is there any problem with that?

MartyG-RealSense, 2021-08-20:
You are very welcome @rythm07   :)

Ideally a specific ROS wrapper version should be matched with the librealsense version listed in the release notes for that particular wrapper.  A ROS2 wrapper for 2.49.0 will likely be released soon (Doronhi develops the wrappers) so it may be worth waiting for the next ROS2 wrapper and then building it on the librealsense 2.49.0 tat you have already installed to avoid having to build librealsense again.

This is because when building the wrapper from source code, librealsense should be installed first and the wrapper secondly, and the wrapper should be built again each time that the librealsense version is updated.

rythm07, 2021-08-20:
Ok, so for now maybe I should uninstall the 2.49 version of librealsense and install the 2.48. Where can I find older versions of the sdk? Can you provide me a link for the 2.48 version?

MartyG-RealSense, 2021-08-20:
The source code folder for librealsense 2.48.0 can be obtained from the link below.  Scroll down to the 'Assets' list at the bottom of the page and left-click on the **Source code (zip)** link to download it as a zip file to your computer.  You can then extract the zip to obtain the librealsense source code folder and build 2.48.0.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0

rythm07, 2021-08-21:
Ok thank you so much for your help and availability! I will try to align the versions and I will tell you whether the problem is solved!

doronhi, 2021-08-24:
Hi,
The example in the README.md file states you should use the absolute path to the config file:
`ros2 launch realsense2_camera rs_launch.py config_file:="'$COLCON_PREFIX_PATH/realsense2_camera/share/realsense2_camera/config/d435i.yaml'"`
Accordingly, if you with to insert the "config_file" value as a default in the launch.py file you should use the full path. In my case it translates into:
```
                           {'name': 'config_file',                  'default': "'/home/doronhi/ros2_ws/install/realsense2_camera/share/realsense2_camera/config/d435i.yaml'", 'description': 'yaml config file'},
```

Considering my running directory is "ros2_ws" I can also use a relative path: `install/realsense2_camera/share/realsense2_camera/config/d435i.yaml`

Basically, the issue is that the file was not found.

Hope it helps.

doronhi, 2021-08-24:
For future cases, please try to copy the code parts instead of a screenshot. It will make it easier to reproduce.

rythm07, 2021-08-24:
Thank you so much @doronhi and @MartyG-RealSense for your help. I installed the 2.48 version of the SDK to match the version of the wrapper and change the path to the absolute path and it's loading the yaml file!

It worked!

MartyG-RealSense, 2021-08-24:
That's excellent news @rythm07 that you were successful - thanks very much for the update!  :)

closed 2021-08-30 by MartyG-RealSense