realsenseai/realsense-ros #2044: d450 module: No RealSense devices were found!

daniele-gusmini, 2021-08-22:
Hi everyone!
I'm using a RealSense d450 module with vision processor D4 v3, I see the camera from RealSense viewer but it is not working using the ROS wrapper. I saw in the release note that the d455 camera is supported, I thought that the d450 module + D4 is the same than d455 camera (also RealSense viewer recognise the camera as d455)

Could you help me?

Thanks

MartyG-RealSense, 2021-08-22:
Hi @daniele-gusmini  Sometimes a RealSense camera will work correctly in the RealSense Viewer but not in the RealSense ROS wrapper.  It sounds as though your D4 V3 and Depth Module board and the interposer joining them together are set up correctly if the camera is working in the Viewer.

May I first ask the questions below about your RealSense ROS setup.

-  Which computer or computing device are you using with the camera (PC, Raspberry Pi, Nvidia Jetson, etc)
-  Which ROS branch are you using (ROS or ROS2)
-  Which method did you use to install the ROS wrapper - building from packages with **Method 1** or building librealsense and the ROS wrapper separately, building the wrapper from source code?
-  Please provide a roslaunch log indicating the errors that occur when you try to launch the wrapper

daniele-gusmini, 2021-08-22:
Hi @MartyG-RealSense, thank you in advance!

The information you asked for:

- I'm using a NVIDIA Jetson Nano
- ROS (1)
- I built the two separately (Method 2)

I used without problem my RealSense D435 with a previous ROS wrapper (v. 2.2.11), I built the v 2.3.1 in a separate workspace to use the D455 since the other release was too old

MartyG-RealSense, 2021-08-22:
Thanks very much for the information.  The RealSense ROS wrapper version should ideally be matched as closely as possible to a specific librealsense SDK version listed in that particular wrapper's release notes.  So if you are using wrapper 2.3.1 then that should be used with librealsense SDK version **2.48.0**.

Furthermore, each time that you change the librealsense SDK version, the ROS wrapper needs to be built again.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.1

daniele-gusmini, 2021-08-23:
When I launch the RealSense node it states that the the Librealsense version is 2.48.0 but I do not upgrade it actually, now I try to rebuild both the library and the node

MartyG-RealSense, 2021-08-29:
Hi @daniele-gusmini  Do you have an update about this case that you can provide, please?  Thanks!

daniele-gusmini, 2021-08-30:
Yes, I solved the problem after some attempts.
I recompiled the RealSense library with RSUSB enabled ( `-DFORCE_RSUSB_BACKEND=true` ), since my module was updated at the latest firmware I have to use librealsense2 v 2.49.0

MartyG-RealSense, 2021-08-30:
That's excellent news @daniele-gusmini  that you achieved a solution - thanks very much for the update!

closed 2021-09-05 by MartyG-RealSense