realsenseai/realsense-ros #2045: NUC RAM gets choked after sometime with 2 d435i

rahulsharma11, 2021-08-23:
Hi,
I am using 2 realsense d435i cameras with my NUC i5(32G RAM, 8cores) and doing rosbag record for both cameras.

I took care for the issue regarding v4l backend and recompiled realsense-ros and it seems perfectly fine working. 

Issue is -
After around 7 mins, my NUC memory gets fully occupied (32GB), and then rosbag record is dropping frames.
I am also doing split to split the record data and played with -b parameter. But no solution.

Any suggestion i should take care here? i compressed the data and tested then it is appearing after 30 mins. But it is happening.

Thanks.

MartyG-RealSense, 2021-08-23:
Hi @rahulsharma11  I have not heard of this particular problem with memory consumption when using rosbag record.  Do you experience the same memory consumption if you use **rosbag record -a** to record all currently published topics with a single command?

There was one case in the link below where a similar problem was occurring when using rosbag record.  The ROS user in that case was recording remotely from a robot using a local Linux machine.  Do you have a similar setup, or are the two D435i attached directly to the computer that is performing the rosbag record operation?

https://answers.ros.org/question/279612/ros-eats-up-ram-while-recording-rosbag-remotely/

rahulsharma11, 2021-08-23:
Hi, 
I have not tested with **rosbag record -a**. Will confirm after testing.

Just for reference the command i used-
**rosbag record -o test.bag --split --duration=15 -b 4096  /camera1/aligned_depth_to_color/camera_info /camera1/aligned_depth_to_color/image_raw /camera1/color/camera_info /camera1/color/image_raw  /camera2/aligned_depth_to_color/camera_info /camera2/aligned_depth_to_color/image_raw /camera2/color/camera_info /camera2/color/image_raw /tf /tf_static**

Also tested with --chunksize=256/1024/2048

The process of record is not remote. Its been done on the NUC only. 

Debug -
Initially it was not even going upto 5 mins and i found there are lots of warning regarding "RS2_USB_STATUS_NO_DEVICE". I found i need to built realsense_camera2 ros from source with V4L bakend... I did that and warning went off. But still memory is getting full after few mins.

MartyG-RealSense, 2021-08-23:
You could try recording a test bag in ROS (without launching the RealSense ROS wrapper) using the ROS tutorial in the link below.  If the memory consumption still occurs then that would indicate that the problem is not related to the RealSense ROS wrapper.

http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data#Recording_data_.28creating_a_bag_file.29

rahulsharma11, 2021-08-23:
Hi,
I didn't get "without Realsense ROS wrapper".

I am running the launchfile with two "rs_camera.launch"  for respective cameras only that publishes the messages. If i do not run this, then how can i get realsense topics to record?

FYI-
<group if="$(arg dual_camera)">
            
                <include file="$(find realsense2_camera)/launch/rs_camera.launch">
                <arg name="camera" value="camera2" />
            <arg name="serial_no" value="serial1" />
                <arg name="align_depth" value="true"/>
                <arg name="depth_width" value="640"/>
                <arg name="depth_height" value="480"/>
                <arg name="color_width" value="1280"/>
                <arg name="color_height" value="720"/>
                <arg name="initial_reset" value="true"/>
                </include>

            
            <include file="$(find realsense2_camera)/launch/rs_camera.launch">
                <arg name="camera" value="camera1" />
                <arg name="serial_no" value="serial2" />
                <arg name="align_depth" value="true"/>
                <arg name="depth_width" value="640"/>
                <arg name="depth_height" value="480"/>
                <arg name="color_width" value="1280"/>
                <arg name="color_height" value="720"/>
                <arg name="initial_reset" value="true"/>

                </include>
     </group>

Please correct me if i am wrong.

MartyG-RealSense, 2021-08-23:
The non-RealSense ROS tutorial that I linked to suggests testing recording by setting up three terminals with 'turtlesim' nodes and then rosbag recording the topics published by those turtle nodes.

Terminal 1:  roscore
Terminal 2: rosrun turtlesim turtlesim_node
Terminal 3: rosrun turtlesim turtle_teleop_key

rahulsharma11, 2021-08-25:
Hi, I tested with the same "http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data#Recording_data_.28creating_a_bag_file.29
"

i can confirm there is no issue with RAM choking.

MartyG-RealSense, 2021-08-25:
Thanks for testing the above tutorial.  So we could conclude from the test results that the memory consumption problem does not occur when using rosbag record only with ROS, without the RealSense ROS wrapper being launched.  But it does occur when the RealSense ROS wrapper has been launched and its topics are being recorded.

Do you experience the problem if you launch with **rs_multiple_devices.launch** instead of rs_camera?

https://github.com/IntelRealSense/realsense-ros#work-with-multiple-cameras

rahulsharma11, 2021-09-01:
Hi, I tried with -b 0 but no luck.
It is throwing error for buffer exceed too quickly.

MartyG-RealSense, 2021-09-01:
The issue seems to occur with **rosbag record** when performing live recording, not specifically with RealSense.  A number of other non-RealSense memory consumption reports related to rosbag record have been posted online over the years.  So this may be a question that is better asked on the **ROS Answers** forum.

https://answers.ros.org/questions/

MartyG-RealSense, 2021-09-07:
Hi @rahulsharma11  Considering the apparent nature of the problem not being specific to RealSense as described above, do you require further assistance with this case, please?  Thanks!

rahulsharma11, 2021-09-07:
No, thanks.

closed 2021-09-07 by rahulsharma11