realsenseai/realsense-ros #2046: Machine shutdown after launching D435

sammar80, 2021-08-23:
Hi , I'm using D435 on nvidia jetson nano with ros-melodic what I'm facing is that once I launch the rs_camera.launch the machine shutdown I tired several sdk versions and the same issue happens when I plug the camera using the provided type-C cable for USB 3 . I appreciate if anyone can help with this problem 
librealsense ver : 2.45.0
realsense-ros ver : 2.3.0
camera firmware : 5.12.13.50

MartyG-RealSense, 2021-08-23:
Hi @sammar80   Could you paste the launch log from your roslaunch into a comment below, please?  Thanks!

sammar80, 2021-08-23:
Hi @MartyG-RealSense  , this is all what I get before the nvidia shutdown
... logging to /home/optimus/.ros/log/a5646b7e-041a-11ec-ae8b-d03745187cad/roslaunch-Autobot-8223.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
]2;/home/optimus/catkin_ws/src/realsense-ros/realsense2_camera/launch/rs_camera.launch
[1mstarted roslaunch server http://Autobot:35413/[0m

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 45 lines cut ...]
[0m[ INFO] [1629727382.328490632]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16[0m
[0m[ INFO] [1629727382.329796247]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8[0m
[0m[ INFO] [1629727382.329886250]: setupPublishers...[0m
[0m[ INFO] [1629727382.346657894]: Expected frequency for depth = 30.00000[0m
[0m[ INFO] [1629727382.494310970]: Expected frequency for color = 30.00000[0m
[0m[ INFO] [1629727382.584277099]: setupStreams...[0m
[0m[ INFO] [1629727382.607547446]: insert Depth to Stereo Module[0m
[0m[ INFO] [1629727382.607655940]: insert Color to RGB Camera[0m
 23/08 10:03:02,682 WARNING [546928845184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[33m[ WARN] [1629727382.807105559]: [0m
[0m[ INFO] [1629727382.733614449]: SELECTED BASE:Depth, 0[0m
[0m[ INFO] [1629727382.791572757]: RealSense Node Is Up![0m
 23/08 10:03:02,809 WARNING [546928845184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/08 10:03:02,861 WARNING [546928845184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/08 10:03:02,914 WARNING [546928845184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/08 10:03:02,965 WARNING [546928845184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 23/08 10:03:03,471 WARNING [546928845184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/08 10:03:03,522 WARNING [546928845184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/08 10:03:03,675 WARNING [546928845184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/08 10:03:03,726 WARNING [546928845184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2021-08-23:
Thank you very much for the log.  As you are not defining a custom configuration in the roslaunch instruction, have you edited the rs_camera lauch file to change its default settings, please?

sammar80, 2021-08-23:
yes I reduced the depth and the color width to 640 here is my launch file , it was working before I upgrade sdk and the realsense ros to the new 
[rs_camera.txt](https://github.com/IntelRealSense/realsense-ros/files/7032754/rs_camera.txt)

sammar80, 2021-08-23:
hey @MartyG-RealSense  this shutdown issue also happening when I'm running the realsense-viewer

MartyG-RealSense, 2021-08-23:
I can't see anything in your custom launch file configuration that would cause a problem.   And as you mention, the RealSense Viewer is causing you shutdown problems too.

What method did you use to install the librealsense SDK and the RealSense ROS wrapper, please?

sammar80, 2021-08-23:
I follow the build from source instructions for the sdk and I used catkin build for the ros wrapper 
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

MartyG-RealSense, 2021-08-23:
If you are patching the kernel then I wonder whether the problem is occurring at that point, since Jetson has its own L4T-based patching script on the Jetson installation instructions page.

https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation#section-building-from-source-using-native-backend

Some Jetson users take the approach of building librealsense from source on Nano using the **RSUSB backend** method, which requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.  An RSUSB installation guide that has often worked for Nano users is in the link below

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

sammar80, 2021-08-23:
Thank you ,I will try that and get back to you

sammar80, 2021-08-24:
Hi I follow the links you sent but nothing solves the problem

MartyG-RealSense, 2021-08-24:
I recalled that there was a past case at https://github.com/IntelRealSense/realsense-ros/issues/1340#issuecomment-676759207 where a computer was shutting down after the launch of a RealSense program.  This case belonged to you and did not have a solution at the end of it.  So can you confirm that you are experiencing the same problem as that previous case, please?

sammar80, 2021-08-24:
No it is not the same case ,  in my case the nvidia keeps shutting down whenever I launch realsense_ros or run realsense-viewer

MartyG-RealSense, 2021-08-24:
I have handled a couple of past cases where a Jetson shuts down and reboots immediately after a launch but it is highly rare.  My understanding is that it is related to power supply (specifically, the camera drawing more power than the board can cope with).  

The link below has an example of a case where a Nano was slowing to a crawl or crashing when using the Viewer and roslaunch.  They were powering their board from the micro USB connection but their crashes cleared up when they switched to a 5v 4A power supply.

https://www.jetsonhacks.com/2019/05/16/jetson-nano-realsense-depth-camera/

Intel's own official Jetson instructions for RealSense recommend enabling the barrel jack connector on the Nano.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

sammar80, 2021-08-24:
Thank you for the links I checked and I'm using the max power for jetson nano . I installed the sdk and ros wrapper again and they are working only if I'm not using the USB 3 cable 

here is the cpu utilizations when running the camera using USB3 cable
[image]

logs for camera using USB2 cable
[realsense-ros-log.txt](https://github.com/IntelRealSense/realsense-ros/files/7041114/realsense-ros-log.txt)

MartyG-RealSense, 2021-08-25:
Your CPU usage percentage is high.  As you are using aJetson board, when you built librealsense from source did you include the build flag **-DBUILD_WITH_CUDA=true** in the CMake build instruction in order to build the librealense SDK with **CUDA** GPU support?

sammar80, 2021-08-25:
No currently I installed the precompiled librealsense SDK using these commands 
sudo apt-get install librealsense2-utils
sudo apt-get install librealsense2-dev

MartyG-RealSense, 2021-08-25:
Thank you.  You mention above that it only works when using a USB2 cable and not with a USB3 cable (hence your providing of a text-file log obtained with a USB2 cable).

  Is your USB 3 cable a Type C to C (a micro-sized connector on each end) instead of a Type A to Type C (a full-size USB connector on one end to connect to the computer and a micro-sized connector on the other end to plug into the camera).  A RealSense team member advises in https://github.com/IntelRealSense/librealsense/issues/4166#issuecomment-500374449 that some C-to-C cables only provide USB 3 capability in one direction.

MartyG-RealSense, 2021-08-31:
Hi @sammar80  Do you have an update about this case that you can provide, please?  Thanks!

sammar80, 2021-08-31:
Hi @MartyG-RealSense I used a visual preset and disable the  auto exposure the d435 camera is working now with the USB3 cable but the issue now is that it is flickering a lot

MartyG-RealSense, 2021-08-31:
Thanks very much for the update.  Are you using the camera in a location with ceiling strip lights?

sammar80, 2021-08-31:
No I am using it in a normal room light and the camera is parallel to the ground

MartyG-RealSense, 2021-09-01:
Which value is the manual depth exposure set to in the visual preset file, please?  The default depth exposure value of the D435 is **8500** and the default RGB exposure is **156**.

sammar80, 2021-09-01:
I am using the default exposure values for the depth and rgb I just disable the auto exposure

MartyG-RealSense, 2021-09-01:
Is it the depth or the RGB that is flickering?

sammar80, 2021-09-02:
sorry @MartyG-RealSense  for the late reply both are flickering but the RGB is flickering more

MartyG-RealSense, 2021-09-02:
And the Jetson shuts down again if you re-enable auto-exposure **and** use the visual preset?

sammar80, 2021-09-02:
yes

MartyG-RealSense, 2021-09-02:
Did you build the RealSense ROS wrapper from packages )**Method 1**) or from source code (**Method 2**)?

If you did build from source with Method 2: when you have tried different SDK versions, have you rebuilt the wrapper each time that you have changed the librealsense version?  This is something that is necessary to do each time the librealsense version is changed if building the wrapper from source with Method 2. 

It is also important to bear in mind that a particular RealSense ROS wrapper version should ideally be matched with a specific librealsense version listed in the release notes for that wrapper version.  So using different librealsense versions with wrapper version 2.3.0 would have negative consequences, since wrapper 2.3.0 is designed specifically for librealsense 2.45.0.

********

May I also ask whether you are applying different settings (such as auto-exposure enabled / disabled) using the roslaunch instruction or the launch file contents, or whether you are changing settings in the dynamic reconfigure interface after launch.  

https://github.com/IntelRealSense/realsense-ros#set-camera-controls-using-dynamic-reconfigure-params

sammar80, 2021-09-02:
initially I was try to build both the librealsense and ros wrapper form source and I was using the compatible versions but eventually I ends up with method 1 using the installation commands  and I got the latest versions 

and for the settings I'm disabling the auto-exposure from realsense-viewer and export the settings then loaded through my roslaunch file

MartyG-RealSense, 2021-09-03:
I reviewed the whole case from the beginning again to look for fresh insights but we have already covered most possibiities.

-  Your Jetson is the only one that I have encountered that shuts itself down when using RealSense, like in your other case at https://github.com/IntelRealSense/realsense-ros/issues/1340#issuecomment-676759207

-  I cannot think of a reason why simply having auto-exposure enabled would make enough difference to the power draw of the camera to cause it to shut down due to insufficient power available.

-  However, in general Jetsons can certainly shut themselves down.  The two main causes are insufficient power or excessive temperature on the Jetson board (e.g the cooling fan is not running fast enough).

-  We covered the power supply aspect in https://github.com/IntelRealSense/realsense-ros/issues/2046#issuecomment-904721016

-  You could try setting the Jetson's fan to run at full speed with the command below to help rule out a temperature related shutdown.

**sudo sh -c 'echo 255 > /sys/devices/pwm-fan/target_pwm'**

sammar80, 2021-09-03:
Tyanks @MartyG-RealSense  I will try the fan settings but this is strange because it was working fine before I upgrad to the new versions of the sdk and ros wrappers

MartyG-RealSense, 2021-09-03:
You could also test whether there is a problem with the json file exported from the Viewer by using the pre-made **DefaultPreset_D435.json** json file from the Visual Presets documentation, which sets auto-exposure to True.  If the Jetson does not shut down when using that file, it could suggest that there is a problem in your Viewer-exported json.

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets#preset-table
https://raw.githubusercontent.com/wiki/IntelRealSense/librealsense/d400_presets/DefaultPreset_D435.json

sammar80, 2021-09-03:
I tried the default presets and also other presets from the link and the same problem is occurring as long the auto exposure is enabled

MartyG-RealSense, 2021-09-03:
Does enabling just one auto-exposure cause the shutdown, or does it only occur if both auto-exposures are enabled?  For example, if you had RGB auto-exposure enabled but depth exposure disabled, does it still shut down?

sammar80, 2021-09-03:
it happens when both auto-exposures are enabled but when the depth auto-exposure is disabled and the RGB exposure is enabled it works fine

MartyG-RealSense, 2021-09-03:
Thanks for the information.  Does it have the same behavior (works but flickers) if depth auto-exposure is enabled and RGB exposure is disabled?

sammar80, 2021-09-03:
I didn't try this combination but when I disable both it doesn't work

MartyG-RealSense, 2021-09-03:
Next, could you try using the rosparam code below to disable a setting called **Auto-Exposure Priority** in the launch file.  This is a separate setting from RGB Auto-Exposure.

When Auto-Exposure is enabled and Auto-Exposure Priority is disabled, the SDK tries to enforce a constant FPS speed instead of allowing FPS to vary.

```
<rosparam>
/camera/rgb_camera/rs435_color_auto_exposure_priority: false
</rosparam>
```

sammar80, 2021-09-03:
So I should enable the depth auto-exposure and set this parameter in the roslaunch ,is that correct?

MartyG-RealSense, 2021-09-03:
I suggest setting the parameter in the launch and then testing with depth auto-exposure **disabled** and RGB exposure enabled first (given that enabling depth auto-exposure usually causes a shutdown).  If it does not make a difference, try it again with both depth and RGB auto-exposure enabled.

sammar80, 2021-09-03:
Thank you @MartyG-RealSense I tried the parameter with auto-exposure disabled/enabled  for the depth and jetson nano didn't shutdown when I enable the depth auto-exposure and it also fixed the depth image flicker still the RGB image flickers

MartyG-RealSense, 2021-09-04:
Great to hear that you made progress!  Does the flicker of the RGB reduce if you set the color FPS in the launch to **60** instead of 30?

MartyG-RealSense, 2021-09-10:
Hi @sammar80  Do you require further assistance with this case, please?  Thanks!

sammar80, 2021-09-10:
Hi @MartyG-RealSense sorry for the delayed reply I tried to change the RGB FPS to 60 from the roslaunch file but it doesn't take effect it remains 30 fps

MartyG-RealSense, 2021-09-10:
In order for a custom configuration to be valid, it must provide three factors - width, height and FPS.  Otherwise, the custom configuration is deemed invalid by the launch and the default stream configuration of the particular camera model is used instead.

Are you able to obtain 60 FPS color if you use the roslaunch instruction below, please?

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=60**

sammar80, 2021-09-10:
Thank you very much @MartyG-RealSense that worked fine and the RGB is not flickering any more

MartyG-RealSense, 2021-09-10:
That's great news, @sammar80 - thanks for the update!   :)

sammar80, 2021-09-10:
Yes finally  it works fine , thank you @MartyG-RealSense  for your patience and time :)

closed 2021-09-16 by MartyG-RealSense