realsenseai/realsense-ros #2047: on registered depth image

Angit16, 2021-08-24:
Hi,

I have a basic query. I am using D435 and D415 cameras. If I want to obtain a registered depth image topic, will it be fine to use `rs_camera.launch` only or I need to use `rs_rgbd.launch`? The question is in connection with the package [rtabmap_ros](http://wiki.ros.org/rtabmap_ros) that expects registered depth image. Also, how do I verify whether the depth image that I am viewing in `rviz` is registered or not? 

I understand that these questions may sound boring to you, but I need to clear out my doubts.

cheers,

MartyG-RealSense, 2021-08-24:
Hi @Angit16   Your questions are not boring at all   :)

The term **registered depth** is not used very commonly on this support forum.  My understanding is that it refers to aligning depth to color.  Is that what you are aiming to achieve, please?

If you do wish to perform depth-color alignment with rs_camera.launch then you can set **align_depth** to True in your roslaunch instruction.

**roslaunch realsense2_camera rs_camera.launch align_depth:=true**

When align_depth is True, the additional topics below are published:

/camera/aligned_depth_to_color/image_raw 
/camera/aligned_depth_to_color/camera_info

If you have set RViz to render **/camera/aligned_depth_to_color/image_raw** then you can feel assured that you are publishing depth-color aligned data.

Angit16, 2021-08-24:
Many thanks for the quick response.

Yes, I was trying to mean the same, aligning depth with color. Also, from L96-97 of `rs_rgbd.launch `depth_registered` argument is declared. I am trying to relate the same with the registered depth image on `/depth/image topic from [rtabmap_ros](http://wiki.ros.org/rtabmap_ros)`. Can you please confirm if they are the same? Also, how do I find the difference between `/camera/aligned_depth_to_color/image_raw and` and ` /camera/depth/image_rect_raw`? They seem to be very similar to each other.

MartyG-RealSense, 2021-08-24:
As the topic names suggest, **/camera/depth/image_rect_raw** contains purely depth data, whilst **/camera/aligned_depth_to_color/image_raw** is a topic containing color data that is aligned to depth that is only published when align_depth is set to True.

If you literally wanted to compare these two topics to each other using ROS to find differences between them then I do not know of a method for doing so.  The only suggestion that my research could find was to compare them using **message_filters**

https://answers.ros.org/question/321357/testing-node-that-compares-two-topics/

In regard to your rtabmap_ros **/depth/image** topic and relating it to the depth data from D435, the discussion in the link below would seem to be relevant to that question.

https://github.com/introlab/rtabmap_ros/issues/318

Angit16, 2021-08-25:
thank you, it is much clearer now.

Angit16, 2021-09-01:
@MartyG-RealSense no thanks. I'm closing it.

closed 2021-09-01 by Angit16