realsenseai/realsense-ros #2053: IR camera topics not publishing for D415

Angit16, 2021-08-29:
Hello, with reference to [this](https://github.com/introlab/rtabmap_ros/issues/565#issuecomment-815762643) link, I tried to obtain IR data from D415 camera. However, I find the following warnings:

`(backend-v4l2.cpp:1358) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 !`
`(backend-v4l2.cpp:1358) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!`
Then,
` 29/08 17:45:11,965 WARNING [139843700741888] (backend-v4l2.cpp:1122) Frames didn't arrived within 5 seconds
`
Also, nothing is publishing on the IR camera topics. Following are the specifications of my OS kernel and other relevant stuff.

ubuntu kernel : 5.4.0-80-generic
librealsense: V2.48

Would you suggest what could be the possible issue here?

MartyG-RealSense, 2021-08-29:
Hi @Angit16  The warning **Pixel format ... likely requires patch for fourcc code W10** suggests that the Ubuntu kernel needs to be patched.  If you built the librealsense SDK from source code and did not patch the kernel, section **4.  Build and apply patched kernel modules** of the SDK source code compilation instructions in the link below explains how to do so for the Ubuntu version that you are using.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

In regard to the IR topics, if you are attempting to publish those topics in the RealSense ROS wrapper then you can enable them by adding the commands below to your roslaunch instruction:

**enable_infra1:=true enable_infra2:=true**

For example:

**roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true**

Angit16, 2021-08-30:
Hello  @MartyG-RealSense 

Thank you for your response. I followed all the steps you mentioned before, already. However, just for my own satisfaction tested it on another system and it worked there. So, I may need to reinstall the kernel patch once again, on the previous system.
You may please close the issue.

MartyG-RealSense, 2021-08-30:
Thanks very much @Angit16 for the update.  

Enabling the infrared topics should not prevent the depth topic from being published.

Angit16, 2021-08-30:
Hi @MartyG-RealSense I checked, the depth topics are not publishing when IRs are enabled.

MartyG-RealSense, 2021-08-30:
There is another case of this problem occurring in https://github.com/IntelRealSense/realsense-ros/issues/1377#issuecomment-693465427

In that particular case the non-publishing of depth was found to be caused by setting different resolutions for the depth and IR topics.  Are you also setting different resolutions for depth and IR, please?

Angit16, 2021-08-30:
resolutions aka width and height of color, depth, and iR are all set the same respectively.

MartyG-RealSense, 2021-08-30:
Does the launch define height, width **and** FPS for color, depth and IR inside the launch file or in the roslaunch instruction?  If custom definitions for width and height were set but a definition for FPS was not provided then the custom configuration would be deemed invalid and the default configuration of that particular camera model would be applied instead.

Angit16, 2021-08-30:
For convenience, put up the `rs_camera.launch` below.

`<launch>
  <arg name="file_name_json" default="$(find realsense2_camera)/launch/presets/MediumDensity/mayank.json"/> 

  
  <arg name="serial_no"           default="102122060989"/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default="$(arg file_name_json)"/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="-1"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="-1"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
[... 57 lines cut ...]
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="stereo_module/exposure/1" value="$(arg stereo_module/exposure/1)"/>
      <arg name="stereo_module/gain/1"     value="$(arg stereo_module/gain/1)"/>
      <arg name="stereo_module/exposure/2" value="$(arg stereo_module/exposure/2)"/>
      <arg name="stereo_module/gain/2"     value="$(arg stereo_module/gain/2)"/>

      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>
      
    </include>
  </group>
</launch>
`

MartyG-RealSense, 2021-08-30:
Please try the launch instruction below to override the stream definitions in the launch file and set all three stream types to 640x480 and 30 FPS.

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30 enable_infra1:=true enable_infra2:=true infra1_width:=640 infra1_height:=480 infra1_fps:=30 infra2_width:=640 infra2_height:=480 infra2_fps:=30**

Angit16, 2021-08-30:
thank you so much, it works.

MartyG-RealSense, 2021-08-30:
You are very welcome, @Angit16 - great to hear that it worked for you   :)

closed 2021-09-01 by Angit16