realsenseai/realsense-ros #2054: low fps on jetson board

Jaelly, 2021-08-30:


MartyG-RealSense, 2021-08-30:
Hi @Jaelly  As you mention seeing the color image on the other PC, are you using a networking connection such as **ssh** to connect remotely from your PC to a Jetson Xavier with the D435i camera attached to it, please?

If you are using networking, slowdown can sometimes occur, especially if using wireless networking (WLAN / wi-fi), as described in the case https://github.com/IntelRealSense/realsense-ros/issues/2052 from this week.

Can you also confirm please that you have installed support for RGBD launch mode.

**ROS Kinetic**
sudo apt-get install ros-kinetic-rgbd-launch

**ROS Melodic**
sudo apt-get install ros-melodic-rgbd-launch

Jaelly, 2021-09-01:
Well, I used without ssh. I just match ip address on bash.
And, I have to make without wire on robot, so I used Wifi.

I also had installed RGBD launch.

MartyG-RealSense, 2021-09-01:
Do you see a positive difference if you turn off wi-fi power saving in Ubuntu with the command below:

**iw dev wlan0 set power_save off**

MartyG-RealSense, 2021-09-13:
Case closed due to no further comments received.

closed 2021-09-13 by MartyG-RealSense

shubham300ska, 2022-01-18:
> 

Hi,
I have same issue, I tried With  ssh also. I lunch-roslaunch realsense2_camera rs_camera.launch ,If I rostopic hz  /camera/color/image_raw then -no new messages and launch rviz with command -rosrun rviz rviz i got Image but is in Slow  image status in rviz is 1 image after 9 or 10 second 
[image]

MartyG-RealSense, 2022-01-18:
Hi @shubham300ska  There was a recent case at https://github.com/IntelRealSense/realsense-ros/issues/2182 where a Jetson NX user could not receive any messages from /camera/color/image_raw when using an ssh connection.

My understanding of the comment https://github.com/IntelRealSense/realsense-ros/issues/2182#issuecomment-989158769 in that discussion is that the RealSense user in the case attributed the problem to use of Ubuntu 20.04 and therefore decided to wait for Nvidia to release a JetPack version that would support Ubuntu 20.04.

Which Ubuntu version are you using, please?