realsenseai/realsense-ros #2055: Setting "High Accuracy" visual preset on D435 launch

aharmat, 2021-08-30:
My understanding is that it's not possible to set the visual preset (e.g. High Accuracy) using the launch file, and the two options are:
1. Use dynamic reconfigure after the camera driver is up
2. Use a json parameter file that sets a non-default visual preset

I'm able to set the visual preset using `rqt_reconfigure`, but my application does not permit manual intervention, so I'm trying to get option 2 to work. Looking at https://dev.intelrealsense.com/docs/d400-series-visual-presets I see a number of json files that are listed, but all the links are broken! Does anyone have a working link for MedResHighAccuracyPreset.json? Or any other suggestion for setting a non-default visual preset on camera startup?

MartyG-RealSense, 2021-08-31:
Hi @aharmat  You can define an **absolute file path** to a json file by including the **json_file_path** command in the roslaunch instruction or in the launch file.  Examples are given below.

**roslaunch**
roslaunch realsense2_camera rs_camera.launch json_file_path:=/path/to/config/DefaultPreset_viewer.json

**Launch file
https://github.com/IntelRealSense/realsense-ros/issues/1890#issuecomment-847813284

In the above link, the launch file instruction **<arg name="json_file_path" value="$(arg HighResMedDensityPreset)"/>is used to load the HighResMedDensityPreset json filename.

I confirm that the json links on the page that you quoted are not working.  You can obtain the json files from the GitHub version of that documentation page though.  The json that you want is called **MedDensityPreset.json**

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets#preset-table
https://raw.githubusercontent.com/wiki/IntelRealSense/librealsense/d400_presets/MidDensityPreset.json

aharmat, 2021-09-02:
Hi Marty, thank you for the quick response. I'm using the realsense d435 in a robotics application where the High Accuracy preset has been working well in our testing.

I tried the HighAccuracyPreset.json that you linked to, but I get the following error from the realsense2_camera driver:
```
process[realsense/realsense2_camera_manager-1]: started with pid [1755]
[ INFO] [1630540614.304258647]: Initializing nodelet with 8 worker threads.
process[realsense/realsense2_camera-2]: started with pid [1756]
[ INFO] [1630540614.742601898]: RealSense ROS v2.2.8
[ INFO] [1630540614.744120133]: Running with LibRealSense v2.28.1
[ INFO] [1630540615.292248762]: getParameters...
[ INFO] [1630540615.762550188]: setupDevice...
 01/09 23:57:00,900 WARNING [547667046800] (types.cpp:48) get_xu(id=2) failed! Last Error: Resource temporarily unavailable
[ERROR] [1630540620.901369022]: An exception has been thrown: get_xu(id=2) failed! Last Error: Resource temporarily unavailable
```

At this point, even if I try re-launching the driver without a json file, I get the following:
```
process[realsense/realsense2_camera_manager-1]: started with pid [1844]
[ INFO] [1630540840.604070767]: Initializing nodelet with 8 worker threads.
process[realsense/realsense2_camera-2]: started with pid [1845]
[ INFO] [1630540841.076792774]: RealSense ROS v2.2.8
[ INFO] [1630540841.076885963]: Running with LibRealSense v2.28.1
[ INFO] [1630540841.647686849]: getParameters...
[ INFO] [1630540842.056340626]: setupDevice...
[ INFO] [1630540842.056521596]: JSON file is not provided
[ INFO] [1630540842.056599265]: ROS Node Namespace: realsense
[ INFO] [1630540842.056647748]: Device Name: Intel RealSense D435
[ INFO] [1630540842.056682726]: Device Serial No: 102422072501
[ INFO] [1630540842.056783820]: Device FW version: 05.11.01.100
[ INFO] [1630540842.056829455]: Device Product ID: 0x0B07
[ INFO] [1630540842.057641088]: Enable PointCloud: Off
[ INFO] [1630540842.057682914]: Align Depth: Off
[ INFO] [1630540842.057731941]: Sync Mode: Off
[ INFO] [1630540842.058510932]: Device Sensors: 
[ INFO] [1630540842.059238687]: Stereo Module was found.
[ INFO] [1630540842.059383144]: RGB Camera was found.
[ INFO] [1630540842.059505263]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1630540842.059558963]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1630540842.059599477]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1630540842.059646264]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1630540842.059703771]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1630540842.059751870]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1630540842.060233339]: num_filters: 0
[ INFO] [1630540842.060283902]: Setting Dynamic reconfig parameters.
 02/09 00:00:47,271 WARNING [547782746512] (types.cpp:48) get_xu(id=2) failed! Last Error: Resource temporarily unavailable
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  get_xu(id=2) failed! Last Error: Resource temporarily unavailable
```

I have to physically unplug and replug the realsense for me to be able to launch without a json file.

I'm running the driver in a Docker container on Ubuntu xenial and ROS kinetic, the installed version of librealsense is `librealsense2.so.2.28`, and I'm using the 2.2.8 tag of `realsense-ros`.

MartyG-RealSense, 2021-09-02:
Given that the error is occurring after the launch tries to access dynamic reconfig parameters, can you confirm please whether you have the **ddynamic_reconfigure** package for Kinetic installed?  If not, it can be installed for ROS Kinetic with the command below:

**sudo apt install ros-kinetic-ddynamic-reconfigure**

aharmat, 2021-09-02:
I can confirm that ddynamic_reconfigure is installed. When I launch the camera with no json file after a fresh plug-in, I can use `rqt_reconfigure` to change parameters on the camera, which is how I was able to see that High Accuracy mode works well for me.

Also, in the first log output I posted, the `Last Error: Resource temporarily unavailable` happens after `setupDevice..` not `Setting Dynamic reconfig parameters`, so I don't believe the problem is related to dynamic reconfigure.

aharmat, 2021-09-02:
I just added the ability to disable dynamic reconfigure setup in `base_realsense_node.cpp`, and found that once the realsense has given me an error from trying to load a json file, if I then launch the driver with dynamic reconfigure disabled the driver comes up! If I launch with dynamic reconfigure enabled, the driver fails after `Setting Dynamic reconfig parameters`.

It seems to me that the realsense's firmware gets into a funny state after it fails to set the json parameters, to the point where just trying to read the reconfigurable parameters on a new launch of the driver causes a `Resource temporarily unavailable` error. I'm guessing this is at the firmware level because an unplug and replug of the sensor hardware solves the problem.

However, where things get interesting is that using the same sensor with the same firmware, I'm able to get json file loading to work on my laptop (x86) running Ubuntu 20.04 with ROS noetic. This implies that it's NOT a firmware problem, but rather a driver problem, either librealsense2 or realsense-ros. What do you think?

aharmat, 2021-09-02:
After some more digging, it seems that we can actually set ALL dynamically reconfigurable parameters using ROS params at startup, but this wasn't working for me, the sensor was entering some kind of error state.

The solution proved to be moving the dynamic reconfigure setup function alter in the initialization chain, which allowed the sensor to enter a stable state before trying to set parameters on it. I made a PR here: https://github.com/IntelRealSense/realsense-ros/issues/2055

MartyG-RealSense, 2021-09-03:
It's great to hear that you were able to develop a solution for the dynamic reconfigure situation, @aharmat 

Thanks so much for creating the PR - I have added an **Enhancement** tag to this case to signify that it should be kept open whilst your PR is active.

doronhi, 2021-11-14:
I added the following lines to rs_camera.launch:
```
  <rosparam>
      /camera/stereo_module/visual_preset: 3
  </rosparam>
```
And tested on 2 VMware machines on my laptop - Ubuntu 20.04 and Ubuntu 16.04 (noetic and kinetic)
Everything seems fine and the visual preset was changed to "High Accuracy" without any error.
I'll locate and try next on a Jetson Xavier.

MartyG-RealSense, 2023-07-18:
Case closed due to PR being closed because of the ROS1 wrapper no longer being updated.

closed 2023-07-18 by MartyG-RealSense