realsenseai/realsense-ros #2056: Realsense-ros on Windows Setup

Ahsayn, 2021-09-01:
Hi

I'm a novice, so please advise if posted incorrectly.

I am trying to setup realsense-ros on Windows 10 (ROS 1) following the instructions provided (installing Realsense SDK, cloning ddynamic-config and realsense-ros repos, etc.) Things work fine in WSL2, but I need this natively on the Windows machine so I can use my D435 and T265 (no USB support in WSL2 yet)

Detailed Procedure:

1. Install ROS 1 on Windows 10 - no issues there
2. Clone and build ddynamic-config - no issues there
3. Clone realsense-ros and switch to 2.3.1 head using "git checkout 2.3.1" command (cannot use linux equivalent provided in official steps on windows terminal: git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`) - no issues noted there
4. catkin_make clean - no issues there

My problems seemingly start at "catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release":

--Terminal readout--

c:\ws>catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
Base path: c:\ws
Source space: c:\ws\src
Build space: c:\ws\build
Devel space: c:\ws\devel
Install space: c:\ws\install
####
#### Running command: "cmake c:\ws\src -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=c:\ws\devel -DCMAKE_INSTALL_PREFIX=c:\ws\install -G Ninja" in "c:\ws\build"
####
-- Using CATKIN_DEVEL_PREFIX: C:/ws/devel
-- Using CMAKE_PREFIX_PATH: C:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows;c:\opt\ros\noetic\x64
-- This workspace overlays: c:/opt/ros/noetic/x64
-- Using PYTHON_EXECUTABLE: C:/opt/ros/noetic/x64/python.exe
-- Using default Python package layout
-- Using empy: C:/opt/ros/noetic/x64/lib/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: False
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - ddynamic_reconfigure
-- ~~  - realsense2_camera
-- ~~  - realsense2_description
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ddynamic_reconfigure'
-- ==> add_subdirectory(ddynamic_reconfigure)
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Create Release Build.
-- realsense2_camera: 2 messages, 0 services
-- +++ processing catkin package: 'realsense2_description'
-- ==> add_subdirectory(realsense-ros/realsense2_description)
-- Configuring done
-- Generating done
-- Build files have been written to: C:/ws/build
####
#### Running command: "ninja -j12 -l12" in "c:\ws\build"
####
ninja: error: 'C:/Program Files (x86)/Windows Kits/10/Lib/10.0.19041.0/um/x64/RpcRT4.Lib', needed by 'C:/ws/devel/bin/realsense2_camera.dll', missing and no known rule to make it
Invoking "ninja -j12 -l12" failed

--Terminal readout ends--

I have checked for the RpcRT4.Lib file in the location noted above and it does not exist. Folder there is C:\Program Files (x86)\Windows Kits\10\Lib\10.0.18362.0

Any advice on how to tackle this issue? I can get things done just fine within WSL, but cannot seem to make progress trying the Windows approach. Thanks!

MartyG-RealSense, 2021-09-01:
Hi @Ahsayn  I checked your procedure carefully and it seems as though you have gone through the same steps as the RealSense ROS wrapper's Windows installation instructions (aside from the git checkout instruction, but specifying the version number instead of using grep is usually the only way to perform the checkout on Windows).

Typically when installing the ROS wrapper on Windows, the **catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release** instruction is where problems are most likely to occur.

Do you have different results if you change-DCATKIN_ENABLE_TESTING to **True** instead of False

Ahsayn, 2021-09-01:
Thanks @MartyG-RealSense. 
I have tried your suggestion without success unfortunately. I am going to investigate that error referenced in my initial post above:
ninja: error: 'C:/Program Files (x86)/Windows Kits/10/Lib/10.0.19041.0/um/x64/RpcRT4.Lib', needed by 'C:/ws/devel/bin/realsense2_camera.dll', missing and no known rule to make it
Invoking "ninja -j12 -l12" failed

I now know that the files that are missing are part of the Windows 10 SDK. I have a newer version than 10.0.19041.0. Renaming the folder to force use of the current version I have leads to other errors, so I will try to find the older version that realsense2_camera.dll is looking for at build time (seems like something I have to do via Visual Studio). If you can shed light on this, then kindly let me know.

Thanks also for the dual-boot suggestion. Carefully considering that or some hacks to get USB access in WSL2 I have read.

MartyG-RealSense, 2021-09-02:
You could also try strictly following the methodology for Windows in the RealSense ROS installation documentation to see whether the differences in your own method affected the outcome.  

The main differences are that in the official instructions, **/catkin_ws/src** - a catkin workspace called catkin_ws with an 'src' subdirectory - is created instead of **c:/ws**, and also that the instruction **catkin_init_workspace** is input directly before **catkin_make clean** (whilst in the 'src' subdirectory, I believe, before switching to the catkin folder to input catkin_make clean).

Ahsayn, 2021-09-05:
No luck yet. I am going to wipe the entire workspace and try this from scratch.

MartyG-RealSense, 2021-09-05:
Thanks very much @Ahsayn for the update - good luck!

Ahsayn, 2021-09-08:
I'm probably going to abandon this issue owing to limited time.

closed 2021-09-08 by Ahsayn

MartyG-RealSense, 2021-09-08:
Okay, thanks very much for the update, @Ahsayn - please do feel free to return to this support forum if you attempt to solve the issue again.