realsenseai/realsense-ros #2059: Adding RealSense cameras to transformation tree

NimaNGRD, 2021-09-03:
Hi everyone,
I am using ROS Foxy together with D435 and T265 cameras.
I want to add the URDF models of cameras to my main URDF file to build the transformation tree using tf2 and state publisher. I should add that I am working on an actual robot, not a simulation.
How can I do this? Appreciate any help and comment.

RealSenseSupport, 2021-09-07:
@NimaNGRD I think this question is more related to ROS2. Could you please try to seek more support from ROS2 project? Thanks!

RealSenseSupport, 2021-09-24:
@NimaNGRD If no other questions this ticket will be closed in 7 days. Thanks!

RealSenseSupport, 2021-10-08:
@NimaNGRD Since no other questions from your side, we closed this ticket now. Thanks!