realsenseai/realsense-ros #2060: d435i with external Odometry not working with Rtabmap.

rahulsharma11, 2021-09-04:
Hi, I am using d435i camera but want to use odometry geneated by external device (with  odometry topic  by external device is "/Odometry").

I did following steps in launch file-
Visual_odometry = "false"
odom_topic = "/Odometry".

Note-
rqt_tf_tree graph is attached for the external device-
[image]

**Launch files configuration-**

  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera" value="camera1" />
    <arg name="serial_no" value="SERIAL NUMBER" />
    <arg name="align_depth" value="true" />
  </include>

  
  <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
    <arg name="args" value="--delete_db_on_start"/>
    <arg name="depth_topic" value="/camera1/aligned_depth_to_color/image_raw"/>
    <arg name="rgb_topic" value="/camera1/color/image_raw"/>
    <arg name="camera_info_topic" value="/camera1/color/camera_info"/>
    <arg name="approx_sync" value="false"/>
    <arg name="visual_odometry" value="false"/>
    <arg name="odom_topic" value="/Odometry"/>
  </include>

  <node pkg="tf" type="static_transform_publisher" name="laserMapping_"
      args="0.0 0.0 0.0 0.0 0.0 0.0 /camera_init /camera1_link 100" />
  

**config in rtabmap.launch-**
frame_id = "camera1_link"
odom_frame_id = "camera_init"

Question-
I can see only current map(pointcloud) but map is not stiching as  if move around (Like SLAM is not happening).

Any suggestions would be appreciable.
Thanks.

rahulsharma11, 2021-09-05:
Hi,
@doronhi , can you please suggest. I need it a bit urgent.
Thanks.

aditdoshi333, 2021-09-05:
Hello,

@doronhi @MartyG-RealSense Please throw some light.

Not able to understand what to set in frame_id and odom_frame_id. Currently, the odometry bag file is publishing tf between camera_init and body. We are not able to understand the connection between camera_link, camera_init, and body. 

As soon as I turn on visual odometry and set frame_id="camera_link" and odom_frame_id="" everything is working. But I want to use external odometry. 

Thank you

MartyG-RealSense, 2021-09-05:
Hi @rahulsharma11 and @aditdoshi333   I will highlight this issue to @doronhi

doronhi, 2021-09-07:
I am sorry. It has been a long time and I honestly don't remember how to configure rtabmap right now. I'll have to play around with it in a way I'm sure you already have.
This question relates much more to rtabmap than to realsense-ros. My best advice is to place this question under the rtabmap repository or ROS-answers perhaps.

rahulsharma11, 2021-09-08:
Ok Thanks.

closed 2021-09-08 by rahulsharma11