realsenseai/realsense-ros #2062: How do multiple devices work synchronously

caobingquan6, 2021-09-05:
Due to the optical interference phenomenon caused by TOF principle, how to realize the normal operation of multiple L515 with overlapping field of view? The hardware synchronization case described on the website (https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup) does not guarantee that multiple cameras will run at high frequencies (such as 30Hz) in real time.

MartyG-RealSense, 2021-09-05:
Hi @caobingquan6  Could you describe your multiple camera setup please, including the following information:

-  How many L515 cameras you are using.
-  If you are connecting all of the cameras to the same computer or using more than one computer.
-  What computer are you using (PC, Nvidia Jetson, Raspberry Pi, etc).
-  Are the cameras attached to a USB hub or plugged directly into the USB ports of the computer.

As described in Section 2 Parts A to D of the L515 multiple camera paper, a range of factors can affect the performance of the multiple camera setup, including the number of cameras used, if the cameras are attached to a USB hub and the capabilities of the computer hardware.

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup#section-2multi-camera-considerations

caobingquan6, 2021-09-05:
Thank you for your reply.

I currently have ten L515 cameras, they are distributed above the room, and some cameras have a certain overlapping field of view. Each camera is connected to a Raspberry Pi, and the point cloud information of the camera is read through ROS. These ten Raspberry Pis are connected to the switch through a network cable, and the switch is connected to a PC (ubuntu). Finally, the point cloud data (30Hz) of each L515 is read through the master-slave mechanism of ROS. The system needs to run in real time.

The problem encountered at present is due to the overlapping field of view between some cameras, which results in abnormal depth data reading, that is, light interference. So I want to solve this situation, is there a solution?

The phenomenon of light interference is shown in Figure 2 on the website (https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup).

MartyG-RealSense, 2021-09-05:
Have you set the **Inter Cam Sync Mode** of each L515 camera to '1' through ROS to define them as Slave cameras, please?

The link below describes how to set the value in ROS during runtime with a **dynamic reconfigure** command.

https://github.com/IntelRealSense/realsense-ros/issues/984#issuecomment-707729244

Alternatively, you could load a **json** camera configuration file during launch that contains a definition for the value of inter_cam_sync_mode

https://github.com/IntelRealSense/realsense-ros/issues/1540

********

You could also check whether using the **Short Range** preset improves your results without using hardware sync, as this L515 preset reduces the **Laser Power** and the **Receiver Gain** values.

caobingquan6, 2021-09-06:
As you said, if I set the Inter Cam Sync Mode of each L515 to 1, do I still need to set the external hardware synchronization at the same time. Because I see the sentence on the website (https://github.com/IntelRealSense/librealsense/issues/7372)"You must use an external trigger for L515 and each camera must have it's option set (slave) separately.".

MartyG-RealSense, 2021-09-06:
The L515 multiple camera hardware sync works differently to the system on the 400 Series stereo cameras.  On the L515 version, there are no Master cameras, as the external signal generator provides the trigger signal.  So **all** cameras are set to an Inter Cam Sync Mode of '1' (which is hardware sync's Slave mode on L515) in order to be able to respond to the trigger pulse from the external signal generator.

Can you confirm please that you have created **sync cables** for your cameras from Digikey components, as described in Section 1 of the L515 multiple camera paper, and that they are attached to the hardware sync port on the camera that is accessed by removing a small cover piece on the camera's outer casing?

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup#section-1connecting-the-cameras 

[image]

caobingquan6, 2021-09-06:
I haven't really connected the sync cable yet. I just thought about how to involve synchronous trigger pulses if multiple cameras all work at 30Hz without being affected by light interference. As stated in the official website, each is given a high-level pulse of 160ms. But this is only a situation where only three frames of images are collected each, I think this situation is not suitable for my needs. My requirement is that all cameras work at 30Hz in real time and get rid of the influence of light interference. May I ask whether this can be achieved through a synchronous optical cable, and whether pulse triggering can meet this demand.

MartyG-RealSense, 2021-09-06:
As mentioned above, the L515 multiple camera sync system is not the same as the one for the 400 Series cameras (which supports a wireless trigger pulse).  As far as I know, the L515 cameras require sync cabling in order to be synced for reduction of interference.

[image]

If you would prefer to use the cameras without sync, I would recommend either:

1.  Trying the **Short Range** preset mentioned earlier to reduce Laser Power and Receiver Gain, or: 

2.  Spacing the cameras further apart from each other if possible whilst still ensuring that there is some overlap and use Intel's RealSense ROS guide for implementing point cloud capture with multiple cameras on multiple computers.  The guide advises how to stitch together a single merged point cloud from multiple individual clouds if you need to take that further step.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

caobingquan6, 2021-09-06:
First of all, thank you for your answers. I still have three questions.

First of all, if the hardware synchronization is set correctly, can we use a certain time interval for each camera to trigger at the beginning to achieve high-frequency operation at different times, similar to Kinect's multi-device synchronization.

Secondly, if I don’t use hardware synchronization, what should I do about the "Trying the Short Range preset mentioned earlier to reduce Laser Power and Receiver Gain"? Is there a tutorial for this part? (It can best be achieved through ROS)

Finally, the guide you provided (https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines) does not seem to consider the problems caused by light interference , And they are not as far apart as you said.

MartyG-RealSense, 2021-09-06:
My understanding is that the trigger signal's pulse width would be determined by the external signal generator that produces the trigger.  As far as I know you would not be able to generate an individual pulse width for each camera though, but may have some control over the point at which particular cameras trigger (as shown in the diagram below from the L515 multiple camera paper).

[image]

You could test the **Short Range** preset in ROS by accessing the **dynamic reconfigure** interface and selecting the preset from a drop-down menu, as described in https://github.com/IntelRealSense/realsense-ros/issues/1294

The dynamic reconfigure interface can be accessed in the ROS terminal after launch using the command below:

**rosrun rqt_reconfigure rqt_reconfigure**

The ROS point cloud guide was written with the 400 Series cameras in mind as the guide pre-dates the release of the L515 model.  400 Series cameras in multicam arrangements do not interfere with each other.

closed 2021-09-18 by MartyG-RealSense