realsenseai/realsense-ros #2063: /usr/local/lib/librealsense2.so.2.45.0  file does not exist.

liangchen97, 2021-09-06:
hi,
I have installed librealsense v.2.49.0, and realsense-viewer v2.49.0 runs normally,
but when I follow https://github.com/IntelRealSense/realsense-ros installation instructions,
an error is reported after I run catkin_make :

/usr/local/lib/librealsense2.so.2.45.0
 but this file does not exist.
```nvidia@nvidia-desktop:~/catkin_ws1$ catkin_make clean
Base path: /home/nvidia/catkin_ws1
Source space: /home/nvidia/catkin_ws1/src
Build space: /home/nvidia/catkin_ws1/build
Devel space: /home/nvidia/catkin_ws1/devel
Install space: /home/nvidia/catkin_ws1/install
####
#### Running command: "make cmake_check_build_system" in "/home/nvidia/catkin_ws1/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/nvidia/catkin_ws1/devel
-- Using CMAKE_PREFIX_PATH: /home/nvidia/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/nvidia/catkin_ws/devel;/opt/ros/melodic
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: False
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - realsense2_description
-- ~~  - realsense2_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense2_description'
-- ==> add_subdirectory(realsense-ros/realsense2_description)
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /usr/local/lib/cmake/realsense2/realsense2Targets.cmake:92 (message):
  The imported target "realsense2::realsense2" references the file

     "/usr/local/lib/librealsense2.so.2.45.0"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/lib/cmake/realsense2/realsense2Targets.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/local/lib/cmake/realsense2/realsense2Config.cmake:35 (include)
  realsense-ros/realsense2_camera/CMakeLists.txt:44 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/catkin_ws1/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/catkin_ws1/build/CMakeFiles/CMakeError.log".
Makefile:1484: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
nvidia@nvidia-desktop:~/catkin_ws1$ 
```
I did delete /usr/local/lib/librealsense2.so.2.45.0
Because after the latest realsense-ros is compiled, there is an API mismatch problem when running (2.45.0and2.49.0)
But I don't understand why the latest ralsense-ros also needs librealsense2.so.2.45.0
What should I do to run realsense-ros under librealsense2.so.2.49.0?

MartyG-RealSense, 2021-09-06:
hi @liangchen97   Can you confirm please if you used the instruction **catkin_make clean** as the wrapper instructions specify instead of just 'catkin_make', please?

liangchen97, 2021-09-06:
Sorry i didn't make it clear.I use catkin_make clean.

MartyG-RealSense, 2021-09-06:
Thank you for the clarification.  Had you installed SDK 2.45.0 on your computer on a previous occasion before installing 2.49.0 now?

liangchen97, 2021-09-06:
yes ,but i don't need it now.Did I not delete it cleanly?

MartyG-RealSense, 2021-09-06:
Did you install the librealsense SDK with Debian packages?  If so, all librealsense related packages can be uninstalled with the command below.

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

liangchen97, 2021-09-06:
Unfortunately, since I am using nvidia jetson nx, I used the installation script to build the 2.45.0 librealsense SDK before
```
./buildPatchedKernel.sh
./installLibrealsense.sh 
```

MartyG-RealSense, 2021-09-06:
It looks as though the JetsonHacks method of building with **installLibrealsense.sh** installs the Debian packages, so using the method in https://github.com/IntelRealSense/realsense-ros/issues/2063#issuecomment-913565379 to remove all packages containing the word 'realsense' may still work.

https://github.com/JetsonHacksNano/installLibrealsense/blob/master/installLibrealsense.sh

liangchen97, 2021-09-06:
I used this command
```dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge```
 but the problem still exists,   it still links to /usr/local/lib/librealsense2.so.2.45.0 file whitch does not exist.

liangchen97, 2021-09-06:
My problem is very similar to this issue#1674

MartyG-RealSense, 2021-09-06:
The official recommended procedure in the RealSense ROS documentation has the following steps.

1.  Install the librealsense SDK on Jetson using Intel's own Jetson installation guide.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

2.  Install the RealSense ROS wrapper from source code with **Method 2.**

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

You have apparently already successfully achieved step 1 using your own choice of librealsense installation method, as you have said that 2.49.0 is working correctly for you.   And if you are using catkin_make then you are apparently already using Method 2.

There is a complication at this point.  Ideally the RealSense ROS wrapper should be matched with a specific librealsense version that is listed in the wrapper's release notes.  But there is not a ROS wrapper version available for librealsense 2.49.0, as very occasionally a wrapper version is not created for a certain librealsense SDK version.  

At the time of writing this, the latest wrapper version is designed for librealsense 2.48.0.  Using the ROS wrapper designed for 2.48.0 (ROS wrapper version 2.3.1) with 2.49.0 may be possible but not ideal.  The ideal situation would be to have librealsense 2.48.0 installed instead.  Given the complications of installing on Jetson though, it may not be worth the risk of changing librealsense version and causing something to break when it is currently working correctly.

The most straightforward solution to the ROS wrapper installation problem may unfortunately be to just wipe the entire installation (both Ubuntu and librealsense / ROS) and start with a fresh install of the OS that has never had librealsense installed.

liangchen97, 2021-09-07:
Thanks for your reply, I understand, I want to try librealsense 2.48.0 and ROS wrapper version 2.3.1 first, because there are some other important files on the device

liangchen97, 2021-09-07:
I am sorry that I still report the same error under librealsense 2.48.0 version
```
CMake Error at /usr/local/lib/cmake/realsense2/realsense2Targets.cmake:92 (message):
  The imported target "realsense2::realsense2" references the file

     "/usr/local/lib/librealsense2.so.2.45.0"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/lib/cmake/realsense2/realsense2Targets.cmake"
```
Is there any other way besides starting over?

MartyG-RealSense, 2021-09-07:
Aside from https://github.com/IntelRealSense/realsense-ros/issues/1674 that you linked to earlier in this discussion, the problem is also discussed in https://github.com/IntelRealSense/realsense-ros/issues/832

In that discussion, **Doronhi** the RealSense ROS wrapper developer suggests in https://github.com/IntelRealSense/realsense-ros/issues/832#issuecomment-507995602 to update the librealsense packages with **sudo apt update && sudo apt upgrade**

liangchen97, 2021-09-08:
I have solved this problem, it seems to be a problem with this command (or the command corresponding to other versions)
```
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1
```
When i use
```git checkout version number```
After that, there will be no errors

MartyG-RealSense, 2021-09-08:
Great news @liangchen97 that you resolved your problem - thanks for the update!

closed 2021-09-14 by MartyG-RealSense

MartyG-RealSense, 2023-03-22:
Hi @Sashraa   If you are attempting to install librealsense from source code on a Jetson board then the instructions for doing so at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 may be helpful.

Alternatively, you could use the **installLibrealsense.sh** build script at the JetsonHacks website to build the SDK on Jetson from packages instead of source code

https://github.com/jetsonhacks/installRealSenseSDK#installlibrealsensesh

MartyG-RealSense, 2023-03-22:
Okay, let's try installing from a build script with the **libuvc backend** installation method at the link below.  That method tends to work well with Pi devices.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Sashraa, 2023-03-22:
make: *** [Makefile:149: all] Error 2
this is the error for  ./libuvc_installation.sh
only 55% was built..

MartyG-RealSense, 2023-03-22:
If the libuvc backend method does not work then there is a more modern procedure for achieving the same goal that can have a greater chance of success, though it is a little more complicated.  Instructions for building the SDK with the RSUSB backend using CMake can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692