realsenseai/realsense-ros #2064: Failed to load nodelet

naoya-kumagai, 2021-09-06:
I am installing the ROS Wrapper with the following commands. 
 
      sudo apt-get install ros-$ROS_DISTRO-realsense2-camera
Then, 

       roslaunch realsense2_camera rs_camera.launch
However, this yields the following error. 
[FATAL] [1630939742.801402600]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'

My kernel version and Ubuntu version are as follows. 
Linux 5.4.72-microsoft-standard-WSL2 x86_64
Ubuntu 20.04.3 LTS

MartyG-RealSense, 2021-09-06:
Hi @naoya-kumagai   Are you installing the ROS wrapper in Linux within Windows using WSL2, please?  If you are then WSL2 does not have support for USB devices.  To use the camera with the ROS wrapper, the Windows version of the wrapper should be installed, or the Ubuntu version of the wrapper installed within a normal Ubuntu OS installation.

If you are using a Windows computer and want to install on Ubuntu but do not have a second computer to put Ubuntu on, you could consider setting up a **dual boot** arrangement on your computer so that you can choose whether to boot into Windows or Linux.

naoya-kumagai, 2021-09-06:
@MartyG-RealSense 
I see, I was not aware of this constraint of WSL2.
I am trying with VirtualBox for now, are there any problems with this approach?

MartyG-RealSense, 2021-09-06:
Intel recommends use of **VMWare Workstation** if a virtual machine needs to be used, in order to provide USB 3 emulation.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation

naoya-kumagai, 2021-09-07:
@MartyG-RealSense 
Thank you for the suggestion! VMWare is working well recognizing the camera.

closed 2021-09-07 by naoya-kumagai

MartyG-RealSense, 2021-09-07:
That's excellent news that VMWare is working for you - thanks for the update   :)