realsenseai/realsense-ros #2070: SLAM with D435i tutorial: PO

naoya-kumagai, 2021-09-08:
I am following the SLAM tutorial. 
The pointcloud map generated by 

    rosrun pcl_ros pointcloud_to_pcd input:=/rtabmap/cloud_map
is not dense. What could be a fix for this? 

[image]

MartyG-RealSense, 2021-09-09:
Hi @naoya-kumagai   I can see color shading in the points, which suggests to me that the color image is being mapped to the point cloud like it should be in the D435i SLAM guide.

As a starting point, you could confirm that the pointcloud is set as unordered (is_dense = true) by including the command **ordered_pc:=false** in your roslaunch instruction.  The point cloud should be unordered by default anyway but it is worth checking to confirm that an ordered point cloud is not being caused by something in the project.

naoya-kumagai, 2021-09-09:
This doesn't seem to make a difference...

MartyG-RealSense, 2021-09-09:
Let's check through the guide steps again.  Can you confirm please that you have realsense2_camera, imu_filter_madgwick, rtabmap_ros and robot_localization installed, as described in https://github.com/IntelRealSense/realsense-ros/issues/2067#issuecomment-915452219 

Does your **Frames** list match the one in the guide, with only 'camera_link' and 'map' ticked?

[image]

And is **Image Topic** set to /camera/color/image/raw ?

closed 2021-09-21 by MartyG-RealSense