realsenseai/realsense-ros #2071: Control depth color stream distance

ranjitkathiriya, 2021-09-09:
Currently, I am able to git Red and yellow colors on my nearest object, It is possible in ros to extend this color to more distance?

[image]

MartyG-RealSense, 2021-09-09:
Hi @ranjitkathiriya  Because the near-range detail at the sides of the image is blue, that makes me think that the center detail may be returning incorrect depth values, and so the colorization over distance does not need to be adjusted.  Would it be possible please to provide an RGB image of the scene that the camera is observing to see if the observed object has features that may be causing bad depth readings, please?

ranjitkathiriya, 2021-09-09:
[image]

Actually, these are my points, and from the color depth stream, I am finding the center of the cow teat. Because of the green color, I can find far teat, but if the color had been red or yellow up to a certain distance, it would have been greater.
Thanks.

MartyG-RealSense, 2021-09-09:
Given how well lit the scene is, you may be able to achieve improved accuracy if you turn off the IR emitter and the dot pattern that is being projected onto the udders and the underbelly section nearest to the camera.  When a scene is well lit, the camera can use the ambient light that is present to analyze surfaces for depth detail instead of analyzing the projected dots cast onto the surface of the object.

```
<rosparam>
/camera/stereo_module/emitter_enabled: false
</rosparam>
```

In the librealsense SDK the color grading over distance scale can be altered by changing the **depth unit scale** value of the camera.  Changing the scale from its default of **0.001** to a value of **0.0001** is a useful method for inproving the depth image when performing close-range sensing.  This is illustrated in https://github.com/IntelRealSense/librealsense/issues/8228 but **Doronhi** the RealSense ROS wrapper developer explains in https://github.com/IntelRealSense/realsense-ros/issues/277#issuecomment-525676873 that the depth unit scale cannot be changed in the ROS wrapper.

ranjitkathiriya, 2021-09-09:
> Given how well lit the scene is, you may be able to achieve improved accuracy if you turn off the IR emitter and the dot pattern that is being projected onto the udders

Based on your point, Can you please explain to me if I turn this thing off I will get point cloud or not? Because For identifying 3D coordinates I do require a point cloud.

MartyG-RealSense, 2021-09-09:
Yes, you can obtain a point cloud if the IR emitter is turned off.  It is a separate component inside the camera from the sensors that projects a semi-random dot pattern onto surfaces in the scene.  It is useful for analyzing surfaces with no texture or low texture detail, as the camera can use the dots as a texture source to analyze for depth.  In a dimly lit scene, turning off the pattern may significantly reduce the detail on the depth image.  If the surfaces are well lit though, like in your image, you may not need the dot pattern.  

Point 5 of the section of Intel's camera tuning guide linked to below estimates that **RMS error** (error over distance) can be reduced by around 30% if the dot pattern is turned off.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-verify-performance-regularly-on-a-flat-wall-or-target

ranjitkathiriya, 2021-09-09:
Thanks for providing solution, 

```
<rosparam>
/camera/stereo_module/emitter_enabled: false
</rosparam>
```

How can I off this can you please! explain about that?

MartyG-RealSense, 2021-09-09:
Do you mean how you can implement it?  If so then you would edit the code into your launch file by opening the launch file with a text editor, such as Ubuntu's **Gedit**, and insert the code.

 If you wanted to test turning the IR emitter off with **dynamic_reconfigure** without having to edit the test file then I believe that the command to input into the ROS terminal would be:

**rosrun dynamic_reconfigure dynparam set /camera/stereo_module emitter_enabled 0**

closed 2021-09-10 by ranjitkathiriya