realsenseai/realsense-ros #2073: Skipped loading plugin with error: XML Document 'c:\opt\ros\noetic\x64\share\test_nodelet/test_nodelet.xml' has no Root Element

MakerWangHQ, 2021-09-10:
**I have build realsense-ros successfully, but when I run 'roslaunch realsense2_camera rs_camera.launch' see the error below.**
**OS**:Windows 10  19042.1165
**ros**:noetic
**librealsense**: 2.48
**realsense-ros**: 2.3.1

C:\catkin_ws>call devel\local_setup.bat

C:\catkin_ws>roslaunch realsense2_camera rs_camera.launch
... logging to C:\Users\IME\.ros\log\5b9854b5-1240-11ec-b91e-8cec4b7d9227\roslaunch-DESKTOP-SI41QI1-11872.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://127.0.0.1:56310/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path:
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename:
 * /camera/realsense2_camera/serial_no:
 * /camera/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method:
 * /camera/realsense2_camera/usb_port_id:
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [13292]
ROS_MASTER_URI=http://127.0.0.1:11311
setting /run_id to 5b9854b5-1240-11ec-b91e-8cec4b7d9227
process[rosout-1]: started with pid [1444]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [6304]
process[camera/realsense2_camera-3]: started with pid [8604]
[ERROR] [1631282822.314210600]: Skipped loading plugin with error: XML Document 'c:\opt\ros\noetic\x64\share\test_nodelet/test_nodelet.xml' has no Root Element. This likely means the XML is malformed or missing..
[ERROR] [1631282822.317347500]: Skipped loading plugin with error: XML Document 'c:\opt\ros\noetic\x64\share\test_nodelet_topic_tools/test/test_nodelets.xml' has no Root Element. This likely means the XML is malformed or missing..
[ INFO] [1631282822.329202700]: Initializing nodelet with 8 worker threads.
[ERROR] [1631282824.096719200]: Skipped loading plugin with error: XML Document 'c:\opt\ros\noetic\x64\share\test_nodelet/test_nodelet.xml' has no Root Element. This likely means the XML is malformed or missing..
[ERROR] [1631282824.100378700]: Skipped loading plugin with error: XML Document 'c:\opt\ros\noetic\x64\share\test_nodelet_topic_tools/test/test_nodelets.xml' has no Root Element. This likely means the XML is malformed or missing..
[ERROR] [1631282824.111819000]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Failed to load library C:/catkin_ws/devel\bin/realsense2_camera.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = C:/catkin_ws/devel\bin/realsense2_camera.dll)
[ERROR] [1631282824.112372500]: The error before refreshing the cache was: Failed to load library C:/catkin_ws/devel\bin/realsense2_camera.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = C:/catkin_ws/devel\bin/realsense2_camera.dll)
[FATAL] [1631282824.113226500]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[31m[camera/realsense2_camera-3] process has died [pid 8604, exit code 4294967295, cmd c:\opt\ros\noetic\x64\lib\nodelet\nodelet.exe load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=C:\Users\IME\.ros\log\5b9854b5-1240-11ec-b91e-8cec4b7d9227\camera-realsense2_camera-3.log].
log file: C:\Users\IME\.ros\log\5b9854b5-1240-11ec-b91e-8cec4b7d9227\camera-realsense2_camera-3*.log[0m

MartyG-RealSense, 2021-09-10:
Hi @MakerWangHQ   I note that just before performing roslaunch, you used the instruction **devel\local_setup.bat**.  The installation documentation of the RealSense ROS wrapper uses **devel\setup.bat**.  As the .bat file names are different, may I begin this investigation of your problem please by asking if you created your own custom .bat file?  Thanks!

MakerWangHQ, 2021-09-11:
I have used the **devel\setup.bat**, and the result is same as use devel\local_setup.bat. I haven't created my own custom .bat file.
I add a situation that there isn't a **'test' directory** in the **'c:\opt\ros\noetic\x64\share\test_nodelet' directory** of computer.

MartyG-RealSense, 2021-09-11:
Bearing in mind your other installation case at https://github.com/IntelRealSense/realsense-ros/issues/2074 it may be worth returning to this question after you have tried a ROS wrapper installation on an installation of the 2.42.0 RealSense SDK made with the installer program to see whether the problem still occurs.

MakerWangHQ, 2021-09-11:
> Bearing in mind your other installation case at #2074 it may be worth returning to this question after you have tried a ROS wrapper build on an installation of the 2.42.0 RealSense SDK made with the installer program to see whether the problem still occurs.

Thank you for your reply!
Only can I install librealsense2 v 2.34.0 using **vcpkg** tools. It seems that there only one version in the vcpkg repository.
I haven't install the librealsense2 v 2.42.0 successfully.

MakerWangHQ, 2021-09-12:
**It have been solved!**
download and **copy** the file **https://github.com/ros/nodelet_core/tree/noetic-devel/test_nodelet/test_nodelet.xml** into **C:\opt\ros\noetic\x64\share\test_nodelet**
[image]

test_nodelset is a package of ros, and ros package can be searched in the [ros website](https://index.ros.org/search/?term=test_nodelet).

closed 2021-09-12 by MakerWangHQ

MartyG-RealSense, 2021-09-12:
It's great that you achieved a solution to this issue, @MakerWangHQ   :)