realsenseai/realsense-ros #2074: How to install new version of realsense2, I use command 'vcpkg install librealsense' get realsense2 v 2.34.0

MakerWangHQ, 2021-09-11:
How to install new version of realsense2, I use command 'vcpkg install librealsense' get realsense2 v 2.34.0.
It is likely that the realsense2 **v 2.42.0** can be installed by using **'vcpkg install'**
[ [realsense2] Update version to v 2.42.0 #16411](https://github.com/microsoft/vcpkg/pull/16411)

MartyG-RealSense, 2021-09-11:
Hi @MakerWangHQ   I would recommend not installing RealSense SDK version **2.34.0** if possible, as this version had an issue that continuously generated timing errors for some RealSense users.

Are you using Windows or Linux, please?  If you are using Windows then I would also recommend installing the SDK with the automated **Intel.RealSense.SDK-WIN10** installer file available on the SDK's **Releases** page instead of using vcpkg.  This file can be found in the 'Assets' file list at the bottom of the information listing for each version.  

Once it is downloaded and run, the SDK can be installed on Windows quickly and easily just by clicking 'Next' a few times without making any changes to the installer's defualt settings.

https://github.com/IntelRealSense/librealsense/releases

If you require version 2.42.0 then it can be found here:

https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0

[image]

MakerWangHQ, 2021-09-11:
Thanks for your reply. Now I want to use realsense-ros on Windows 10 and I follow the [official tutorial](https://github.com/IntelRealSense/realsense-ros). Can I use realsense-ros if I install the SDK with the automated Intel.RealSense.SDK-WIN10 installer file available on the SDK's Releases page instead of using vcpkg

MartyG-RealSense, 2021-09-11:
The RealSense ROS wrapper is not included in the installer program.  You can install librealsense that way, and then afterwards you can build the ROS wrapper from source code using **Method 2** of the official guide.  When using Method 2, the correct order of installation is the librealsense SDK first and then the ROS wrapper secondly after the SDK has been installed.

https://github.com/IntelRealSense/realsense-ros#method-2-the-realsense-distribution

It is also important to remember that a particular RealSense ROS wrapper version should be closely matched with a specific version of librealsense listed in the release notes for that wrapper version.  So if you are using ROS1 (Kinetic, Melodic or Noetic) and were installing librealsense 2.42.0 then you would need to build ROS wrapper version **2.2.22** from source.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.22

MakerWangHQ, 2021-09-11:
> The RealSense ROS wrapper is not included in the installer program. You can install librealsense that way, and then afterwards you can build the ROS wrapper from source code using **Method 2** of [...]

Which realsense2 version can be installed using command 'vcpkg install librealsense' ?  I hope install newerful version of realsense, but I can only get the realsense2 v 2.34.0 . Is there some solution to install librealsense 2.42.0 by using **vcpkg** tools.

MartyG-RealSense, 2021-09-11:
As far as I know, when using vcpkg you do not have a choice in which particular SDK version it installs.  If you do need to use that installation method because the automated installer will not work for you, https://github.com/IntelRealSense/realsense-ros/issues/1509#issuecomment-753930031 may provide useful guidance.

MakerWangHQ, 2021-09-12:
> As far as I know, when using vcpkg you do not have a choice in which particular SDK version it installs. If you do need to use that installation method because the automated librealsense installer will not work for you, [#1509 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/1509#issuecomment-753930031) and [IntelRealSense/librealsense#6462 (comment)](https://github.com/IntelRealSense/librealsense/issues/6462#issuecomment-641450420) may provide useful guidance.

many thanks, I have successfully installed the realsense-ros, the ros version that I installed is noetic and its default vcpkg only can install realsense2 v 2.34.0. 
[ #1509 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/1509) is helpful. The newest vcpkg in https://github.com/microsoft/vcpkg, can install v2.42.0. and I installed realsense v 2.42.0. `vcpkg install realsense2:x64-windows`
[image]

closed 2021-09-12 by MakerWangHQ

MartyG-RealSense, 2021-09-12:
That's great to hear, @MakerWangHQ - thanks very much for the update about your success   :)