realsenseai/realsense-ros #2075: Wrong frame and point cloud data in using realsense-ros

Tom127, 2021-09-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Operating System & Version |   { Linux (Ubuntu 16.04)   |                        | 
| Language                            |  {C++/opencv/pcl }                          | 

Hi,  I am using  [realsense-ros](https://github.com/IntelRealSense/realsense-ros) driver, and I ran `roslaunch realsense2_camera demo_pointcloud.launch` in the terminal and generated point cloud, which rostopic name was `/camera/depth/color/points`, and then I subscribed to this rostopic with the following code:

```
#include <ros/ros.h>
#include <iostream>
#include <sstream>
#include <string>
#include <sensor_msgs/PointCloud2.h>
using namespace std;

void process_cb (const sensor_msgs::PointCloud2ConstPtr& input)
{
        std::cout << "frame id========" << input->header.frame_id << std::endl;
}

int main (int argc, char** argv)
  {    
	// Initialize ROS
	ros::init (argc, argv, "detector");
	ros::NodeHandle nh;

	ros::Subscriber sub_ = nh.subscribe("/camera/depth/color/points", 1, process_cb);

	while(ros::ok())
       {
          ros::spinOnce();
       }
  }
```

But the default frame_id of this topic is `camera_depth_optical_frame`, and its point data(x,y,z) is wrong for the real world, I want to subscribe to the data of this topic in the `camera_link` frame, which is right for the real world. How could I do?

MartyG-RealSense, 2021-09-13:
Hi @Tom127  The **demon_pointcloud** launch file has not been updated in over 3 years at the time of writing this.  I would recommend instead following the ROS wrapper documentation's pointcloud tutorial based on the modern **rs_camera.launch** and setting 'Fixed Frame' in RViz to **camera_link** as illustrated in the tutorial.

https://github.com/IntelRealSense/realsense-ros#point-cloud

Tom127, 2021-09-13:
> Hi @Tom127 The **demo_pointcloud** launch file has not been updated in over 3 years at the time of writing this. I would recommend instead following the ROS wrapper documentation's pointcloud tutorial based on the modern **rs_camera.launch** (equipped with a pointcloud filter in the roslaunch instruction) and setting 'Fixed Frame' in RViz to **camera_link** as illustrated in the tutorial.
> 
> https://github.com/IntelRealSense/realsense-ros#point-cloud

@MartyG-RealSense @doronhi Thanks for your reply, I tried `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
`, but the frame_id of this rostopic(`/camera/depth/color/points`) still `camera_depth_optical_frame`, and its point data(x,y,z) is wrong for the real world, I want to subscribe to the data of this topic in the camera_link frame, which is right for the real world.

Tom127, 2021-09-13:
I am not to display the point cloud in RViz. I want to use the point cloud data in the  rostopic `/camera/depth/color/points`, but the point cloud data inside now is inconsistent with the actual data of the target, because it is in the `camera_depth_optical_frame`, but I want to use the point_cloud data in the `camera_link`.

MartyG-RealSense, 2021-09-13:
Some people define a custom value for frame_id as a rosparam in the launch file with **param name="frame_id"**.  For example:

`<param name="frame_id" type="string" value="$(arg frame_id)"/>`

https://github.com/IntelRealSense/realsense-ros/issues/1803 has an example of defining such settings in an rtabmap_ros launch file.

[image]

[image]

Tom127, 2021-09-14:
@MartyG-RealSense sorry, I didn't get what you mean. I am wondering if I can modify the realsense-ros driver so that I can publish the data under the `camera_link` frame to this rostopic `/camera/depth/color/points`. If so, what should I do? This is very convenient to use. The data under the `camera_depth_optical_frame` is meaningless to us. Now the driver has obviously added the tf transformation from `camera_link` frame to `camera_depth_optical_frame` frame.

point cloud under `camera_link` frame
[image]

point cloud under `camera_depth_optical_frame`
[image]

MartyG-RealSense, 2021-09-14:
I will highlight your question to **Doronhi** the RealSense ROS wrapper developer.  Thanks very much for your patience.

doronhi, 2021-09-19:
Hi @Tom127 ,
The pointcloud is calculated inside the librealsense2 library using the native coordinate system of the camera, which is the "camera_depth_optical_frame" coordinate system. The wrapper does not transform the points into a new coordinate system. 
It gives the information that allows you to transform the points to the coordinate system of your choice in the form of TF and frame_id - just as you mentioned.
With that mechanism available in the ROS system I don't see much use in changing the API to allow a user to define a specific coordinate system to the pointcloud and in increasing the CPU usage and code complexity by transforming the points inside the wrapper.

closed 2021-10-03 by MartyG-RealSense