realsenseai/realsense-ros #2077: RS Camera Image Raw stops publishing when launching with pointcloud

Dverstandig, 2021-09-13:
Hello,

I am running into an issue when trying to launch the realsense camera using the realsense-ros wrapper.

When launching with the pointcloud set to true, I get the following error, and image_raw stops publishing:

(handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
(sensor.cpp:517) acquire_power failed: failed to set power state

However, the node is able to launch fine without the pointcloud.

I am on version 2.3.1 of the realsense ros wrapper and 2.48 of librealsense. 

Any help would be appreciated!

MartyG-RealSense, 2021-09-13:
Hi @Dverstandig  As a first step in investigating your problem, could you test whether including **initial_reset:=true** in your roslaunch instruction to reset the camera at launch makes a difference, please.

For example:

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud initial_reset:=true**

Dverstandig, 2021-09-13:
Thanks for the quick reply. Even with initial reset, I get this error:

 13/09 15:23:40,743 ERROR [546375188864] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
[ WARN] [1631546620.745432174]: Device 1/1 failed with exception: failed to set power state
 13/09 15:23:40,743 ERROR [546391974272] (sensor.cpp:517) acquire_power failed: failed to set power state
 13/09 15:23:40,745 WARNING [546391974272] (rs.cpp:306) null pointer passed for argument "device"

As a side note, I have been launching the pointcloud by setting the enable_pointcloud param to true. Should I be launching it using filters instead?

MartyG-RealSense, 2021-09-13:
enable_pointcloud was deprecated as far back as the previous ROS wrapper **realsense_camera** for the original generation of RealSense cameras.  The documentation says simply that it was set to false by default because of "performance issues" and that using **filters:=pointcloud** is recommended instead.  enable_pointcloud should still work though, even though it is not the recommended method of enabling a pointcloud.

Do you experience the error if you use a custom camera configuration to set depth and color to a low resolution of 640x480 and 30 FPS for both using the roslaunch instruction below?

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30 filters:=pointcloud**

Dverstandig, 2021-09-14:
I do not seem to have this problem when running just the realsense2_camera launch file. 
The problem occurs when I launch this with the rest of my stack, and only if the pointcloud is set. Could it be a power problem (I am running this on a jetson)?

Is this something you've encountered when trying to use the realsense as part of a bigger project?

MartyG-RealSense, 2021-09-14:
Are you using a Jetson Nano, please?  If you are then Intel recommend in their Jetson documentation to enable the **barrel jack** power connector to provide extra power.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

Dverstandig, 2021-09-14:
I am using a jetson xavier nx

MartyG-RealSense, 2021-09-14:
I recalled that there have been issues with Xavier NX and pointclouds, as described in https://github.com/IntelRealSense/realsense-ros/issues/1967

In that discussion, a RealSense user in https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-876615270 said that they were able to get it working successfully by using older SDK / wrapper versions.

closed 2021-09-17 by Dverstandig

MartyG-RealSense, 2021-09-22:
Hi @AswinkarthikeyenAK  I would recommend avoiding using librealsense SDK 2.41.0 if possible, as it had an issue with V4L that was corrected in version 2.42.0.

https://github.com/IntelRealSense/librealsense/pull/8234

AswinkarthikeyenAK, 2021-09-22:
Hi @MartyG-RealSense ,
Thanks for that. 
I am using v2.42.0 and Ros version v2.2.22.

I notice that the ros node stops automatically after few seconds. Is there any way to fix that. Thanks

```
 roslaunch realsense2_camera rs_camera.launch
... logging to /home/.ros/log/c0ee3da2-1ba3-11ec-9c09-7412b3bb7b4b/roslaunch-desktop-6255.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.2:38851/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
[... 81 lines cut ...]
[ INFO] [1632311495.775908064]: insert Depth to Stereo Module
[ INFO] [1632311495.776074976]: insert Color to RGB Camera
 22/09 12:51:35,884 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 12:51:35,935 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 12:51:35,985 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 12:51:36,036 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1632311496.087477088]: SELECTED BASE:Depth, 0
[ INFO] [1632311496.120636832]: RealSense Node Is Up!
[ WARN] [1632311496.148542048]: 
 22/09 12:51:36,149 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 12:51:36,200 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 12:51:36,251 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 12:51:36,302 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 12:51:36,412 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 12:51:36,463 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 12:51:36,513 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 22/09 12:51:38,824 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 12:51:43,879 WARNING [547846029696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/.ros/log/c0ee3da2-1ba3-11ec-9c09-7412b3bb7b4b/camera-realsense2_camera-2*.log
```
[roslaunch-lams-desktop-6255.log](https://github.com/IntelRealSense/realsense-ros/files/7210451/roslaunch-lams-desktop-6255.log)

MartyG-RealSense, 2021-09-22:
@AswinkarthikeyenAK   Does stability improve if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch?

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**