realsenseai/realsense-ros #2078: Issue on starting demo of rs_camera.launch

GitNzino, 2021-09-14:
Hi everyone,

I had some issue on starting the demo examples of realsense camera. I followed all the instruction that are on the README file and i don't understand where is the error.

Thank you in advice, have a nice day!
Lorenzo

[image]

MartyG-RealSense, 2021-09-14:
Hi @GitNzino  The **_ZN20ddynamic_reconfigure** error may occur if there is more than one instance of the **ddynamic_reconfigure** component installed on your computer, as described in https://github.com/IntelRealSense/realsense-ros/issues/1922#issuecomment-856491772

Some RealSense users resolved the problem by doing a complete wipe and clean reinstall of their system.

Another took the approach of uninstalling ddynamic_reconfigure with the instruction **uninstall apt remove ddynamic*** and then reinstalling ddynamic_reconfigure

A quick way to install it for Melodic is with the instruction **sudo apt-get install ros-melodic-ddynamic-reconfigure**

GitNzino, 2021-09-14:
Thank you, after checking i have only one ddynamic_recongifure installed on my pc.
What should i do for using the ros package?

thank you for help!

MartyG-RealSense, 2021-09-14:
https://github.com/IntelRealSense/realsense-ros/issues/838 is the main reference for this problem.  As mentioned earlier, some RealSense users solved it with a complete system wipe and re-installation, whilst others suggested methods that did not involve a wipe. 

Could you tell me about the method that you used to install the librealsense SDK and the RealSense ROS wrapper that led to the problem that you are experiencing, please?  Thanks!

GitNzino, 2021-09-15:
I installed all the libraries with `sudo apt install` only the `realsense-ros`  package is installed in catkin workspace.

In your opinion is it possible to use the python code in python and use it in ros?

Anyway I continue to investigate the other methods mentioned in your comment with the hope to solve the problem.

MartyG-RealSense, 2021-09-15:
You can access Python from ROS by using a Python **ROS node** script, as discussed in https://github.com/IntelRealSense/librealsense/issues/6367  

A system wipe is a wipe and fresh reinstall of everything on the computer - Ubuntu, librealsense and ROS.

Good luck with your investigations and please do update here when you have new results.

GitNzino, 2021-09-16:
Thank you for the example reported.

The following question are written in order to try to use some realsense functionality without doing a system wipe (because it could be take away time in rebuilding all the other scripts and procedure):

-  Is it possible to use in ROS the [example written for python](https://github.com/IntelRealSense/librealsense/blob/0fa6c4f1a000059cbb80a09ee5efa54b281b22c0/wrappers/python/examples/align-depth2color.py), so using the librealsense API in python to dialogue with the camera and the write the acquired information from the camera to a subscriber node for other purpose?
- It is possible taking a [ROS example](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/scripts/show_center_depth.py) putting in a new personal package and use them without  `ddynamic_reconfigure`?

In the end i tried most of the suggested action to start the rs_camera.launch without doing a system wipe but for the moment nothing seem works properly, i hope that there is a method to remedy this issue.

Thank you very much for the help and support, hope a nice day for you!

MartyG-RealSense, 2021-09-16:
I am not aware of an example of **align-depth2color.py** being launched from ROS.  As the 'ROS example' script **show_center_depth.py** is a pyrealsense2 script though, this suggests that align-depth2color.py (also a pyrealsense2 script) may be launchable from ROS.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/scripts/show_center_depth.py#L8

**Doronhi** the RealSense ROS wrapper developer provides more information about how show_center_depth.py works in https://github.com/IntelRealSense/realsense-ros/issues/1086#issuecomment-718477110

However, you do not necessarily need a Python script to perform alignment.  If you add the term **align_depth:=true** to your roslaunch instruction then it publishes a pair of depth-to-color topics called **/camera/aligned_depth_to_color/image_raw** and **/camera/aligned_depth_to_color/camera_info**.  A pointcloud, if enabled, will be built based on the aligned_depth_to_color image.

In regard to your second question: if you are askng if a ROS Python example script could be converted for non-ROS use in Python then it may be easier to use a pyrealsense2 script that performs a similar function.  The Python wrapper has a list of example programs at the link below, and many others can be found on the librealsense forum.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples

GitNzino, 2021-09-17:
Thank you for help again!

What I want to do is using the python wrapper example in ROS to overcome the problem of ddynamic_reconfigure, but I'm not sure that it is easy or possible.

Alternatively to use the` /camera/aligned_depth_to_color/image_raw` and `/camera/aligned_depth_to_color/camera_info` is mandatory to have a workng ddynamic _reconfigure?

Sorry for the question but if is possible i would like to avoid a system wipe because this camera step is the last step of an entire system that involves many programs, library and scripts already working.

MartyG-RealSense, 2021-09-17:
As far as I am aware, **ddynamic_reconfigure** is a necessary installation requirement for the ROS1 wrapper when installing it from source with **Method 2**.  

It is not a described part though of the **Method 1** installation (which installs librealsense and the wrapper together from packages).

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

GitNzino, 2021-09-20:
Thank you!

This week I'll try to do the system wipe, hope that all go right.
After them is needed to do `sudo apt install ros-melodic-ddynamic_reconfigure && sudo apt install ros-melodic-realsense2-*` and then import the raw and depth frames using the appropriate node. I think it comes by itself that i need to launch the `roslaunch realsense2_camera rs_camera.launch` before I start the ros script that reads the node output.

Can you confirm?

MartyG-RealSense, 2021-09-20:
Processes that depend upon the output of the RealSense ROS wrapper (such as the camera data topics that it publishes) do usually have to be launched after the ROS wrapper launch has completed.

GitNzino, 2021-09-20:
Sure, understand thank you!

Do you prefer that I close the issue and if in few day I have new similar question I will open another issue or is better that i continue writing there?

MartyG-RealSense, 2021-09-20:
If the new question is similar to this one then it's fine to keep this case open and comment here.  If the question is a different subject then you can create a new issue for it.  Thanks!

GitNzino, 2021-09-22:
Hi 
I have just do the system wipe, but a similar error still occur. The error reported by terminal is the following:
[image]

What do you suggest to do now?
Thank you for help

MartyG-RealSense, 2021-09-22:
It looks as though it is now a different error to the one at the start of this case (which was **ZN20ddynamic_reconfigure** and is now **_ZN2cv3MatC1Ev**).  My research indicates that it seems to be particularly related to **OpenCV**.  Are you making use of OpenCV in your application, please?

GitNzino, 2021-09-22:
I'm really pleased that now `ddynamic_reconfigure` works! 

Now I'm trying to execute `roslaunch realsense2_camera rs_camera.launch` that is the script produced by you. 

If is required to install OpenCV can you tell me what is the required package?
Thank you

MartyG-RealSense, 2021-09-22:
OpenCV is not part of the installation process for the RealSense ROS wrapper, so I'm not sure why an error typically associated with OpenCV would occur if you have done a wipe and just installed Ubuntu, librealsense, ROS and the ROS wrapoer.

When OpenCV is used with librealsense, a minimum of OpenCV 3.4 is recommended.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv#opencv-samples-for-intel-realsense-cameras

All the guides that I have seen for installing OpenCV on Ubuntu involve a requirement for Python.

https://docs.opencv.org/master/d2/de6/tutorial_py_setup_in_ubuntu.html

GitNzino, 2021-09-22:
Thank you for good answer, but there is already installed OpenCV in this PC as you can see in the picture below:

[image]

Maybe is necessary to declare some `export PATH ` in the bash or something other?

MartyG-RealSense, 2021-09-23:
Could you describe the installation steps that you took after wiping your computer's drive, please?  For example:

1.  Installed Ubuntu
2.  Installed librealsense
3.  Installed ROS
4.  Installed the RealSense ROS wrapper

GitNzino, 2021-09-23:
Yes sure: 

1. Installed Ubuntu
2. Installed librealsense (via apt)
3. Installed ROS
4. Installed the RealSense ROS wrapper (via apt)
5. Installed ddynamic_reconfigure (via apt)
6. Installed OpenCV (via apt)
7. Now i'm trying to install OpenCV from source

MartyG-RealSense, 2021-09-23:
Is OpenCV a required component of the project that you are working on, please?

Have you tested the RealSense ROS wrapper after completing stage 5, before adding OpenCV?

GitNzino, 2021-09-23:
If I'm not in wrong i have understand that for obtain the node from the realsense camera for is to launch the `roslaunch realsense2_camera rs_camera.launch` and if i digit on the terminal `sudo apt install ros-melodic-realsense2-camera` appears the following result.

```
lorenzo@ubuntu-18-04:~$ sudo apt install ros-melodic-realsense2-camera-*
[sudo] password for lorenzo: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'ros-melodic-realsense2-camera-dbgsym' for glob 'ros-melodic-realsense2-camera-*'
The following additional packages will be installed:
  python-opencv ros-melodic-cv-bridge ros-melodic-realsense2-camera
The following NEW packages will be installed:
  python-opencv ros-melodic-cv-bridge ros-melodic-realsense2-camera
  ros-melodic-realsense2-camera-dbgsym
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 4.462 kB of archives.
After this operation, 8.363 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
```
So i suppose that OpenCV is needed by realsense package to works properly. 

For these reason it is impossible to test Realsense ROS packages without OpenCV.

Thank you for help.

MartyG-RealSense, 2021-09-23:
Do you get the same set of messages if you use the installation instruction below:

**sudo apt-get install ros-melodic-realsense2-camera**

GitNzino, 2021-09-24:
Thank you now works!

I think that `sudo apt-get install ros-melodic-realsense2-camera` + a clean `catkin_make` of all the workspace fixed  the previous bug.

Thank you for your kindness, hope you a nice day!
Lorenzo

closed 2021-09-24 by GitNzino

MartyG-RealSense, 2021-09-24:
You are very welcome, Lorenzo.  That's great to hear that you were successful.  :)