realsenseai/realsense-ros #2079: Can we apply "opensource_tracking.launch"  to SLAM with D455?

Yoshihito-M, 2021-09-15:
Hi every one.
 I'm trying SLAM with D455 camera following the tutorial [https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i].
However, I get the errors when I run the command,"roslaunch realsense2_camera opensource_tracking.launch".

rgbd.rviz appears but it displays no PointCloud and the error message on "Grid" Tab as follows: 
"Fixed Frame: Fixed frame [map] does not exist .

I'm trying to fix the errors by using past issue "https://github.com/IntelRealSense/realsense-ros/issues/1153", but I can't solve these issues at the moment.
I think that the points of difference between my issue and above past issue are 
1) I install VMware workstation 16 Player to use ubuntu 16.04 as guest OS,
2) I use D455 camera not  D435.

Can you help me fix these issues?
Thanks for your help.

MartyG-RealSense, 2021-09-15:
Hi @Yoshihito-M   You should be able to use the D435i SLAM guide with your D455.

Could you confirm please that the 'Fixed Frame' option in RViz hs been set to **map**, as suggested in https://github.com/IntelRealSense/realsense-ros/issues/1153

[image]

Does it also make any difference to the outcome if you enable the IMU topics by adding **enable_gyro:=true enable_accel:=true** to the end of your opensource_tracking roslaunch instruction?  For example:

**roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true**

Yoshihito-M, 2021-09-16:
HI, @MartyG-RealSense 

I appreciate your advice to my issues.
Your advice is very helpful for me and I can fix these issues.
Thanks!

closed 2021-09-16 by Yoshihito-M

MartyG-RealSense, 2021-09-16:
You are very welcome, @Yoshihito-M - thanks for the update!