realsenseai/realsense-ros #2080: IMU measurement freezes and has unstable frequency with ros driver of D455i

YuxuanSnow, 2021-09-15:
Hi Realsense developers,

i use the launch file to launch the d455 camera. The launch file I used can be found at the end of the issue.

I have two questions:
1. The imu measurement freezes sometimes for no reason. As you can see, the frequency of the topic "/imu" is very unstable and sometimes even no messages from this topic. I want to have the topic of IMU measurement with stable frequency (which is should have), how can I do?
[image]

2. The topic "aligned_depth_to_color/image_raw" and "depth/image_rect_raw" have no clear difference in my observation. If I want to read the depth of the pixel in the color image, should I use "aligned_depth_to_color" topic? Or it is fine whatever I use ?

Thank you for your support!

Yuxuan

Lauchfile:
```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="d455"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="depth_width"         default="848"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="848"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="848"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="depth_fps"           default="90"/>
  <arg name="infra_fps"           default="90"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="200"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="filters"                   default="pointcloud"/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="true"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>

  <rosparam>
      /d455/rgb_camera/enable_auto_exposure: true
      /d455/rgb_camera/auto_exposure_priority: false
      /d455/stereo_module/inter_cam_sync_mode: 1
  </rosparam>
  
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
      <arg name="depth_height"             value="$(arg depth_height)"/>
      <arg name="enable_depth"             value="$(arg enable_depth)"/>

      <arg name="color_width"              value="$(arg color_width)"/>
      <arg name="color_height"             value="$(arg color_height)"/>
      <arg name="enable_color"             value="$(arg enable_color)"/>

      <arg name="infra_width"              value="$(arg infra_width)"/>
      <arg name="infra_height"             value="$(arg infra_height)"/>
      <arg name="enable_infra1"            value="$(arg enable_infra1)"/>
      <arg name="enable_infra2"            value="$(arg enable_infra2)"/>

      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>
</launch>
```

MartyG-RealSense, 2021-09-15:
Hi @YuxuanSnow  

1.  There are recommended pairings of values for the RealSense IMU's gyro and accelerometer.  They are 200 for gyro and 63 for accelerometer, or 400 for gyro and 250 for accelerometer.  As you are already using 200 for gyro in your launch file, could you try changing accel_fps to **63** please to see whether it makes a positive difference?

2.  Another RealSense ROS user recently asked about the difference between these two topics because they looked the same to that user.  My response is in https://github.com/IntelRealSense/realsense-ros/issues/2047#issuecomment-904665666

YuxuanSnow, 2021-09-15:
For the first issue, I think it is better, but the frequency still varies. Is it unavoidable if I want to use the ROS version? Because the message type "std_msgs/imu" have angular velocity and gyro velocity together, but they sometimes cannot be provided simultaneously because of the Hardware. According to my observation, the realsense-viewer doesn't suffer from this because the gyro and accelero are separately displayed.

For the second problem, I found this in the issue you noticed.
>As the topic names suggest, /camera/depth/image_rect_raw contains purely depth data, whilst /camera/aligned_depth_to_color/image_raw is a topic containing color data that is aligned to depth that is only published when align_depth is set to True.
And I am not sure about the meaning. The alignment is "align depth data to color image" or "align color data to the depth image"? Also the word "align", does that mean after the alignment, the same pixel in the depth image and the color image should have the same object location?

With best regards,
Yuxuan

MartyG-RealSense, 2021-09-15:
The gyro and accel topics in ROS are combined together into the single **/imu** topic if the launch includes a definition for **unite_imu_method**.  Otherwise, there are seperate gyro and accel topics published if unite_imu_method is not defined.

[image]

The **align_depth** instruction aligns **depth to color** and publishes that alignment on two aligned-data topics, **/camera/aligned_depth_to_color/image_raw** and **/camera/aligned_depth_to_color/camera_info**

A depth to color alignment adjusts the field of view (FOV) size of the depth stream to match the FOV size of the color stream.  More information about the principles of alignment can be found in the opening paragraph of the link below.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align#overview

YuxuanSnow, 2021-09-16:
Thank you for the reply.
Still for the imu: it seems like the accelerometer doesn't support 63 hz. 
[image]
The imu measurement sometime still has no message output. Since I already use the linear interpolation as the unite_imu_method, I don't know why that happens. Could it about the queue size of the ros imu publisher?

YuxuanSnow, 2021-09-16:
I tested on another PC and found out the problem: it seems like that the all USB ports of my laptop are problematic. The same camera and the same launch file work very well on another PC.
Thank you for answering my questions.
With best regards,
Yuxuan

closed 2021-09-16 by YuxuanSnow

MartyG-RealSense, 2021-09-16:
I was just looking at your case now and was about to respond further, so thanks very much for your timely update  :)