realsenseai/realsense-ros #2082: apt-get install ros-dashing-realsense2-camera fails

colinator, 2021-09-20:
$ sudo apt-get install ros-$ROS_DISTRO-realsense2-camera
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-dashing-realsense2-camera

Same for foxy...

colinator, 2021-09-20:
Answered. First, do this:

sudo apt update && sudo apt install curl gnupg2 lsb-release
sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key  -o /usr/share/keyrings/ros-archive-keyring.gpg
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/ros2.list > /dev/null

colinator, 2021-09-20:
(ubuntu 18.04)

VasyaSmolyar, 2022-07-11:
Do you have any success on this?

MartyG-RealSense, 2022-07-13:
Hi, the RealSense ROS2 wrapper is now based on the current **ros2_beta** branch introduced in November 2021 rather than the **ros2** branch that was in use in September 2021 when this case was originally created.

https://index.ros.org/r/realsense2_camera/

The ros2_beta branch can be built from source code for ROS2 Dashing and Ubuntu 18.04.

MartyG-RealSense, 2022-07-19:
Hi @VasyaSmolyar  Do you require assistance, please?  Thanks!

closed 2022-07-25 by MartyG-RealSense