realsenseai/realsense-ros #2084: Timestamp of color frame and depth frame

YuxuanSnow, 2021-09-21:
Dear developers,

i want to ask that what's the definition of the timestamp? Is it exposure start time? Or is it the stop time of the exposure? Or even is it the mid-exposure point?

Besides, only depth stream can be successfully triggered by a signal with my current driver. Is it already fixed for color stream in the new driver?

With best regards,
Yuxuan

MartyG-RealSense, 2021-09-21:
Hi @YuxuanSnow  There are a range of different timestamps that have different characteristics.  In https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803 a RealSense team member provides a list of timestamp types and descriptions for them.

Regarding your second question about how only the depth stream can be triggered, can you provide more information about this problem please?  Thanks!

YuxuanSnow, 2021-09-21:
Thanks, the description of timestamp is clear and understandable.

For the second question: i have a trigger board, which can send 30 Hz signal to D455 camera. For the current driver i use, only the depth stream can be triggered. That means, depth frame is frozen until the triggering signal arrives. However, the color frame is independent of the triggering signal. My colleague asked you before, and the answer was only the depth frame triggering driver is accomplished, and the color frame triggering function would be released later. Therefore, i want to ask if you already released that function for color frame.

With best regards,
Yuxuan

MartyG-RealSense, 2021-09-21:
If you are currently setting your slave camera's Inter Cam Sync Mode to '2' (Slave) then setting it to '3' should set the camera to Full Slave mode, where color is synchronized as well as depth.

[image]

YuxuanSnow, 2021-09-22:
Hi,

i also saw this document. However, it doesn't work as i expected. 

I set the inter_cam_sync_mode to 0, 1, 2, 3, both depth stream and color stream are not frozen. When i set the variable to 4, which is not referred in the document, the depth frame is frozen, and color frame is not frozen. I am confused. Please tell me how i should set up the launch file.

I attach the launch file i used:
```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="d455"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="depth_width"         default="848"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="848"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="848"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="depth_fps"           default="90"/>
  <arg name="infra_fps"           default="90"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="63"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="filters"                   default="pointcloud"/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="true"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>

  <rosparam>
      /d455/rgb_camera/enable_auto_exposure: true
      /d455/rgb_camera/auto_exposure_priority: false
      /d455/stereo_module/inter_cam_sync_mode: 4
  </rosparam>
  
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
      <arg name="depth_height"             value="$(arg depth_height)"/>
      <arg name="enable_depth"             value="$(arg enable_depth)"/>

      <arg name="color_width"              value="$(arg color_width)"/>
      <arg name="color_height"             value="$(arg color_height)"/>
      <arg name="enable_color"             value="$(arg enable_color)"/>

      <arg name="infra_width"              value="$(arg infra_width)"/>
      <arg name="infra_height"             value="$(arg infra_height)"/>
      <arg name="enable_infra1"            value="$(arg enable_infra1)"/>
      <arg name="enable_infra2"            value="$(arg enable_infra2)"/>

      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>
</launch>
```

MartyG-RealSense, 2021-09-22:
When using **genlock** mode (4 and above), the rules differ in regard to the FPS that should be set for master and slave.  This is described in section **2.  Principles of Operation** of the 'External Synchronization' white-paper document.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-2-principles-of-operation

The following quote from this section describes how the slave camera FPS should be a multiple of the Master FPS in genlock mode.

**********

One Intel RealSense camera can be placed into “Master” mode, while all other cameras are placed into “Genlock” mode. 
 However, the same frame rate restrictions apply as outlined in Table 1.  So, in order to have all cameras stream at 30fps, for example, the Master must be set to 30fps, and all Genlock cameras must be natively set to 90fps modes with burst count set to 1.  The master camera will then deliver the trigger at 30fps, and all Genlock cameras will trigger in sync with the master.

YuxuanSnow, 2021-09-23:
Thank you.

As you can see in the launch file, i already set the the camera into Genlock mode (inter_camera_syn=4). However, only depth frame is only captured when the triggering signal arrives. Should i set the FPS of color stream also to 90 to solve this problem?

MartyG-RealSense, 2021-09-23:
As stated in the External Synchronization paper, setting the Inter Cam Sync Mode to '3' sets the camera to **Full Slave** mode, where RGB is synchronized in addition to depth.  Setting it to '4' or above sets it to **Genlock Slave** mode.

[image]

In regard to syncing to RGB in mode 4 and above though, Section 4 point 3 of the paper suggests a scenario where a non-RealSense RGB camera (such as an HD camera) is set as the Master camera so that RealSense slave cameras can synchronize with it.  This suggests that mode 4+ is different to mode 3 and that both depth and RGB will not be synchronized if a RealSense camera is set to mode 4 or higher.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-4-genlock-usages

MartyG-RealSense, 2021-09-29:
Hi @YuxuanSnow  Do you require further assistance with this case, please?  Thanks!

YuxuanSnow, 2021-09-29:
Thank you for taking care of this. I still didn't figure out how to let it work properly, but currently I don't have access to the lab and change to use the camera. I would open an issue again in the future in the case that I still need your help.

Appreciate and have a nice day.

closed 2021-09-29 by YuxuanSnow

MartyG-RealSense, 2021-09-30:
Thanks very much @YuxuanSnow for the update.  Please do feel welcome to open a new issue in the future if you need help.

wt160, 2022-11-23:
@MartyG-RealSense Hi, thanks for the explaination above. I read through librealsense #2188 issue. But I am still not sure how those different timestamps(Sensor_timestamp << Frame_timestamp.
Backend_timestamp << Time_of_arrival.)  relate to the timestamps in the header of ros messages. 
Could you please give some insight on my question?
Thank you very much!!