realsenseai/realsense-ros #2085: How to Disable GLSL on Ubuntu(ROS)

Yoshihito-M, 2021-09-22:
Hi, everyone.
I posted isuue “#2079” that asked how to apply "opensource_tracking.launch" to SLAM with D455.
My trouble seemed to be fixed by the answer to that, but I still have this trouble.
I note my troubles in more detail as following;
・rgbd.rviz appears but it displays PointCloud for about half a minute. However, it displays no PointCloud after that.
・Even if killing the launch file and rebooting terminal, it displays no PointCloud.

Meanwhile, I found the issue “#8110” that note RealSense user could not run the Viewer correctly because the GLSL graphics functions were enabled in the Viewer's settings.
In fact, I had same issue mentioned at “#8110” and I could fix that by the advice.

Considering the issue, I guess that rviz displays no PointCloud when I don’t disable these options on Ubuntu(ROS).
However, I’m not sure how to do that.

I’d like you to advice me.

MartyG-RealSense, 2021-09-22:
GLSL would not be enabled by default as support for it has to be deliberately programmed into an application, as is the case with the RealSense Viewer.  So the cause of your RViz pointcloud problem is likely to be something else.

As you are able to display a pointcloud for half a minute, that suggests too that your ROS setup is okay and there is perhaps a communication problem that stops pointcloud data from continuing to be produced.

Do you have any error or warning messages in your ROS terminal when the problem occurs, please?

Yoshihito-M, 2021-09-27:
@MartyG-RealSense 
Thanks for your comment.
I'm sorry for taking long to write you back.

my published issue has a mistake, "half a minute" is incorrect and "30 seconds" is correct.
when the problem occurs, I get the following error message.

-----------------------------------------------------
[ WARN] [1631755128.536455102]: Could not get transform from t265_odom_frame to d400_link after 0.200000 seconds (for stamp=1631755127.842958)! Error="Lookup would require extrapolation into the past.  Requested time 1631755127.842957973 but the earliest data is at time 1631755127.847455025, when looking up transform from frame [d400_link] to frame [t265_odom_frame]. canTransform returned after 0.20089 timeout was 0.2.".
[ WARN] [1631755128.536636603]: We received odometry message, but we cannot get the corresponding TF t265_odom_frame->d400_link at data stamp 1631755127.842958s (odom msg stamp is 1631755127.847455s). Make sure TF of odometry is also published to get more accurate pose estimation. This warning is only printed once.
 15/09 18:18:48,709 ERROR [139778964055808] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
[ INFO] [1631755129.082265931]: rtabmap (1): Rate=1.00s, Limit=0.000s, RTAB-Map=0.5360s, Maps update=0.0005s pub=0.0021s (local map=1, WM=1)
 15/09 18:19:25,129 WARNING [140071936177920] (tm-device.cpp:1291) SLAM_ERROR Speed

[ WARN] [1631755747.058147942]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info

-----------------------------------------------------

MartyG-RealSense, 2021-09-27:
Your log mentions the warning **Could not get transform from t265_odom_frame to d400_link**.   I do not know why this message would be generated by the **opensource_tracking.launch** launch file, as I would expect it from the **rs_d400_and_t265.launch** launch file.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_d400_and_t265.launch#L49

Can you confirm please if you are only using a D455 camera and are not also using a RealSense T265 Tracking Camera?

closed 2021-10-09 by MartyG-RealSense