realsenseai/realsense-ros #2086: Convert depth map to point cloud

caobingquan6, 2021-09-22:
Hello, the camera model I am using is L515. I want to know how the depth map and point cloud are converted in the code. But I did not find a detailed calculation process in the code.

I found that the conversion is directly obtained by the following code. But please answer for the specific conversion process.

<rs_frame.h>
/**
* When called on Points frame type, this method returns a pointer to an array of 3D vertices of the model
* The coordinate system is: X right, Y up, Z away from the camera. Units: Meters
* \param[in] frame       Points frame
* \param[out] error      If non-null, receives any error that occurs during this call, otherwise, errors are ignored
* \return                Pointer to an array of vertices, lifetime is managed by the frame
*/
rs2_vertex* rs2_get_frame_vertices(const rs2_frame* frame, rs2_error** error);

In other words, I want to understand the camera parameters and detailed calculation process of the depth map and point cloud conversion.

MartyG-RealSense, 2021-09-22:
Hi @caobingquan6   A better way to calculate a point cloud that is generated using vertices would be to use **points.get_vertices**, a process that is described step by step in the SDK's C++ pointcloud tutorial guide linked to below.  Full source code for the project can also be found at te top of that page.

https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud

caobingquan6, 2021-09-22:
Yes, I saw this. But I want to know what is the specific calculation process of points.get_vertices()? Because the conversion of depth images into point clouds requires computation. But I did not find this calculation process.

What I found is as follows:

<rs_frame.hpp>
        const vertex* get_vertices() const
        {
            rs2_error* e = nullptr;
            auto res = rs2_get_frame_vertices(get(), &e);
            error::handle(e);
            return (const vertex*)res;
        }

Then:
<rs_frame.h>
/**
* When called on Points frame type, this method returns a pointer to an array of 3D vertices of the model
* The coordinate system is: X right, Y up, Z away from the camera. Units: Meters
* \param[in] frame       Points frame
* \param[out] error      If non-null, receives any error that occurs during this call, otherwise, errors are ignored
* \return                Pointer to an array of vertices, lifetime is managed by the frame
*/
rs2_vertex* rs2_get_frame_vertices(const rs2_frame* frame, rs2_error** error);

However, the above process does not have a detailed calculation process. This is what I need.

MartyG-RealSense, 2021-09-22:
I'm not aware of a reference that describes points.get_vertices in the detail that you have in mind, though further details about how it works can be found in https://github.com/IntelRealSense/librealsense/issues/1783

caobingquan6, 2021-09-22:
In other words, we can convert the depth image into a point cloud through the camera's internal parameter matrix (such as focal length, aperture center, etc.). Where can I get this information?

MartyG-RealSense, 2021-09-22:
The L515 data sheet document provides the technical details of the L515's color sensor.

https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet

[image]

You can also access the L515's intrinsic and extrinsic information in real-time with the the rs-enumerate-devices tool's calibration mode by launching the tool with the instruction **rs-enumerate-devices -c**

[image]

closed 2021-10-05 by MartyG-RealSense