realsenseai/realsense-ros #2087: Cannot get RGB image and pointcloud when using "filters:=pointcloud" in command

JiahaoPlus, 2021-09-22:
Hi, I am using Jetson TX2 (arm64) and installed the realsense-ros using "sudo apt-get install ros-$ROS_DISTRO-realsense2-camera".
When I run "roslaunch realsense2_camera rs_camera.launch", I can see RGB (/camera/color/image_raw) and depth (/camera/depth/image_rect_raw) in rviz.
However, when I run "roslaunch realsense2_camera rs_camera.launch filters:=pointcloud", the depth (/camera/depth/image_rect_raw) dropped to around 8 fps, and there are no RGB (/camera/color/image_raw) and pointcloud (/camera/depth/color/points).
How should I deal with this? Thanks.

JiahaoPlus, 2021-09-22:
**The logs:**
$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to /home/administrator/.ros/log/c1a219a4-1af8-11ec-b6e9-00044bf87676/roslaunch-cpr-ethz05-16593.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.131.1:42393/

SUMMARY
-------

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 81 lines cut ...]
[ INFO] [1632346931.853047603]: insert Color to RGB Camera
[ INFO] [1632346931.952146744]: SELECTED BASE:Depth, 0
[ INFO] [1632346932.002131835]: RealSense Node Is Up!
 22/09 23:42:12,021 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[ WARN] [1632346932.170506263]: 
 22/09 23:42:12,171 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 23:42:12,222 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 23:42:12,272 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 23:42:12,424 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1632346932.477303842]: publishing unordered pointcloud.
[ WARN] [1632346933.021866561]: No stream match for pointcloud chosen texture Process - Color
 22/09 23:42:13,160 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 23:42:13,211 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 23:42:13,363 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 23:42:13,414 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 22/09 23:42:22,030 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 23:42:31,088 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/09 23:48:51,413 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/09 23:48:52,464 WARNING [546987561344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2021-09-23:
Hi @JiahaoPlus   There have been some similar cases with Jetson boards in the last couple of months (problems occur when launching with pointcloud enabled but not when pointcloud is not used), as described in https://github.com/IntelRealSense/realsense-ros/issues/2077

As a starting point in investigating your case, could you try the roslaunch instruction below please?

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30 filters:=pointcloud initial_reset:=true**

Does this instruction make any difference to your results?

JiahaoPlus, 2021-09-23:
Hi @MartyG-RealSense , thanks for your reply!
I saw the other issues and the solutions there cannot solve my problem.
I also tried the instruction you mentioned above, and I still cannot get the color images and pointclould.

MartyG-RealSense, 2021-09-23:
And the same happens if you use this?

**roslaunch realsense2_camera rs_camera.launch align_depth:=true filters:=pointcloud**

JiahaoPlus, 2021-09-23:
Yes, the same...

MartyG-RealSense, 2021-09-24:
Next, can you test whether you can obtain a pointcloud with an **rs_rgbd** launch using the instructions below:

1.  Install support for an rgbd launch on Melodic: 

**sudo apt-get install ros-melodic-rgbd-launch**

2.  Perform an rs_rgbd roslaunch:

**roslaunch realsense2_camera rs_rgbd.launch**

JiahaoPlus, 2021-09-24:
Thanks. Using rs_rgbd.launch I can get the "/camera/depth_registered/points", but it's normally less than 1Hz, and sometimes shows "no new messages".

MartyG-RealSense, 2021-09-24:
Let's look at this issue from the perspective of your ROS wrapper having been installed from packages with **sudo apt-get install ros-melodic-realsense2-camera** ('Method 1' of the ROS wrapper installation instructions).  

I recall that **Doronhi** the RealSense ROS wrapper has stated in https://github.com/IntelRealSense/realsense-ros/issues/1177#issuecomment-674291080 that CUDA graphics acceleration support for devices with an Nvidia GPU (such as Jetson boards) is not included in the ROS distribution packages.  Since CUDA support can provide acceleration for alignment, pointcloud and color conversion operations, a lack of CUDA support in your librealsense SDK build may negatively affect performance on your Jetson when enabling pointcloud.

MartyG-RealSense, 2021-09-30:
Hi @JiahaoPlus  Do you require further assistance with this case, please? Thanks!

JiahaoPlus, 2021-09-30:
Yes I am still trying it. Will update later.

MartyG-RealSense, 2021-09-30:
Thanks very much @JiahaoPlus - I look forward to your update.  Good luck!

JiahaoPlus, 2021-10-01:
Hi @MartyG-RealSense , now the realsense-ros still only supports librealsense 2.48 (not the newest 2.49), right? So I should compile librealsense 2.48 and then run realsense-ros? Thanks

MartyG-RealSense, 2021-10-01:
Yes, build librealsense SDK 2.48.0 first and then wrapper 2.3.1 secondly.  There is not currently a ROS wrapper for matching with 2.49.0.

JiahaoPlus, 2021-10-01:
Thanks! I saw the latest ros wrapper is 3.2.2. What's the difference between it and 2.3.1 (why should I use 2.3.1)? Thanks for your answer

MartyG-RealSense, 2021-10-01:
Version numbers that start with '2' are the ROS1 wrapper for the Kinetic, Melodic and Noetic branches.  Version numbers that start with '3' are the ROS2 wrapper for the Dashing, Eloquent, Foxy and Galactic branches.  So if you are using Melodic then you should use version numbers that start with '2', such as 2.3.1.

JiahaoPlus, 2021-10-04:
Hi @MartyG-RealSense , I built the librealsense SDK 2.48.0 first and then wrapper 2.3.1. One issue is that when I set the fps to 15, it is still 30. 

**My command:**
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud color_fps:=15 depth_fps:=15
... logging to /home/administrator/.ros/log/cc1dd55c-22ce-11ec-a034-00044bf87676/roslaunch-cpr-ethz05-7470.log

**Summary:**
PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 59 lines cut ...]
[ INFO] [1633361523.529103800]: Stereo Module was found.
[ INFO] [1633361523.546625480]: RGB Camera was found.
[ INFO] [1633361523.547376394]: Motion Module was found.
[ INFO] [1633361523.547541354]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1633361523.547639755]: Add Filter: pointcloud
[ INFO] [1633361523.549203535]: num_filters: 1
[ INFO] [1633361523.549294511]: Setting Dynamic reconfig parameters.
[ WARN] [1633361523.655473588]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 165000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1633361523.659249630]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1633361523.730922596]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1633361523.795206805]: Done Setting Dynamic reconfig parameters.
[ INFO] [1633361523.797596316]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1633361523.801568486]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1633361523.801698823]: setupPublishers...
[ INFO] [1633361523.818704662]: Expected frequency for depth = 30.00000
[... cut ...]
 04/10 17:32:04,299 WARNING [546660409728] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1633361524.301900938]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 04/10 17:32:04,612 WARNING [546668802432] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
 04/10 17:37:23,302 ERROR [546652017024] (ds5-options.cpp:88) Asic Temperature value is not valid!

MartyG-RealSense, 2021-10-04:
From ROS wrapper version **2.2.22** onwards, a custom stream configuration in the roslaunch instruction must specifiy three factors for each stream type being used - **width, height and FPS**.  If these 3 factors are not provided then the custom configuration is considered invalid by the roslaunch and the default stream configuration of the particular camera model being used is applied instead.  This is why your FPS changed to 30.

Please try the roslaunch instruction below:

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=848 depth_height:=480 depth_fps:=15 color_width:=1280 color_height:=720 color_fps:=15**

JiahaoPlus, 2021-10-05:
Thanks! @MartyG-RealSense 
Now when cpu is not busy, everything works fine. 
But when cpu is busy, the fps will drop.

JiahaoPlus, 2021-10-05:
Hi @MartyG-RealSense, thanks for your reply. I confirm that "I built the librealsense SDK 2.48.0 first and then wrapper 2.3.1", as illustrated in [#2087 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2087#issuecomment-933614507).
I saw the GPU was used when running rs_camera.launch, but I guess CPU is still needed in the procedure, and a busy cpu may cause fps dropping problem. If it is a normal phenomenon, I guess I can close this issue.
Thanks.

MartyG-RealSense, 2021-10-05:
If CUDA support is enabled in librealsense then it can only assist with a few specific types of processing operations (pointclouds, alignment and color conversion).  Other processing tasks that do not fall into those categories would be handled by the CPU and be vulnerable to processing 'spikes' that cause FPS lag.

If auto-exposure is enabled then disabling another option called **Auto-Exposure Priority** may help to protect against FPS drops.  This is because when Auto-Exposure is enabled and Auto-Exposure Priority is disabled, RealSense should attempt to enforce a constant FPS rate.  A method for doing this in the RealSense ROS wrapper is described in https://github.com/IntelRealSense/realsense-ros/issues/1349#issuecomment-681695259

JiahaoPlus, 2021-10-05:
Will try to disable Auto-Exposure Priority. Thanks a lot!

MartyG-RealSense, 2021-10-11:
Hi @JiahaoPlus  I apologize for the delay in responding further.  Do you still require assistance with this case, please?  Thanks!

MartyG-RealSense, 2021-10-12:
-  Jetson TX2 is a powerful computing device, so it may not be the case that you need a 'better' CPU.  There seems to be an issue for some RealSense ROS users with generating pointclouds on Jetson boards in general.

-  You should be able to set Auto-Exposure Priority on a D455 using the **/camera/rgb_camera/rs435_color_auto_exposure_priority: false** D435 command.  You could check afterwards whether it has been disabled by viewing the **dynamic_reconfigure** visual interface after launch with the command below:

**rosrun rqt_reconfigure rqt_reconfigure**

https://github.com/IntelRealSense/realsense-ros#set-camera-controls-using-dynamic-reconfigure-params

[image]

JiahaoPlus, 2021-10-19:
No, thanks!

closed 2021-10-19 by JiahaoPlus

lovedreamms, 2024-12-25:
Hello, I have encountered the same problem as well, and I want to know what the final solution is. I have read it several times, and the preliminary conclusion is that using self-compiled librealsense and rosrealsense2_camera can be accelerated with CUDA, so will it finally solve the problem?
If possible, please tell me the specific version that should be installed now

MartyG-RealSense, 2024-12-25:
Hi @lovedreamms  Are you using the ROS1 or ROS2 RealSense wrapper, please?

Whilst enabling CUDA support when compiling librealsense from source code will reduce the processing burden on the CPU when using pointcloud or alignment, it only works with Nvidia Jetson computing boards.

If you are using the ROS2 wrapper on a PC, you can accelerate pointclouds by building the wrapper from source code with 'colcon build' and adding the flag **UILD_ACCELERATE_GPU_WITH_GLSL** to the colcon build instruction.

`colcon build --cmake-args '-DBUILD_ACCELERATE_GPU_WITH_GLSL=ON'`

***********

If you are using the ROS1 wrapper and cannot obtain a pointcloud then a solution that typically works is to perform an **RGBD** launch.  The ROS1 wrapper does not have built-in support for an RGBD launch and that support must be first installed manually using one of the instructions below, depending on which ROS1 branch you are using. Have you installed this support, please?

**Kinetic**
sudo apt-get install ros-kinetic-rgbd-launch

**Melodic**
sudo apt-get install ros-melodic-rgbd-launch

**Noetic**
sudo apt-get install ros-noetic-rgbd-launch

Then once RGBD support is installed, [erform the RGBD launch.

`roslaunch realsense2_camera rs_rgbd.launch`

rs_rgbd.launch should by default generate a point cloud that is published at **depth_image_proc**