realsenseai/realsense-ros #2088: allow_no_texture_points setting width

pltlg, 2021-09-23:
Hello! 
I would like to get the biggest pointcloud possible from the D435i cam.

To achieve this I set the "allow_no_texture_points" param to true so the "/d435/depth/color/points" topic provides depth points even with no RGB values. (Since the depth FOV is bigger than RGB cam FOV)
The result is promising, I got uncolored depth points for near and far areas. (Z axis)
Based on the specification the width of the depth camera is bigger so I would sugggest that the topic should give additional (uncolored) depth points for the sides (X axis).

Do you have any suggestion how can I get the full FOV of depth camera?

MartyG-RealSense, 2021-09-23:
Hi @pltlg  When the pointcloud filter is used, the output is based on the **aligned_depth_to_color** image.  The way that depth to color alignment works is that the size of the depth FOV is changed to the size of the color FOV.  

This means that on camera models where the color FOV is smaller than the depth FOV (D435 and D435i), the outer regions of the depth image can be excluded from the aligned image.  This is because the smaller color FOV means that the color sensor literally could not see the areas of the scene that the depth sensor's wider FOV could see.  **allow_no_texture_points** enables the area of cut-off detail to be colored with zeros.  

If **color to depth** alignment is used in librealsense applications instead of depth to color then the image is stretched out to fit the larger depth FOV size.  I cannot think of a way to do color-to-depth alignment in the RealSense ROS wrapper though, other than to perhaps access librealsense from ROS via a Python **node script**.

The link below discusses color to depth alignment in librealsense using Python.

https://github.com/IntelRealSense/librealsense/issues/2560

The **scripts** folder of the RealSense ROS wrapper also provides some example Python scripts.

https://github.com/IntelRealSense/realsense-ros/tree/development/realsense2_camera/scripts

MartyG-RealSense, 2021-09-30:
Hi @pltlg  Do you require further assistance with this case, please? Thanks!

pltlg, 2021-09-30:
As you can see in the picture, if I set the allow_no_texture_points property the /d435/depth/color/points provides bigger depth view but the left side of the image is missing. (On the right side it tries to strech the RGB values to fit?)
Anyway, I would like to get the full depth data.

[image]

MartyG-RealSense, 2021-09-30:
I am not aware of stretching taking place on the right-side of the image during depth to color alignment in the RealSense ROS wrapper.  On the left edge of the image though there may be an **invalid depth band** area where depth data is absent, even before depth is aligned to color.  More information about this phenomenon is in https://github.com/IntelRealSense/librealsense/issues/6311#issuecomment-620261361

Would it be possible to move the camera further away to increase the amount of detail that fits into the field of view?

Aternatively, if you are able to use a second camera, you could conceivably place it to the left of the first camera, capture a point cloud from that second camera and then stitch the two individual point clouds together into a single cloud using the guide in the link below in order to widen the overall view.

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

chenhanpolyu, 2022-07-10:
> allow_no_texture_points

I also see this with my d435i camera, by setting "allow_no_texture_points = true", I get a bigger colored point cloud, but the left side is missing compared with the original point cloud

MartyG-RealSense, 2022-07-10:
Hi @chenhanpolyu  The ROS wrapper documentation states: "The depth FOV and the texture FOV are not similar.  By default, pointcloud is limited to the section of depth containing the texture.  You can have a full depth to pointcloud, coloring the regions beyond the texture with zeros, by setting allow_no_texture_points to true".

On the D435 and D435i camera models, the size of the color field of view (FOV) is smaller than the depth FOV size.  So on these camera models, when depth is aligned to color the outer regions of the depth image may be excluded from the aligned image.  The smaller color FOV size means that there are no color pixels to match with the depth pixels on the outer regions of the image, because the color sensor cannot see all of the areas of the real-world scene that the depth sensor's larger FOV can.

allow_no_texture_points can fill out the areas of missing texture data on the pointcloud but it cannot render color pixels that never existed in the first place because the color FOV could not see those areas of the scene.