realsenseai/realsense-ros #2089: Issue on running show_center_depth.py

GitNzino, 2021-09-24:
Hi!

I'm opening this issue because I found problem in running `rosrun realsense2_camera show_center_depth.py` in fact in my terminal appears this:
```
' @ error/constitute.c/WriteImage/1037.
from: can't read /var/mail/sensor_msgs.msg
from: can't read /var/mail/sensor_msgs.msg
from: can't read /var/mail/cv_bridge
' @ error/constitute.c/WriteImage/1037.
' @ error/constitute.c/WriteImage/1037.
' @ error/constitute.c/WriteImage/1037.
' @ error/constitute.c/WriteImage/1037.
/home/lorenzo/fake_workspace/src/realsense_ros/realsense2_camera/scripts/show_center_depth.py: line 9: syntax error near unexpected token `('
/home/lorenzo/fake_workspace/src/realsense_ros/realsense2_camera/scripts/show_ce'ter_depth.py: line 9: `if (not hasattr(rs2, 'intrinsics')):
```
I can confirm that my `realsense_ros` wrapper is succesfully installed because i see them on `rostopic` and on rviz.
I already tried to do `chmod +x *.py` in the scripts folder, but without any significant results.

What should i do?

Thank you for help!

MartyG-RealSense, 2021-09-24:
Hi @GitNzino  In https://github.com/IntelRealSense/realsense-ros/commit/62d36696999b2fbe100e6f92158dc5746a8d5efa#commitcomment-45621558 **Doronhi** the RealSense ROS wrapper developer describes how to run the **show_center_depth.py** example script.

pele15, 2021-09-25:
Hi!
If you are using an Ubuntu OS, its probably because the file was created in Windows, you might have to convert it to a Unix file using dos2unix:

For installation, run:

sudo apt install dos2unix

After that, you can convert files in either direction using one of the commands

dos2unix /path_to_your_file

Saipriyavk, 2021-09-25:
Hi!
If you are using an Ubuntu OS, its probably because the file was created in Windows, you might have to convert it to a Unix file using dos2unix:

For installation, run:

sudo apt install dos2unix

For conversion, you can run:

dos2unix /path_to_your_file

GitNzino, 2021-09-26:
@Saipriyavk but i think that all the ROS script should be created for linux os for the reason that ROS works only on ubuntu or i'm wrong?

@MartyG-RealSense thank you for answer, i'm trying them and them i will return with a certain response!

Thank you both!

Saipriyavk, 2021-09-26:
You might also need to add **#!/usr/bin/env python3** as the first line in the python file

GitNzino, 2021-09-27:
Thank you @Saipriyavk
I have just tried but nothing changes, sorry

GitNzino, 2021-09-30:
@MartyG-RealSense With python command the script runs, but there are not any method to use the `roslaunch` or `rosrun` command?

MartyG-RealSense, 2021-09-30:
The guide in the link below may be helpful for running a Python script using rosrun.  You only need to use step 2, and ignore step 2.5 which relates to C++.

https://www.theconstructsim.com/difference-run-code-ros-python-cpp/

MartyG-RealSense, 2021-10-07:
Hi @GitNzino  Do you require further assistance with this case, please?  Thanks!

GitNzino, 2021-10-07:
At the end I'm using python examples running in a ROS code, but I was not able to run Ros example. 
I'm closing the issue for now