realsenseai/realsense-ros #2090: How to set emitter_enabled to 0?

abylikhsanov, 2021-09-26:
Hi,

How can I set to 0 the emitter_enabled parameter to 0? (To turn off the emitter completely)
I tried to do:

`<param name="stereo_module/emitter_enabled" type="bool" value="false" />` , but it does not seem to work.

I am using ROS Noetic.

MartyG-RealSense, 2021-09-26:
Hi @abylikhsanov  **Doronhi** the RealSense ROS wrapper developer says in https://github.com/IntelRealSense/realsense-ros/issues/817#issuecomment-813382924 that emitter_enabled needs to be set to '0' instead of 'false' now.

stevemartinov, 2021-09-28:
Even after setting:
`<param name="stereo_module/emitter_enabled" type="int" value="0" />
`
But it does not seem to work

MartyG-RealSense, 2021-10-04:
Hi @abylikhsanov and @stevemartinov  do either of you require further assistance with this subject, please?  Thanks!

closed 2021-10-10 by MartyG-RealSense